coolsnowwolf/lede #7620: 我发现了导致AX6和AX3600无限重启的真正罪魁祸首是NSS！！！！  根本不是无线驱动的锅！！！！

hohhhhh, 2021-08-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ √] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
路由器之前无限重启，我去翻阅了相关的帖子后先后怀疑无线内存泄露、插件冲突等，但排查验证后发现明显不是，又打开.config分析原因，正在分析过程中突然想到：所有重启的机器都有一个共同点！！！那就是NSS ！！！
然后我去软件包里尝试手工卸载了qca-nss的所有项目，卸载好重启之后路由器神奇的稳定了！！！！
插件什么的随便开 再也不会重启了，啥事都没有了  但卸载之后日志会有不停的报错误

### (2) 路由器型号和固件版本

A：AX6、AX3600   OpenWrt R21.8.6 / LuCI Master (git-21.211.30217-03784c3)

### (3) 详细日志

A：

Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.370332] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.378101] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.384042] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.391820] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.397747] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.405062] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.411513] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.419215] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.425198] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.432489] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.439028] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.447643] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.484692] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:30 2021 kern.err kernel: [ 2033.516351] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:34 2021 kern.err kernel: [ 2037.731201] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:34 2021 kern.err kernel: [ 2037.731350] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED
Sat Aug  7 15:52:37 2021 kern.err kernel: [ 2040.838895] wlan1: NSS TX failed with error[8]: NSS_TX_FAILURE_NOT_ENABLED

hcym, 2021-08-07:
7800编译了，一直没敢刷，😂

Kaze1027, 2021-08-07:
早就发现了，所以一直在编译sfe版本，在群里也给大雕提过，不过大雕现在在做806x

hohhhhh, 2021-08-07:
> 早就发现了，所以一直在编译sfe版本，在群里也给大雕提过，不过大雕现在在做806x

我也是，刚才在qq上得到了一位大佬的帮助，现在也开始编译无ecm版本，诶？我怎么看老哥你这ID这么像那个大佬？刚才在QQ帮忙的不会是你吧 不会这么巧吧..................

ja5onwu, 2021-08-07:
> > 早就发现了，所以一直在编译sfe版本，在群里也给大雕提过，不过大雕现在在做806x
> 
> 我也是，刚才在qq上得到了一位大佬的帮助，现在也开始编译无ecm版本，诶？我怎么看老哥你这ID这么像那个大佬？刚才在QQ帮忙的不会是你吧 不会这么巧吧..................

无ECM版本编译出来.还会不会像卸载这样显示报错?还有想问下.无ECM版本是不是只需要把qca-nss-ecm-64 一个去掉就好了?

xsm1997, 2021-08-07:
关于这个报错，去mac80211里面把nss support关掉就好了。

xsm1997, 2021-08-07:
nss-ecm导致随机重启的问题，我关注很久了，在另一个issue  #7523 里面也有提到，所以为了稳定我现在都是不编译nss-ecm。

ja5onwu, 2021-08-07:
> 关于这个报错，去mac80211里面把nss support关掉就好了。

具体怎么操作呢?

xsm1997, 2021-08-07:
> > 关于这个报错，去mac80211里面把nss support关掉就好了。
> 
> 具体怎么操作呢?

make menuconfig，然后找到mac80211这个包（在device drivers/wireless drivers还是什么里面），Enter键进去，里面有个什么nss support，取消掉就好。

不过我记得最新版本不编译nss之后这个勾自动就会去掉（找不到这个项目），如果是用opkg卸载的nss-ecm的话，还请去掉nss-ecm重新编译。

（以上全凭记忆，难免有疏漏，请见谅）

coolsnowwolf, 2021-08-07:
nss  也有用来做无线加速的，卸载后无线能力也会下降

Boos4721, 2021-08-08:
[image]
我是真没遇见重启 这些插件全开

ramondelee, 2021-08-08:
> [image]
> 我是真没遇见重启 这些插件全开

在网络里看看NSS运行没有？

ramondelee, 2021-08-08:
编译不要NSS的包，如果要用SFE，在app配置里开SFE会自动取消NSS

Boos4721, 2021-08-08:
> 编译不要NSS的包，如果要用SFE，在app配置里开SFE会自动取消NSS

> > [image]
> > 我是真没遇见重启 这些插件全开
> 
> 在网络里看看NSS运行没有？

没有nss能用?.jpg

ramondelee, 2021-08-08:
@Boos4721 没看懂

AmadeusGhost, 2021-08-08:
然而你这个日志是 nss 给无线加速的部分出错了🌚

xsm1997, 2021-08-08:
就很玄学，一部分人从始至终不重启，另一部分人从nss支持之初就重启...

hohhhhh, 2021-08-08:
> 就很玄学，一部分人从始至终不重启，另一部分人从nss支持之初就重启...

是的，很玄学，重启的机器不管怎么编译只要加nss-ecm必重启，不重启的机器怎么搞都不重启，目前只知道它会导致疯狂重启，但不知道它为什么会导致重启

hohhhhh, 2021-08-08:
> 然而你这个日志是 nss 给无线加速的部分出错了🌚

因为开issue要日志...而开issue的时候机器上只有这个日志......

hohhhhh, 2021-08-08:
> 然而你这个日志是 nss 给无线加速的部分出错了🌚

因为开issue要日志...而开issue的时候机器已经卸载了nss-ecm并重启过，只有这个日志了......

ramondelee, 2021-08-08:
> 然而你这个日志是 nss 给无线加速的部分出错了🌚

我的AX6不要无线，求改配置的方法！！
试过编译配置里删除kmod-qca-nss-ecm-64，编译后NSS不运行

ja5onwu, 2021-08-08:
我的就没重启过,但是会SSRP用着用着内存占用越来越多,然后导致OOM.大雕之前说是应该去掉NSS加密支持就好了.但没说怎么单独去掉NSS加密支持.

ramondelee, 2021-08-08:
根据楼主的报错和恩山一个贴说闭源OP用无线显示CPU占用较高但不重启，是不是因为用NSS加速无线导致的重启呢？

hohhhhh, 2021-08-09:
> 根据楼主的报错和恩山一个贴说闭源OP用无线显示CPU占用较高但不重启，是不是因为用NSS加速无线导致的重启呢？

不清楚是不是因为这个  我反复卸载和编译试过之后只能确定就是NSS导致的重启，但重启是不是因为NSS加速了无线 我没法确定，我个人能力只能到这了

hcym, 2021-08-09:
7800和7500v2没什么事，也没什么质的变化，白忙活😜

xsm1997, 2021-08-09:
做个调查，排除下问题：用nss不重启的各位，电源适配器是用的AX6的自带电源吗？还是其他路由器的电源拿过来直接插的？

xsm1997, 2021-08-09:
非常像Kernel Panic，重启之前一瞬间蓝色指示灯会停住（保持原来的状态大概1s），然后重启。但是抓不到Log......

wuiiled, 2021-08-10:
AX3600不重启，原装适配器。内存泄漏会在50M剩余的情况下稳定下来，已经坚持了4天无重启，8.6源码

ja5onwu, 2021-08-10:
我的也是AX3600,不重启,只会内存一直减少最后OOM.不在乎OOM的.可以一直不重启用的.毕竟OOM后内存又回到200的状态了.楼上说的AX6会重启,可能真的是电源原因.毕竟NSS是硬件加速,启动后.AX6的1A电源带不动导致重启也可能的.

jingleijack, 2021-08-10:
> 我的AX3600，不重启，内存内存一直最后OOM。不在乎OOM的。可以一直不重启用的。OOM后内存又回到200的状态了。楼上说的AX6会重启，可能真的是电源原因。NSS是硬件加速，启动后。AX6的1A电源带不动导致重启也可能的。

AX6没有遇到重启！重启的全部是AX3600的用户！全部是3600的用户！悲剧的3600.

jingleijack, 2021-08-10:
AX6无限重启？别乱带节奏！论坛群里基本上所有的重启问题都是AX3600用户！别啥都把AX6带上“！nss驱动对AX6非常友好。
AX3600水土不服！ax3600确实对NSS有点问题。具体原因，也没有一个log重启日志。悲剧的AX3600

xsm1997, 2021-08-10:
@jingleijack 我的AX6和我朋友的AX6，开了nss都会无限重启的。

jingleijack, 2021-08-10:
> @jingleijack我的AX6和我朋友的AX6，看一眼nsss整体无限重启的。

这个和NSS没啥关系，可能极端程度出现重启，其他没问题，AK11目前所出现的问题就是内存泄漏还有有一点点，问题不大，真个扯不上重启的问题！刷机的步骤！升级的流程，缓存的清除等都有关系。假如真有你说的无限重启的问题，那应该没人刷这个固件了，源码的更新也就废了，所有出现问题先找自身原因，排出后，可能是源码的问题，那才是真的问题。

jingleijack, 2021-08-10:
> @jingleijack我的AX6和我朋友的AX6，看一眼nsss整体无限重启的。

目前很多AX6用户都在用openwrt日常使用，很稳定。重启的问题不存在。刷机姿势流程狠关键，还有集成的插件有冲突导致无限重启这个是AX6测试的时候最常见的问题。lean默认驱动设置，基本上没问题，不要刷聪明集成其他不明白的驱动就行。

xsm1997, 2021-08-10:
> > @jingleijack我的AX6和我朋友的AX6，看一眼nsss整体无限重启的。
> 
> 目前很多AX6用户都在用openwrt日常使用，很稳定。重启的问题不存在。刷机姿势流程狠关键，还有集成的插件有冲突导致无限重启这个是AX6测试的时候最常见的问题。lean默认驱动设置，基本上没问题，不要刷聪明集成其他不明白的驱动就行。

实测默认编译也会出现重启问题，确实是AX6会重启，而且AX6重启的人不在少数。你不重启不代表别人不会重启。我作为一个Contributor编译的事情还是会一些的......

Kaze1027, 2021-08-10:
ax6用nss确实会重启，重启表现为开机后在不定期时间突发的数次断时间重启，我电脑有线连接着表现为网口down了又up又down这样反复5~6次左右然后再继续正常工作一段时间之后再次发生这种重启，同时无线并不会OOM，并且我发现路由lan口尽量不接设备会好转很多，这来自一次我给朋友刷带nss的固件，想看看能运行多久，结果一晚上9个多小时起来并没有重启，因此怀疑这个重启和nss的有线驱动有点关系。目前日志没有抓，没时间去拆机。说ax6不会重启的你可以尝试找台ax6来用大雕源码默认编译试一下。

hohhhhh, 2021-08-10:
> AX6无限重启？别乱带节奏！论坛群里基本上所有的重启问题都是AX3600用户！别啥都把AX6带上“！nss驱动对AX6非常友好。
> AX3600水土不服！ax3600确实对NSS有点问题。具体原因，也没有一个log重启日志。悲剧的AX3600

你的不重启不代表别人的不重启，不会重启的AX6和AX3600确实有，但另外一些会重启的AX6和AX3600 就算啥插件都不加，只要有nss-ecm就是会无限重启，且这些无限重启的机器在移除nss-ecm后，无限重启状况皆消失。

hohhhhh, 2021-08-10:
> AX6无限重启？别乱带节奏！论坛群里基本上所有的重启问题都是AX3600用户！别啥都把AX6带上“！nss驱动对AX6非常友好。
> AX3600水土不服！ax3600确实对NSS有点问题。具体原因，也没有一个log重启日志。悲剧的AX3600

你的不重启不代表别人的不重启，不会重启的AX6和AX3600确实也有，那是真的稳，但另外一些会重启的AX6和AX3600，就算啥插件都不手动加全默认，只要有nss-ecm就是会无限重启，且这些无限重启的机器在移除nss-ecm后，无限重启状况皆消失。

admin6750217, 2021-08-11:
从论坛，到qq裙，在到贴吧，在到github，反馈重启的就那么几个人，唉，退货换一个吧。瞎折腾啥。还基本都是AX3600或者从qsdk刷至op的机油。不能用不用呗。

nwpuwy, 2021-08-11:
我的AX6昨天第一次刷OPENWRT，也是无限重启，只要WAN口连上光猫后，1到5分钟必重启，后面刷回官方固件了

rsd1987, 2021-08-13:
我也是AX6 开启 NSS 不会重启 也不会泄露内存

rsd1987, 2021-08-13:
目前是刷的这个 https://github.com/rsd1987/Redmi-AX6-DCJ

ja5onwu, 2021-08-16:
> > > 关于这个报错，去mac80211里面把nss support关掉就好了。
> > 
> > 
> > 具体怎么操作呢?
> 
> make menuconfig，然后找到mac80211这个包（在device drivers/wireless drivers还是什么里面），Enter键进去，里面有个什么nss support，取消掉就好。
> 
> 不过我记得最新版本不编译nss之后这个勾自动就会去掉（找不到这个项目），如果是用opkg卸载的nss-ecm的话，还请去掉nss-ecm重新编译。
> 
> （以上全凭记忆，难免有疏漏，请见谅）

请问编译时候怎么去掉NSS,路径是?

Kaze1027, 2021-08-17:
> > > > 关于这个报错，去mac80211里面把nss support关掉就好了。
> > > 
> > > 
> > > 具体怎么操作呢?
> > 
> > 
> > make menuconfig，然后找到mac80211这个包（在device drivers/wireless drivers还是什么里面），Enter键进去，里面有个什么nss support，取消掉就好。
> > 不过我记得最新版本不编译nss之后这个勾自动就会去掉（找不到这个项目），如果是用opkg卸载的nss-ecm的话，还请去掉nss-ecm重新编译。
> > （以上全凭记忆，难免有疏漏，请见谅）
> 
> 请问编译时候怎么去掉NSS,路径是?

在Kernal - Network support里找到qca-nss-ecm，将其设置为M或者N

Headcrabed, 2021-08-18:
openwrt论坛那边的版本他们测试是连接设备较少的情况下会OOM，但是连接设备较多的情况下反而不会出现OOM的情况。不知道和这里讨论的问题是否有关。

Rhystar, 2021-09-07:
你开ipv6了吗？我的mf269，开启ipv6后会重启，关闭ipv6后能正常使用，会不会是ipq8071a的nss和ipv6冲突

Headcrabed, 2021-09-07:
> 你开ipv6了吗？我的mf269，开启ipv6后会重启，关闭ipv6后能正常使用，会不会是ipq8071a的nss和ipv6冲突

这个应该是高通开源代码的问题，他们的开源代码相当混乱

idelame, 2021-09-08:
同问题，开了NSS会无限重启，现在编译sfe的版本。不敢用NSS。

xsm1997, 2021-09-10:
> 你开ipv6了吗？我的mf269，开启ipv6后会重启，关闭ipv6后能正常使用，会不会是ipq8071a的nss和ipv6冲突

关了ipv6，一天多没重启，我再测试测试

xsm1997, 2021-09-15:
几天测试下来，关了IPv6真的不会重启了，我一天一重启也不会OOM。@hohhhhh 和其他发生重启现象的人，是不是也在使用IPv6？我尝试取消编译luci-proto-ipv6 odhcpd odhcp6c，并且禁用掉WAN和LAN的IPv6分配，5天了重来没发生过重启了。

westlia, 2021-12-14:
> > @jingleijack我的AX6和我朋友的AX6，看一眼nsss整体无限重启的。
> 
> 这个和NSS没啥关系，可能极端程度出现重启，其他没问题，AK11目前所出现的问题就是内存泄漏还有有一点点，问题不大，真个扯不上重启的问题！刷机的步骤！升级的流程，缓存的清除等都有关系。假如真有你说的无限重启的问题，那应该没人刷这个固件了，源码的更新也就废了，所有出现问题先找自身原因，排出后，可能是源码的问题，那才是真的问题。

我就是AX6，为了开启ipv6折腾了两天，我一直怀疑我是不是哪里配置错了。
用你的Almighty-Edition12月12日版本不开启ipv6的情况不会自动重启，一开ipv6，故障和这位说的一模一样（Kaze1027：重启表现为开机后在不定期时间突发的数次断时间重启，我电脑有线连接着表现为网口down了又up又down这样反复5~6次左右然后再继续正常工作一段时间之后再次发生这种重启）。
换了这个没编译nss的固件（https://github.com/excelwang/OpenWrt-AX6-Action/releases/tag/2021.07.25-0150）开ipv6也不会自动重启。
{上面是没开BT的情况，用你的固件开bt+ipv6是秒死，我先打开bt然后编辑dhcp文件，再重启odhcpd，lan立刻就断开。换了没编译nss的固件开bt也没问题。我也不懂是不是nss的问题，至少我换这位兄弟编译的固件解决了问题。}

xsm1997, 2022-04-07:
> hi guys, I got some update there. Maybe there is a fix ongoing. Seems a init.d script cause this problem. reference: https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/6229 Fix(maybe) : [bitthief/openwrt@c197c54](https://github.com/bitthief/openwrt/commit/c197c54c8e97a996fd56cf7eed1b6af9a21ea96e)
> 
> Could anyone help to build and test this fix?

However I think I should just remove the init.d script in qca-nss-ecm and manually load the module later after boot, as described in the forum thread, which has not been done in the openwrt repo and nss-package repo of the author.

Need further testing. I will update in a few days.

xsm1997, 2022-04-08:
> hi guys, I got some update there. Maybe there is a fix ongoing. Seems a init.d script cause this problem. reference: https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-ax3600/55049/6229 [...]

Just did what I said, the repo is working perfectly without random rebooting even with ipv6 enabled. And issues mentioned in #7523 never occur again. I cannot test for memory leak related to wireless, because it never occurred to me even in this repo. BTW the author said he does not get memory leak.

For users who want to use, just remove the init.d and sysctl.d in the Makefile of qca-nss-ecm, and it will not get kernel panic on boot. 

https://github.com/bitthief/nss-packages/blob/7e3217436e8e8fc1869cff08d2ed0167699f727d/qca/qca-nss-ecm/Makefile#L44
https://github.com/bitthief/nss-packages/blob/7e3217436e8e8fc1869cff08d2ed0167699f727d/qca/qca-nss-ecm/Makefile#L49-L50

And then add a line in rc.local with `modprobe ecm` to load ecm.ko when boot completed. It is a race condition for loading nf-conntrack with nss-ecm altogether. And it will not get kernel panic if postponed.

Do not forget to add the scripts to enable rps of ethernet (https://github.com/coolsnowwolf/lede/issues/7296#issuecomment-877644290) https://github.com/coolsnowwolf/lede/blob/847a60f53b8f59bb3f784e10963a42a8d3b6ab41/target/linux/ipq807x/base-files/etc/init.d/bootcount#L24-L27
I get a 1Gbps hardware offloading with almost no CPU usage, and almost 800Mbps software offloading using Xray. Hooray!

closed 2023-12-22 by aiamadeus