coolsnowwolf/lede #7621: 编译报错

fqbone, 2021-08-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
mipsel-openwrt-linux-musl-gcc -I. -I./ -EL -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DZLIB_CONST -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/qin/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-4.4=ffmpeg-4.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include -DPIC -fpic   -march=24kc -std=c11 -fomit-frame-pointer -fPIC -pthread -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -Wdeclaration-after-statement -Wall -Wdisabled-optimization -Wpointer-arith -Wredundant-decls -Wwrite-strings -Wtype-limits -Wundef -Wmissing-prototypes -Wno-pointer-to-int-cast -Wstrict-prototypes -Wempty-body -Wno-parentheses -Wno-switch -Wno-format-zero-length -Wno-pointer-sign -Wno-unused-const-variable -Wno-bool-operation -Wno-char-subscripts -Os -fno-math-errno -fno-signed-zeros -fno-tree-vectorize -Werror=format-security -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=vla -Wformat -fdiagnostics-color=auto -Wno-maybe-uninitialized    -MMD -MF fftools/ffmpeg_hw.d -MT fftools/ffmpeg_hw.o -c -o fftools/ffmpeg_hw.o fftools/ffmpeg_hw.c
mipsel-openwrt-linux-musl-gcc -I. -I./ -EL -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DZLIB_CONST -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/qin/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-4.4=ffmpeg-4.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include -DPIC -fpic   -march=24kc -std=c11 -fomit-frame-pointer -fPIC -pthread -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -Wdeclaration-after-statement -Wall -Wdisabled-optimization -Wpointer-arith -Wredundant-decls -Wwrite-strings -Wtype-limits -Wundef -Wmissing-prototypes -Wno-pointer-to-int-cast -Wstrict-prototypes -Wempty-body -Wno-parentheses -Wno-switch -Wno-format-zero-length -Wno-pointer-sign -Wno-unused-const-variable -Wno-bool-operation -Wno-char-subscripts -Os -fno-math-errno -fno-signed-zeros -fno-tree-vectorize -Werror=format-security -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=vla -Wformat -fdiagnostics-color=auto -Wno-maybe-uninitialized     -MMD -MF fftools/cmdutils.d -MT fftools/cmdutils.o -c -o fftools/cmdutils.o fftools/cmdutils.c
fftools/cmdutils.c: In function 'print_all_libs_info':
fftools/cmdutils.c:1137:5: warning: 'avresample_version' is deprecated [-Wdeprecated-declarations]
     PRINT_LIB_INFO(avresample, AVRESAMPLE, flags, level);
     ^~~~~~~~~~~~~~
In file included from fftools/cmdutils.c:37:
./libavresample/avresample.h:148:10: note: declared here
 unsigned avresample_version(void);
          ^~~~~~~~~~~~~~~~~~
fftools/cmdutils.c:1137:5: warning: 'avresample_configuration' is deprecated [-Wdeprecated-declarations]
     PRINT_LIB_INFO(avresample, AVRESAMPLE, flags, level);
     ^~~~~~~~~~~~~~
In file included from fftools/cmdutils.c:37:
./libavresample/avresample.h:158:13: note: declared here
 const char *avresample_configuration(void);
             ^~~~~~~~~~~~~~~~~~~~~~~~
mipsel-openwrt-linux-musl-gcc -I. -I./ -EL -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DZLIB_CONST -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/qin/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-4.4=ffmpeg-4.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include -DPIC -fpic   -march=24kc -std=c11 -fomit-frame-pointer -fPIC -pthread -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include/opus -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/include -Wdeclaration-after-statement -Wall -Wdisabled-optimization -Wpointer-arith -Wredundant-decls -Wwrite-strings -Wtype-limits -Wundef -Wmissing-prototypes -Wno-pointer-to-int-cast -Wstrict-prototypes -Wempty-body -Wno-parentheses -Wno-switch -Wno-format-zero-length -Wno-pointer-sign -Wno-unused-const-variable -Wno-bool-operation -Wno-char-subscripts -Os -fno-math-errno -fno-signed-zeros -fno-tree-vectorize -Werror=format-security -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=vla -Wformat -fdiagnostics-color=auto -Wno-maybe-uninitialized    -MMD -MF fftools/ffmpeg.d -MT fftools/ffmpeg.o -c -o fftools/ffmpeg.o fftools/ffmpeg.c
fftools/ffmpeg.c: In function 'init_input_stream':
fftools/ffmpeg.c:2963:9: warning: 'thread_safe_callbacks' is deprecated [-Wdeprecated-declarations]
         ist->dec_ctx->thread_safe_callbacks = 1;
         ^~~
In file included from ./libavformat/avformat.h:312,
                 from fftools/ffmpeg.c:43:
./libavcodec/avcodec.h:1812:9: note: declared here
     int thread_safe_callbacks;
         ^~~~~~~~~~~~~~~~~~~~~
mipsel-openwrt-linux-musl-gcc -Llibavcodec -Llibavdevice -Llibavfilter -Llibavformat -Llibavresample -Llibavutil -Llibpostproc -Llibswscale -Llibswresample -L/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/qin/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro  -EL -march=24kc -Wl,--as-needed -Wl,-z,noexecstack -Wl,--warn-common -Wl,-rpath-link=:libpostproc:libswresample:libswscale:libavfilter:libavdevice:libavformat:libavcodec:libavutil:libavresample   -o ffmpeg_g fftools/ffmpeg_opt.o fftools/ffmpeg_filter.o fftools/ffmpeg_hw.o fftools/cmdutils.o fftools/ffmpeg.o  -lavdevice -lavfilter -lavformat -lavcodec -lswresample -lswscale -lavutil  -lm -L/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lasound -pthread -lm -lm -lbz2 -L/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lz -L/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lgnutls -pthread -lm -L/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lz -L/home/qin/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lfdk-aac -L/home/qin/lede/staging_dir/target- [... cut ...]
time: package/feeds/packages/ffmpeg/full/compile#309.20#25.77#324.74
    ERROR: package/feeds/packages/ffmpeg failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/home/qin/lede'
make[1]: *** [package/Makefile:110: /home/qin/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qin/lede'
make: *** [/home/qin/lede/include/toplevel.mk:230：world] 错误 2

minlang112, 2021-08-07:
似乎是缺少依赖libatomic ？

kivid, 2021-08-14:
一样的日志

miansan, 2021-08-16:
同样的错误

clk01, 2021-09-11:
相同的错误，望大佬查看一下

closed 2022-02-12 by Beginner-Go