coolsnowwolf/lede #7627: CR660X有时候会突然重启

hzjixiang, 2021-08-08:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

2. 详细叙述

 (1) 具体问题

A：最近发现路由器经常会重启，有时候一天重启两三次，主要发生在看视频（油管、抖音）时。

### (2) 路由器型号和固件版本

A：CR6608 R21.7.26 /

### (3) 详细日志

A：Sun Aug  8 23:12:46 2021 daemon.info hostapd: wlan0: STA 48:95:07:a4:87:9b IEEE 802.11: authenticated
Sun Aug  8 23:12:46 2021 daemon.info hostapd: wlan0: STA 48:95:07:a4:87:9b IEEE 802.11: associated (aid 2)
Sun Aug  8 23:12:46 2021 daemon.notice hostapd: wlan0: STA-OPMODE-SMPS-MODE-CHANGED 48:95:07:a4:87:9b static
Sun Aug  8 23:12:46 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 48:95:07:a4:87:9b
Sun Aug  8 23:12:46 2021 daemon.info hostapd: wlan0: STA 48:95:07:a4:87:9b RADIUS: starting accounting session D4272A6AFF43853B
Sun Aug  8 23:12:46 2021 daemon.info hostapd: wlan0: STA 48:95:07:a4:87:9b WPA: pairwise key handshake completed (RSN)
Sun Aug  8 23:12:46 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 48:95:07:a4:87:9b
Sun Aug  8 23:12:47 2021 daemon.warn odhcpd[1974]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Aug  8 23:13:55 2021 daemon.warn odhcpd[1974]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Aug  8 23:20:22 2021 daemon.warn odhcpd[1974]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Aug  8 23:26:35 2021 daemon.warn odhcpd[1974]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Aug  8 23:27:47 2021 kern.alert kernel: [89840.157738] CPU 2 Unable to handle kernel paging request at virtual address 0000003f, epc == 8ded6144, ra == 8d0ab674

之后就重启了

lexin8, 2021-08-09:
为什么要用这种机器刷openwrt，MTK无线跑不满就算了，重启有一定几率关闭状态，主路由又不如D2 512M内存。

hzjixiang, 2021-08-09:
> 为什么要用这种机器刷openwrt，MTK无线跑不满就算了，重启有一定几率关闭状态，主路由又不如D2 512M内存。

wifi6速度还可以，信号也不错，OP系统七百M也够用了。重启关闭是啥意思？？ 这机器256M内存足够用啊，基本上都剩余150M以上。 D2基本都是二手了，6608是新的，小米官方质保一年。

TXH201622, 2021-08-09:
7621 非dsa的有线驱动是有问题的，有莫名重启现象，尤其是开了HWNAT后。我用DSA的非常稳定。

liwenjie119, 2021-08-09:
一台D2一台CR660x，买cr660x就是因为ax6买不到买个低配wifi6，目前很稳定，没有遇到重启

coolsnowwolf, 2021-08-09:
Switch 版本的 HWNAT 不兼容 IPv6，请不要开启使用 IPv6

MeIsReallyBa, 2021-08-09:
> Switch 版本的 HWNAT 不兼容 IPv6，请不要开启使用 IPv6

和ipv6没关系，dsa的hwnat 一样不支持ipv6，就是switch的有线驱动有问题。

AmadeusGhost, 2021-08-09:
跟 ipv6 没关系，看样子是 crypto-eip93 的锅🌚

coolsnowwolf, 2021-08-10:
我去？

liwenjie119, 2021-08-12:
附上newifi3的日志，重启后wifi没起来，手动启动wifi才起来，大佬判断一下是哪里的问题

`Mon Aug  9 13:54:03 2021 kern.notice kernel: [    0.000000] Linux version 5.4.132 (runner@fv-az87-271) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3617-129bfbf30)) #0 SMP Fri Aug 6 10:07:10 2021
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] MIPS: machine is Newifi-D2
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon Aug  9 13:54:03 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Aug  9 13:54:03 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Aug  9 13:54:03 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Zone ranges:
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Early memory node ranges
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Mon Aug  9 13:54:03 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Mon Aug  9 13:54:03 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00042882
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00042882
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] Memory: 509412K/524288K available (6160K kernel code, 257K rwdata, 1292K rodata, 1236K init, 238K bss, 14876K reserved, 0K cma-reserved, 262144K highmem)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon Aug  9 13:54:03 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] CPU Clock: 1100MHz
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xfdb1960d81, max_idle_ns: 440795240157 ns
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3475018995 ns
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.000007] sched_clock: 32 bits at 550MHz, resolution 1ns, wraps every 3904515583ns
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.007799] Calibrating delay loop... 732.36 BogoMIPS (lpj=3661824)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.073986] pid_max: default: 32768 minimum: 301
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.078752] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.085949] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Aug  9 13:54:03 2021 kern.info kernel: [    0.096063] rcu: Hierarchical SRCU implementation.
[... cut ...]
Mon Aug  9 13:54:46 2021 user.notice : Added device handler type: veth
Mon Aug  9 13:54:46 2021 user.notice : Added device handler type: bridge
Mon Aug  9 13:54:46 2021 user.notice : Added device handler type: Network device
Mon Aug  9 13:54:46 2021 user.notice : Added device handler type: tunnel
Mon Aug  9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is enabled
Mon Aug  9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is setting up now
Mon Aug  9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is now up`

liwenjie119, 2021-08-12:
内核日志：
[    0.000000] Linux version 5.4.132 (runner@fv-az87-271) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3617-129bfbf30)) #0 SMP Fri Aug 6 10:07:10 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00042882
[    0.000000] Readback ErrCtl register=00042882
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509412K/524288K available (6160K kernel code, 257K rwdata, 1292K rodata, 1236K init, 238K bss, 14876K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 1100MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xfdb1960d81, max_idle_ns: 440795240157 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3475018995 ns
[    0.000007] sched_clock: 32 bits at 550MHz, resolution 1ns, wraps every 3904515583ns
[    0.007799] Calibrating delay loop... 732.36 BogoMIPS (lpj=3661824)
[    0.073986] pid_max: default: 32768 minimum: 301
[    0.078752] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085949] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096063] rcu: Hierarchical SRCU implementation.
[    0.101310] smp: Bringing up secondary CPUs ...
[    7.764137] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    7.764145] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    7.764156] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    7.764249] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.166068] Synchronize counters for CPU 1: done.
[    7.543870] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    7.543877] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    7.543884] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    7.543934] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.256713] Synchronize counters for CPU 2: done.
[    7.633435] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    7.633442] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    7.633450] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    7.633504] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.341913] Synchronize counters for CPU 3: done.
[    0.371757] smp: Brought up 1 node, 4 CPUs
[    0.379614] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.389371] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.396351] pinctrl core: initialized pinctrl subsystem
[    0.402956] NET: Registered protocol family 16
[... 837 lines cut ...]
[   80.327471] device rai0 entered promiscuous mode
[   80.333686] br-lan: port 3(rai0) entered blocking state
[   80.338952] br-lan: port 3(rai0) entered forwarding state
[   95.918384] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[   96.140525] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(4C-F2-02-B4-72-0E) !!!
[   96.148283] Receive EAPoL-Key frame from STA 4C-F2-02-B4-72-0E wcid(1)
[   96.158337] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(4C-F2-02-B4-72-0E) !!!
[   96.166280] Receive EAPoL-Key frame from STA 4C-F2-02-B4-72-0E wcid(1)
[   96.172865] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[   96.172865] 
[   96.227208] Rcv Wcid(1) AddBAReq
[... cut ...]
[   98.527195] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[   98.534959] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(2)
[   98.543500] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[   98.551237] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(2)
[   98.557877] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[   98.557877] 
[   98.632056] Rcv Wcid(2) AddBAReq
[   98.635319] Start Seq = 00000000

ericwang2006, 2021-08-14:
我是分别从不同的商家买了2台cr6608, 一台刷dsa驱动固件, 一台刷交换机固件, 都发生过自动重启现象

hzjixiang, 2021-08-14:
> 我是分别从不同的商家买了2台cr6608, 一台刷dsa驱动固件, 一台刷交换机固件, 都发生过自动重启现象(ipv6均未开启)

我也测试了DSA驱动的，重启的日志报错一样 。 内核的故障 epc ra 都一样，不知道谁懂这个？

Sat Aug 14 22:32:52 2021 kern.alert kernel: [  297.263183] CPU 0 Unable to handle kernel paging request at virtual address 0000003f, epc == 8d5d6144, ra == 8d66b674

hzjixiang, 2021-08-14:
> 附上newifi3的日志，重启后wifi没起来，手动启动wifi才起来，大佬判断一下是哪里的问题
> @coolsnowwolf
> Mon Aug 9 13:54:03 2021 kern.notice kernel: [ 0.000000] Linux version 5.4.132 (runner@fv-az87-271) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3617-129bfbf30)) #0 SMP Fri Aug 6 10:07:10 2021
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] printk: bootconsole [early0] enabled
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] MIPS: machine is Newifi-D2
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Initrd not found or empty - disabling initrd
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] VPE topology {2,2} total 4
> Mon Aug 9 13:54:03 2021 kern.warn kernel: [ 0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> Mon Aug 9 13:54:03 2021 kern.warn kernel: [ 0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> Mon Aug 9 13:54:03 2021 kern.warn kernel: [ 0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Zone ranges:
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Normal [mem 0x0000000000000000-0x000000000fffffff]
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] HighMem [mem 0x0000000010000000-0x0000000023ffffff]
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Movable zone start for each node
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Early memory node ranges
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] node 0: [mem 0x0000000000000000-0x000000001bffffff]
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] node 0: [mem 0x0000000020000000-0x0000000023ffffff]
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] On node 0 totalpages: 131072
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] Normal zone: 576 pages used for memmap
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] Normal zone: 0 pages reserved
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] Normal zone: 65536 pages, LIFO batch:15
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] HighMem zone: 65536 pages, LIFO batch:15
> Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
> Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14_4096 Mon Aug 9 13:54:03 2021 kern.debug kernel: [ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 130496 Mon Aug 9 13:54:03 2021 kern.notice kernel: [ 0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Writing ErrCtl register=00042882 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Readback ErrCtl register=00042882 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] Memory: 509412K/524288K available (6160K kernel code, 257K rwdata, 1292K rodata, 1236K init, 238K bss, 14876K reserved, 0K cma-reserved, 262144K highmem) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] rcu: Hierarchical RCU implementation. Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies. Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] NR_IRQS: 256 Mon Aug 9 13:54:03 2021 kern.notice kernel: [ 0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] CPU Clock: 1100MHz Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xfdb1960d81, max_idle_ns: 440795240157 ns Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3475018995 ns Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.000007] sched_clock: 32 bits at 550MHz, resolution 1ns, wraps every 3904515583ns Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.007799] Calibrating delay loop... 732.36 BogoMIPS (lpj=3661824) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.073986] pid_max: default: 32768 minimum: 301 Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.078752] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.085949] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear) Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.096063] rcu: Hierarchical SRCU implementation. Mon Aug 9 13:54:03 2021 kern.info kernel: [ 0.101310] smp: Bringing up secondary CPUs ... Mon Aug 9 13:54:03 2021 kern.warn kernel: [ 7.764137] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes. Mon Aug 9 13: [... cut ...]
> Mon Aug 9 13:54:46 2021 user.notice : Added device handler type: bridge
> Mon Aug 9 13:54:46 2021 user.notice : Added device handler type: Network device
> Mon Aug 9 13:54:46 2021 user.notice : Added device handler type: tunnel
> Mon Aug 9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is enabled
> Mon Aug 9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is setting up now
> Mon Aug 9 13:54:48 2021 daemon.notice netifd: Interface 'lan' is now up

> 我去？

我测试了 DSA驱动也是一样的问题。而且重启的报错地址完全一样，请大佬帮忙看看
address 0000003f, epc == 8d5d6144, ra == 8d66b674

xiaoguoyi, 2021-08-15:
稳定版仓库有没有这个型号，可以试试

junexie-op, 2021-08-19:
一样的问题，刷了padavan也会自动重启。想刷回原厂固件了。。。

hzjixiang, 2021-08-19:
> 一样的问题，刷了padavan也会自动重启。想刷回原厂固件了。。。

HIBOY的PDCN不会重启

junexie-op, 2021-08-19:
> > 一样的问题，刷了padavan也会自动重启。想刷回原厂固件了。。。
> 
> HIBOY的PDCN不会重启

还是会重启的每天都会重启，我现在用其他版本的3天重启一次。

ericwang2006, 2021-08-19:
> 一样的问题，刷了padavan也会自动重启。想刷回原厂固件了。。。

我做了个刷回的教程，希望能帮到你，目前我觉得也不排除是硬件批次的问题。
https://youtu.be/Y_l_9N3xh24

junexie-op, 2021-08-20:
> > 一样的问题，刷了padavan也会自动重启。想刷回原厂扣了。。。
> 
> 我好像是刷回的教程，希望能帮到你，目前我觉得还不能修复是硬件扫描的问题。
> https://youtu.be/Y_l_9N3xh24

好的，我先换K2P了，等修复好再拿出来用。

junexie-op, 2021-08-24:
刷了small_5的V3.3.0-7固件问题解决。

coolsnowwolf, 2026-09-06:
CR660X 报告只有一条未带符号解析的内核 paging fault，缺少完整崩溃栈、版本配置和复现方式；旧硬件/旧内核问题无法归因到当前仓库，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf