coolsnowwolf/lede #7637: 请问编译的时候如何修改左侧菜单栏的名称呢？

liyuhaolol, 2021-08-09:
比如NAS这个选项，我看其他人编译的固件，有的叫“网络存储”，有的叫“存储”，有的叫“NAS”，他们是如何进行修改菜单名称的呢？我尝试并不能修改lua来完成文本的改动，因为这个地方应该是走了国际化的简体中文翻译，文本改动并不是写死在lua里的，但是如何修改简体中文的翻译，我又没有头绪了。请问各位有什么办法修改后编译么？
本议题不是BUG。如果惹来作者不高兴，会关闭。

nitsort, 2021-08-09:
https://github.com/coolsnowwolf/lede/blob/5a001287e541a20fd2c9e9a00d2ffeddfa1740b1/package/lean/luci-app-vsftpd/po/zh-cn/vsftpd.po#L16
/feeds/luci/modules/luci-base/po/zh-cn/
package的po

coolsnowwolf, 2021-08-10:
修改语言包里面的翻译即可

liyuhaolol, 2021-08-10:
> 修改语言包里面的翻译即可
感谢解答，已经会弄了

liyuhaolol, 2021-08-10:
> https://github.com/coolsnowwolf/lede/blob/5a001287e541a20fd2c9e9a00d2ffeddfa1740b1/package/lean/luci-app-vsftpd/po/zh-cn/vsftpd.po#L16
> 
> 
> /feeds/luci/modules/luci-base/po/zh-cn/
> package的po

感谢解答，已经会弄了，谢谢

closed 2021-08-10 by liyuhaolol