coolsnowwolf/lede #7639: libpcap 编译失败

tomclassplus, 2021-08-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

SHELL= flock /workdir/openwrt/tmp/.libpcap-1.10.0.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "libpcap-1.10.0.tar.gz" "ed285f4accaf05344f90975757b3dbfe772ba41d1c401c2648b7fa45b711bdd4" "" "http://www.us.tcpdump.org/release/" "http://www.tcpdump.org/release/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.us.tcpdump.org/release/libpcap-1.10.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 90  912k   90  822k    0     0  1098k      0 --:--:-- --:--:-- --:--:-- 1096k
100  912k  100  912k    0     0  1216k      0 --:--:-- --:--:-- --:--:-- 1215k
Hash of the downloaded file does not match (file: 8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e, requested: ed285f4accaf05344f90975757b3dbfe772ba41d1c401c2648b7fa45b711bdd4) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.tcpdump.org/release/libpcap-1.10.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  912k  100  912k    0     0  1342k      0 --:--:-- --:--:-- --:--:-- 1342k
Hash of the downloaded file does not match (file: 8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e, requested: ed285f4accaf05344f90975757b3dbfe772ba41d1c401c2648b7fa45b711bdd4) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/libpcap-1.10.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  912k  100  912k    0     0  8076k      0 --:--:-- --:--:-- --:--:-- 8076k
Hash of the downloaded file does not match (file: 8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e, requested: ed285f4accaf05344f90975757b3dbfe772ba41d1c401c2648b7fa45b711bdd4) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/libpcap-1.10.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  8  912k    8 81662    0     0  73371      0  0:00:12  0:00:01  0:00:11 73305
100  912k  100  912k    0     0   527k      0  0:00:01  0:00:01 --:--:--  527k
Hash of the downloaded file does not match (file: 8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e, requested: ed285f4accaf05344f90975757b3dbfe772ba41d1c401c2648b7fa45b711bdd4) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/libpcap-1.10.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:104: /workdir/openwrt/dl/libpcap-1.10.0.tar.gz] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/libs/libpcap'
time: package/libs/libpcap/compile#0.29#0.12#4.38
    ERROR: package/libs/libpcap failed to build.
make[2]: *** [package/Makefile:116: package/libs/libpcap/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

lhn1989, 2021-08-10:
R4S也有同样的错误

xingkong0113, 2021-08-10:
等[PR 7636](https://github.com/coolsnowwolf/lede/pull/7636)合并了问题应该就可以解决了。

minfengyk, 2021-08-10:
libpcap-1.10.0.tar.gz下载失败，网络不好，换节点或换梯子。也可以直接下载回来，放dl文件夹，再继续编译。

xingkong0113, 2021-08-10:
我今天早上用 github actions编译也报这个错，国外网络不至于吧！我用电脑测试过错误日志里的地址，可以正常下载。

xingkong0113, 2021-08-10:
x86和redmi ac2100都有。

EnnawYang, 2021-08-10:
...等pr合并

tomclassplus, 2021-08-10:
前几个地址都正确，但hash校验不过，最后一个地址404

minfengyk, 2021-08-10:
我不是专业人士，只是和大家交流一下，不对的地方请指正。我的理解，curl和https下载使用的协议不一样，所以有时curl有时下的很慢或下载失败，https却可以顺利下载。我昨天编译遇见类似问题，香港节点https都下不回来，换美国节点才下回来，放dl文件夹，顺利编译成功。

WYC-2020, 2021-08-10:
明显的hash不对，最后一次直接404访问失败 最后下载失败，自然出错了

closed 2021-08-10 by coolsnowwolf

renchao05, 2021-08-12:
是因为hash不对导致的，解决办法：先修改/package/libs/libpcap/Makefile里面的PKG_HASH:=8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e
然后重新编译试一下，正常修改以后就可以了，如果再不行就使用这个链接http://www.us.tcpdump.org/release/libpcap-1.10.0.tar.gz下载后，上传到dl文件夹。