coolsnowwolf/lede #7643: pushbot问题

xiangfeidexiaohuo, 2021-08-10:
跟着固件一起编译，打开就是报错：

[image]

[image]

单独安装ipk也会报错：

[image]

返回再安装一次ipk，就ok了

感觉有点玄学

xiangfeidexiaohuo, 2021-08-10:
@Beginner-Go  

有点迷茫呀

Coolkids, 2021-08-10:
大概原因是htm文件每一行代码都是^M结尾 在vim编辑时候会显示出来
sed -i "s/^M//" filename 一下或者重新编辑下htm文件就能解决问题

Beginner-Go, 2021-08-10:
我缺少 iputils-arping 依赖 麻烦发我下

xiangfeidexiaohuo, 2021-08-10:
> iputils

大雕这源码有呀，iputils 这个

Beginner-Go, 2021-08-10:
我没有编译环境 需要ipk安装依赖

xiangfeidexiaohuo, 2021-08-10:
> 我没有编译环境 需要ipk安装依赖
[iputils-arping.zip](https://github.com/coolsnowwolf/lede/files/6960843/iputils-arping.zip)

里面有x86和aarch64的

xiangfeidexiaohuo, 2021-08-10:
> 大概原因是htm文件每一行代码都是^M结尾 在vim编辑时候会显示出来
> sed -i "s/^M//" filename 一下或者重新编辑下htm文件就能解决问题

试过，不是这个原因。

Beginner-Go, 2021-08-10:
[image]
没有问题

xiangfeidexiaohuo, 2021-08-10:
> [image]
> 没有问题

我装2次 就没问题，第一次装会提示那些错误。那个luci-app-serverchan也是。

Beginner-Go, 2021-08-10:
安装其他luci插件会报错?

ali2liu, 2021-08-10:
我的也遇到了

对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/pushbot/log'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/simpleform'.
A runtime error occured: /usr/lib/lua/luci/template.lua:74: Failed to load template 'pushbot/log'.
Error while parsing template '/usr/lib/lua/luci/view/pushbot/log.htm':
Syntax error in /usr/lib/lua/luci/view/pushbot/log.htm:7: unfinished string near '"

xiangfeidexiaohuo, 2021-08-10:
> 安装其他luci插件会报错?

其他不会 ，或者这个 pushbot上个版本 也没事

Beginner-Go, 2021-08-10:
第1  2张图片的报错 我这边遇到

Beginner-Go, 2021-08-10:
试试rm -rf /tmp/luci*

xiangfeidexiaohuo, 2021-08-10:
> rm -rf /tmp/luci*

没用，虽然我再安装一次 能用，但是带在固件里，刷完固件 打开 就是错误。

ali2liu, 2021-08-10:
编译出来的版本是3.12-9，到https://github.com/zzsj0928/luci-app-pushbot/releases下载3.12-10版本的ipk安装上可以了。

xiangfeidexiaohuo, 2021-08-10:
> 编译出来的版本是3.12-9，到[https://github.com/zzsj0928/luci-app-pushbot/releases下载3.12-10版本的ipk安装上可以了。](https://github.com/zzsj0928/luci-app-pushbot/releases%E4%B8%8B%E8%BD%BD3.12-10%E7%89%88%E6%9C%AC%E7%9A%84ipk%E5%AE%89%E8%A3%85%E4%B8%8A%E5%8F%AF%E4%BB%A5%E4%BA%86%E3%80%82)

东西是一样的，那边版本号加1而已。我知道再安装一次，就好了。，

但是带在固件里，打开就是错误啊，还要再装一次，带在固件里的意义在哪？所以　不能根本解决。

xingkong0113, 2021-08-10:
可以下载试试看我这个[编译的](https://github.com/xingkong0113/Actions-OpenWrt/actions/runs/1070887588)固件，我测试过，刷入是可以用luci-app-pushbot的。

Beginner-Go, 2021-08-10:
应该不是插件问题

Beginner-Go, 2021-08-10:
> 可以下载试试看我这个[编译的](https://github.com/xingkong0113/Actions-OpenWrt/actions/runs/1070887588)固件，我测试过，刷入是可以用luci-app-pushbot的。

你是用哪个源？

xingkong0113, 2021-08-10:
> > 可以下载试试看我这个[编译的](https://github.com/xingkong0113/Actions-OpenWrt/actions/runs/1070887588)固件，我测试过，刷入是可以用luci-app-pushbot的。
> 
> 你是用哪个源？

当然是 lede

fonlan, 2021-08-11:
确实是换行符的问题，把/usr/lib/lua/luci/view/pushbot路径下的htm文件的换行符换成Unix格式的就行了

fonlan, 2021-08-11:
> 确实是换行符的问题，把/usr/lib/lua/luci/view/pushbot路径下的htm文件的换行符换成Unix格式的就行了。
> 还有一个问题是这个目录下缺少client.htm。

好像不止这些htm文件，其他脚本文件的换行符也是windows格式的。。。

ybist, 2021-08-11:
> 确实是换行符的问题，把/usr/lib/lua/luci/view/pushbot路径下的htm文件的换行符换成Unix格式的就行了。
> 还有一个问题是这个目录下缺少client.htm。

换行符改了，配置  项下的正常了，但是 在线设备  还是500错误

500 内部服务器错误
对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/pushbot/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/simpleform'.
A runtime error occured: /usr/lib/lua/luci/template.lua:74: Failed to load template 'pushbot/client'.
Error while parsing template '/usr/lib/lua/luci/view/pushbot/client.htm':
No such file or directory
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

WYC-2020, 2021-08-11:
这个插件问题 一个在回车符，另外一个说client.htm不存在那个我是重新吧这句话cat>/usr/lib/lua/luci/view/pushbot/client.htm<<EOF删除重新粘贴，用的这个WinSCP工具编辑的，怀疑应该也是哪里不可见字符导致，你们自己去试下吧

Beginner-Go, 2021-08-11:
@xingkong0113 迁移过去的

Beginner-Go, 2021-08-11:
> @Beginner-Go 哈喽，lede那边的源是谁迁过去的？
> 我比对了从我源拉的和lede那边拉下来的，是不一样的。
> 
> **_zzsj0928/luci-app-pushbot: ./luasrc/view/pushbot/pushbot_status.htm_**
> 
> ```
> :~/lede$ cat -A ./package/luci-app-pushbot/luasrc/view/pushbot/pushbot_status.htm 
> <script type="text/javascript">//<![CDATA[$
> XHR.poll(3, '<%=url([[admin]], [[services]], [[pushbot]], [[status]])%>', null,$
> ^Ifunction(x, data) {$
> ^I^Ivar tb = document.getElementById('pushbot_status');$
> ^I^Iif (data && tb) {$
> ^I^I^Iif (data.running) {$
> ^I^I^I^Ivar links = '<em><b><font color=green>pushbot <%:RUNNING%></font></b></em>';$
> ^I^I^I^Itb.innerHTML = links;$
> ^I^I^I} else {$
> ^I^I^I^Itb.innerHTML = '<em><b><font color=red>pushbot <%:NOT RUNNING%></font></b></em>';$
> ^I^I^I}$
> ^I^I}$
> ^I}$
> );$
> //]]>$
> </script>$
> <style>.mar-10 {margin-left: 50px; margin-right: 10px;}</style>$
> <fieldset class="cbi-section">$
> ^I<p id="pushbot_status">$
> ^I^I<em><%:Collecting data...%></em>$
> ^I</p>$
> </fieldset>$
> ```
> 
> **_coolsnowwolf/lede: ./package/lean/luci-app-pushbot/luasrc/view/pushbot/pushbot_status.htm_**
> 
> ```
> :~/lede$ cat -A ./package/lean/luci-app-pushbot/luasrc/view/pushbot/pushbot_status.htm 
> <script type="text/javascript">//<![CDATA[^M$
> XHR.poll(3, '<%=url([[admin]], [[services]], [[pushbot]], [[status]])%>', null,^M$
> ^Ifunction(x, data) {^M$
> ^I^Ivar tb = document.getElementById('pushbot_status');^M$
> ^I^Iif (data && tb) {^M$
> ^I^I^Iif (data.running) {^M$
> ^I^I^I^Ivar links = '<em><b><font color=green>pushbot <%:RUNNING%></font></b></em>';^M$
> ^I^I^I^Itb.innerHTML = links;^M$
> ^I^I^I} else {^M$
> ^I^I^I^Itb.innerHTML = '<em><b><font color=red>pushbot <%:NOT RUNNING%></font></b></em>';^M$
> ^I^I^I}^M$
> ^I^I}^M$
> ^I}^M$
> );^M$
> //]]>^M$
> </script>^M$
> <style>.mar-10 {margin-left: 50px; margin-right: 10px;}</style>^M$
> <fieldset class="cbi-section">^M$
> ^I<p id="pushbot_status">^M$
> ^I^I<em><%:Collecting data...%></em>^M$
> ^I</p>^M$
> </fieldset>^M$
> ```

怎么你这个pr把文件全给编辑了一遍？

Beginner-Go, 2021-08-11:
你Closed吧 我来重新迁移下

EnnawYang, 2021-08-11:
> 
> 怎么你这个pr把文件全给编辑了一遍？
不用编辑，项目直接覆盖就可以了

zzsj0928, 2021-08-11:
我直接删除，然后重新拉的我那边的源。
显示是全部编辑而已

zzsj0928, 2021-08-11:
> 你Closed吧 我来重新迁移下

我就是重新迁移的

Beginner-Go, 2021-08-11:
嗯 刚刚试了
确实会这样 也是服了，搬运都能出bug 搞得把我给整不会了

zzsj0928, 2021-08-11:
> > 确实是换行符的问题，把/usr/lib/lua/luci/view/pushbot路径下的htm文件的换行符换成Unix格式的就行了。
> > 还有一个问题是这个目录下缺少client.htm。
> 
> 好像不止这些htm文件，其他脚本文件的换行符也是windows格式的。。。

因为我看到这个回复的时候也很纳闷，我不是一个Windows使用者，我的环境是macOS加VM Linux，所以不可能出现Windows换行符。
后来我两个都git下来才发现上面那个问题所在的。

Beginner-Go, 2021-08-11:
我也是在Linux环境下 完成提交等一系列操作，出现这个问题，把我给整不会了

Beginner-Go, 2021-08-11:
那个client.htm文件 是在插件运行后自动生成的

zzsj0928, 2021-08-11:
> 嗯 刚刚试了
> 确实会这样 也是服了，搬运都能出bug 搞得把我给整不会了

不过还是感谢 @xingkong0113  帮我合并到lede源 😄

Beginner-Go, 2021-08-11:
@zzsj0928 看看serverchan有没有这个情况

zzsj0928, 2021-08-11:
> @zzsj0928 看看serverchan有没有这个情况

嗯，我一起看下

xingkong0113, 2021-08-11:
当时提交pull request的时候没有修改过任何文件，只把.git目录删了。

WYC-2020, 2021-08-11:
大型翻车现场 哈哈

xingkong0113, 2021-08-11:
下次提交pull request看来要用wsl

zzsj0928, 2021-08-11:
> 当时提交pull request的时候没有修改过任何文件，只把.git目录删了。

这个着实有点古怪，你是在Windows下git的？

xingkong0113, 2021-08-11:
> > 当时提交pull request的时候没有修改过任何文件，只把.git目录删了。
> 
> 这个着实有点古怪，你是在Windows下git的？

是的。

zzsj0928, 2021-08-11:
> @zzsj0928 看看serverchan有没有这个情况

serverchan 正常

```
:~/lede/package/luci-app-serverchan$ cat -A luasrc/view/serverchan/status.htm 
<script type="text/javascript">//<![CDATA[$
XHR.poll(3, '<%=url([[admin]], [[services]], [[serverchan]], [[status]])%>', null,$
^Ifunction(x, data) {$
^I^Ivar tb = document.getElementById('serverchan_status');$
^I^Iif (data && tb) {$
^I^I^Iif (data.running) {$
^I^I^I^Ivar links = '<em><b><font color=green>serverchan <%:RUNNING%></font></b></em>';$
^I^I^I^Itb.innerHTML = links;$
^I^I^I} else {$
^I^I^I^Itb.innerHTML = '<em><b><font color=red>serverchan <%:NOT RUNNING%></font></b></em>';$
^I^I^I}$
^I^I}$
^I}$
);$
//]]>$
</script>$
<style>.mar-10 {margin-left: 50px; margin-right: 10px;}</style>$
<fieldset class="cbi-section">$
^I<p id="serverchan_status">$
^I^I<em><%:Collecting data...%></em>$
^I</p>$

```

xingkong0113, 2021-08-11:
gitfor windows+TortoiseGit

closed 2021-08-12 by coolsnowwolf