coolsnowwolf/lede #7646: 希望x86也增加AC58U那种调整cpu调度模式的luci。

ggllioio, 2021-08-10:
原因是现在用的z8350做软路由，但是在op下cpu始终5-600mhz，即使使用率占满也很少上升。导致软nat连300多m都跑不满。
手动执行
echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
policy1
policy2
policy3
将4个核心都改成性能模式，即可稳定在高频状态，性能大幅提升。
查到似乎是因为有些凌动cpu升频降频跟桌面u不太一样。应该有很多u有类似问题。在op论坛上也搜到很多同样问题。
现在是靠把命令加入rc.local来实现开机自动修改。如果有luci就方便多了，同时也能方便测试其它的调度模式。
希望维护者们能考虑下。谢谢。

coolsnowwolf, 2021-08-11:
好提议

WYC-2020, 2021-08-11:
动态适应不错

ifuturenan, 2021-08-23:
wyse 3040,应该是楼主所说的z8350，在不接dp视频负载的情况下开机，CPU默认频率始终为480MHz。使用了楼主所说的手动执行的方法，但没有效果。

ggllioio, 2021-08-25:
> wyse 3040,应该是楼主所说的z8350，在不接dp视频负载的情况下开机，CPU默认频率始终为480MHz。使用了楼主所说的手动执行的方法，但没有效果。

负载一定要接的，一直用假负载没遇到过问题。
以前用小马3700遇到过没负载睿频不正常的情况。后来类似场景就都加上假负载了。

chuangbo, 2021-12-10:
同样希望 luci-app-cpufreq 可以支持 x86。发现 j1900 默认是 powersave 模式，连接数一多就 CPU 100% 了，手动调节到 performance 以后正常了。

我看了一下代码，其实想自己改然后提 PR，但是不确定要如何安装测试。。。

coolsnowwolf, 2026-09-06:
这是针对 x86 CPU 调度器的 LuCI 功能请求，不是已确认的仓库 bug；不同 x86 平台 cpufreq 能力差异很大，当前没有足够通用的实现范围，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf