coolsnowwolf/lede #7647: AX6的lan接口改wan口不能正常通信

kob, 2021-08-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：将lan绑定的物理接口从默认的eth1、eth2、eth3改为eth1、eth2，准备将eth3留作他用
      新建一个接口命名为gm，绑定物理接口eth3
      将gm接口的ip设置为192.168.1.99 掩码255.255.255.0  网关为192.168.1.1（光猫ip）
       设置完成后ssh进op，ping 192.168.1.1  发现不通

### (2) 路由器型号和固件版本

A：  路由器型号AX6  固件版本OpenWrt R21.8.6 / LuCI Master (git-21.211.30217-03784c3)

### (3) 详细日志

A：
`6: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 9c:9d:7e:86:82:91 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.99/24 brd 192.168.1.255 scope global eth3
       valid_lft forever preferred_lft forever
    inet6 fe80::9e9d:7eff:fe86:8291/64 scope link
       valid_lft forever preferred_lft forever
`
`[12587.170593] nss-dp 3a001800.dp5 eth3: PHY Link is down
[12590.136676] nss-dp 3a001400.dp3 eth1: PHY Link up speed: 1000
[12590.136763] br-lan: port 5(eth1) entered blocking state
[12590.141566] br-lan: port 5(eth1) entered forwarding state
[12590.260673] nss-dp 3a001800.dp5 eth3: PHY Link up speed: 1000
[12592.306347] nss-dp 3a001800.dp5 eth3: PHY Link is down
[12595.412731] nss-dp 3a001800.dp5 eth3: PHY Link up speed: 1000
[12646.426075] nss-dp 3a001800.dp5 eth3: PHY Link is down
[12649.496803] nss-dp 3a001800.dp5 eth3: PHY Link up speed: 1000
[12649.496882] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
[12691.988991] nss-dp 3a001800.dp5 eth3: PHY Link is down
[12695.060962] nss-dp 3a001800.dp5 eth3: PHY Link up speed: 1000
[12695.061042] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
`
`root@OpenWrt:~# ip route
default via 119.96.24.1 dev pppoe-wan proto static
default via 192.168.1.1 dev eth3 proto static metric 99
60.166.169.63 via 119.96.24.1 dev pppoe-wan proto static
119.96.24.1 dev pppoe-wan proto kernel scope link src 119.96.25.152
172.17.17.0/24 dev wg0 proto static scope link
192.168.1.0/24 dev eth3 proto static scope link metric 99
192.168.6.0/24 dev br-lan proto kernel scope link src 192.168.6.1
192.168.11.0/24 dev wg0 proto static scope link
192.168.14.0/24 dev wg0 proto static scope link
root@OpenWrt:~# ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1): 56 data bytes
`

hahaher123, 2021-08-11:
网段都冲突了，这不是固件bug，自己设置的问题。

kob, 2021-08-11:
难道AX6的四个网口是独立网口么

VergilGao, 2021-08-13:
> 难道AX6的四个网口是独立网口么

https://openwrt.org/zh/docs/guide-user/network/vlan/switch_configuration

通过这个页面的方法确认是4个独立网口还是用内置交换机分割出来的网口。

L-vivi, 2021-08-16:
> 难道AX6的四个网口是独立网口么

很确定的告诉你，是4个独立的网口

kob, 2021-08-25:
> > 难道AX6的四个网口是独立网口么
> 
> 很确定的告诉你，是4个独立的网口

那就奇怪了，为啥从lan分离出一个接口接光猫就不能通信了

kob, 2021-08-31:
> > 难道AX6的四个网口是独立网口么
> 
> 很确定的告诉你，是4个独立的网口
通过ip link命令查询，eth1 eth2 eth3  三个接口的mac地址一模一样，应该可以确定，这三个应该不是独立接口
那么我想分离一个出来的话，是不是就应该做vlan然后不打标记呢

coolsnowwolf, 2026-09-06:
AX6 配置将独立物理口接入另一网段，但日志显示 eth3 PHY 反复掉线，且路由表和网关配置不完整；无法确认是当前 DSA/NSS 驱动问题，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf