coolsnowwolf/lede #7648: 打算使用R2S配一个USB无线网卡做无线路由器

joesdu, 2021-08-10:
求推荐一些免驱或者能够驱动的无线网卡.

AmadeusGhost, 2021-08-10:
可以试试 mt7612u 芯片的网卡，mt76 驱动就支持

ricksuzade-maker, 2021-08-11:
华硕 AC55 硕果仅存的软路由无线网卡

joesdu, 2021-08-11:
> 华硕 AC55 硕果仅存的软路由无线网卡

感谢

joesdu, 2021-08-12:
想咨询下RTL8812AU的芯片能发射WIFI吗?

EnnawYang, 2021-08-12:
> 想咨询下RTL8812AU的芯片能发射WIFI吗?

可以，但8812这边自带的驱动不行，要用隔壁的，还是不建议用rtl

joesdu, 2021-08-12:
> > 想咨询下RTL8812AU的芯片能发射WIFI吗?
> 
> 可以，但8812这边自带的驱动不行，要用隔壁的，还是不建议用rtl

能给个教程看看吗?要是实在是不好弄我还是用路由器加R2S做WIFI吧😂(❁´◡`❁)想着若是只用R2S加一个无线网卡能方便很多和小很多.

EnnawYang, 2021-08-12:
> 
> 能给个教程看看吗?要是实在是不好弄我还是用路由器加R2S做WIFI吧😂(❁´◡`❁)想着若是只用R2S加一个无线网卡能方便很多和小很多.

驱动：  
`svn co https://github.com/immortalwrt/immortalwrt/branches/master/package/kernel/rtl8812au-ac package/rtl8812au-ac`  
配置文件加一行，或者从 `make menuconfig` 里面选中：  
`CONFIG_PACKAGE_kmod-rtl8812au-ac=y`    
我再提醒下，还是不建议使用rtl的usb网卡

joesdu, 2021-08-12:
> > 能给个教程看看吗?要是实在是不好弄我还是用路由器加R2S做WIFI吧😂(❁´◡`❁)想着若是只用R2S加一个无线网卡能方便很多和小很多.
> 
> 驱动：
> `svn co https://github.com/immortalwrt/immortalwrt/branches/master/package/kernel/rtl8812au-ac package/rtl8812au-ac`
> 配置文件加一行，或者从 `make menuconfig` 里面选中：
> `CONFIG_PACKAGE_kmod-rtl8812au-ac=y`
> 我再提醒下，还是不建议使用rtl的usb网卡
> 如果和树莓派搭配使用的话要关闭自带的wifi，不然会有点冲突

好的,非常感谢😘💕

closed 2021-08-12 by joesdu