coolsnowwolf/lede #7650: HC5962最新编译WAN DHCP无法获得地址

rshi0212, 2021-08-11:
手上有个8月1日编译的固件，升级最新编译的，始终WAN DHCP就拿不到IP了，降级到8月1日的就没问题，下面是升级后的log。

Wed Aug 11 08:17:31 2021 kern.notice kernel: [    0.000000] Linux version 5.4.138 (rshi0212@Latitude-E6540) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3629-0f59ed510)) #0 SMP Tue Aug 10 12:20:42 2021
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] MIPS: machine is HiWiFi HC5962
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Zone ranges:
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000]   HighMem  empty
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Early memory node ranges
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Wed Aug 11 08:17:31 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=0000010a
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=0000010a
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] Memory: 250088K/262144K available (6171K kernel code, 250K rwdata, 1296K rodata, 1292K init, 238K bss, 12056K reserved, 0K cma-reserved, 0K highmem)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Wed Aug 11 08:17:31 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.007802] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.073973] pid_max: default: 32768 minimum: 301
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.078767] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.085967] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.096416] rcu: Hierarchical SRCU implementation.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.101752] smp: Bringing up secondary CPUs ...
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.585266] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.585276] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.585288] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    9.585389] CPU1 revision is: 0001992f (MIPS 1004Kc)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.166536] Synchronize counters for CPU 1: done.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    4.099162] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    4.099171] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    4.099180] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    4.099235] CPU2 revision is: 0001992f (MIPS 1004Kc)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.257269] Synchronize counters for CPU 2: done.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.181471] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.181480] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Aug 11 08:17:31 2021 kern.warn kernel: [    9.181489] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    9.181551] CPU3 revision is: 0001992f (MIPS 1004Kc)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.342452] Synchronize counters for CPU 3: done.
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.372317] smp: Brought up 1 node, 4 CPUs
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.380689] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.390453] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.397467] pinctrl core: initialized pinctrl subsystem
Wed Aug 11 08:17:31 2021 kern.info kernel: [    0.404130] NET: Registered protocol family 16
Wed Aug 11 08:17:31 2021 kern.debug kernel: [    0.419805] FPU Affinity set after 11720 emulations
[... 679 lines cut ...]
[   27.834318] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   27.844002] PPP generic driver version 2.4.2
[   27.850188] PPP MPPE Compression module registered
[   27.856614] NET: Registered protocol family 24
[   27.865012] kmodloader: done loading kernel modules from /etc/modules.d/*
[   29.378069] random: crng init done
[   64.437153] br-lan: port 1(wlan1) entered blocking state
[   64.442715] br-lan: port 1(wlan1) entered disabled state
[   64.449349] device wlan1 entered promiscuous mode
[   65.893338] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   65.900344] br-lan: port 1(wlan1) entered blocking state
[   65.905671] br-lan: port 1(wlan1) entered forwarding state
[   65.915006] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   65.946016] br-lan: port 2(wlan0) entered blocking state
[... cut ...]
[   65.968199] br-lan: port 2(wlan0) entered forwarding state
[   67.184535] br-lan: port 2(wlan0) entered disabled state
[   67.390428] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   67.397259] br-lan: port 2(wlan0) entered blocking state
[   67.402727] br-lan: port 2(wlan0) entered forwarding state
[  117.688825] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  117.696780] NFSD: Using legacy client tracking operations.
[  117.702404] NFSD: starting 10-second grace period (net f00000a0)

rshi0212, 2021-08-11:
这个是降级后的log

Wed Aug 11 08:19:08 2021 kern.info kernel: [    5.369510] CPU3 revision is: 0001992f (MIPS 1004Kc)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.342448] Synchronize counters for CPU 3: done.
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.372314] smp: Brought up 1 node, 4 CPUs
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.380672] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.390438] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.397433] pinctrl core: initialized pinctrl subsystem
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.404212] NET: Registered protocol family 16
Wed Aug 11 08:19:08 2021 kern.debug kernel: [    0.421038] FPU Affinity set after 11720 emulations
Wed Aug 11 08:19:08 2021 kern.warn kernel: [    0.438600] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    0.449494] random: fast init done
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.453202] clocksource: Switched to clocksource GIC
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.459929] NET: Registered protocol family 2
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.464384] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.472147] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.480521] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.488117] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.495210] TCP: Hash tables configured (established 2048 bind 2048)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.501624] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.508102] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.515284] NET: Registered protocol family 1
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.519577] PCI: CLS 0 bytes, default 32
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.763139] 4 CPUs re-calibrate udelay(lpj = 2924544)
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.769717] workingset: timestamp_bits=14 max_order=16 bucket_order=2
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.788087] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.793860] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.807253] mt7621_gpio 1e000600.gpio: registering 32 gpios
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.813019] mt7621_gpio 1e000600.gpio: registering 32 gpios
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.818871] mt7621_gpio 1e000600.gpio: registering 32 gpios
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.825352] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.833052] printk: console [ttyS0] disabled
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.837356] 1e000c00.uartlite: ttyS0 at MMIO 0x1e000c00 (irq = 19, base_baud = 3125000) is a 16550A
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.846337] printk: console [ttyS0] enabled
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.854600] printk: bootconsole [early0] disabled
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.866293] mt7621-nand 1e003000.nand: Using programmed access timing: 31c07388
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.873724] nand: device found, Manufacturer ID: 0xc8, Chip ID: 0xd1
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.880046] nand: ESMT NAND 128MiB 3,3V 8-bit
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.884410] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.891948] mt7621-nand 1e003000.nand: ECC strength adjusted to 4 bits
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.898494] mt7621-nand 1e003000.nand: Using programmed access timing: 31c07388
Wed Aug 11 08:19:08 2021 kern.info kernel: [    0.905789] Scanning device for bad blocks
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.882034] 7 fixed-partitions partitions found on MTD device mt7621-nand
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.888815] Creating 7 MTD partitions on "mt7621-nand":
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.894044] 0x000000000000-0x000000080000 : "u-boot"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.900356] 0x000000080000-0x000000100000 : "debug"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.906550] 0x000000100000-0x000000140000 : "factory"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.912878] 0x000000140000-0x000000540000 : "kernel"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.919170] 0x000000540000-0x0000021c0000 : "ubiconcat0"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.925962] 0x0000021c0000-0x000002240000 : "bdinfo"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.932196] 0x000002240000-0x000007f80000 : "ubiconcat1"
Wed Aug 11 08:19:08 2021 kern.notice kernel: [    2.939833] Concatenating MTD devices:
[... cut ...]
[   63.319618] br-lan: port 4(wlan1) entered blocking state
[   63.325060] br-lan: port 4(wlan1) entered forwarding state
[   63.481518] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   63.488473] br-lan: port 5(wlan0) entered blocking state
[   63.493865] br-lan: port 5(wlan0) entered forwarding state
[  106.743747] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  106.751605] NFSD: Using legacy client tracking operations.
[  106.757288] NFSD: starting 10-second grace period (net f00000a0)

Hey-Leo, 2021-08-11:
好像接口并不能取到正确的MAC

xiangfeigao, 2021-08-11:
这几天确实这样

rshi0212, 2021-08-11:
还好手上保留了一个8月1日编译的，5台HC5962刷了都是这个问题，差点死翘翘。

rshi0212, 2021-08-11:
MTK方案老是老，但是性能够用啊，200兆~500兆接入宽带，WiFi 5G 866m，毕竟没有那么多老设备支持WiFi6，带机固定+移动20~40台设备，科学上网一点问题也没有。
虽然大佬这边好像重心都在ipq路由上了，ipq路由价格不贵，但是扛不住MTK老设备多啊，全换也是不小成本。
谢谢，大佬不要放弃MTK！

AmadeusGhost, 2021-08-11:
大更新不要保留配置升级，建议进 breed 重刷固件。

rshi0212, 2021-08-11:
> 大更新不要保留配置升级，建议进 breed 重刷固件。

不管是保留配置，还是重置配置，还是进breed刷factory，都没用。

tcsr200722, 2021-08-12:
进breed看看你的MAC有没有改变

rshi0212, 2021-08-12:
保留配置降级固件，直接就恢复正常。

你的意思是MAC可能因为升级固件而改变？

我晚点用newifi d2试验一下。

tcsr200722, 2021-08-13:
我的b70就会

发自我的iPhone

 
 
 
 &gt;
 &gt; 进breed看看你的MAC有没有改变
 &gt;
 &gt; —
 &gt; <https://github.com/coolsnowwolf/lede/issues/7650#issuecomment-897243057&gt;, 
 &gt;

Hey-Leo, 2021-08-15:
现在可以了吗，我想重新编译了

rshi0212, 2021-08-15:
暂时还没有哦～

发自我的iPhone

closed 2021-08-25 by coolsnowwolf