coolsnowwolf/lede #7651: 科学上网故障转移困惑

lexin8, 2021-08-11:
热门的SSR、passwall、vssr、openclash，原理都差不多，都是在分流功能里做优化。但是中转节点故障转移功能就不好使，tcpping到节点是通的，但是无法上网的。中转节点插件只是检测A到B ping通就行，B到C就检测不了。
大部分人基本都是用这种一两百包年的中转节点吧！各位是怎么解决的？

这类廉价的节点经常需要更新，更新后节点顺序变化的话主服务器节点就会变。
个人觉得如果能像PC端v2rayN那样检测真实连接速度然后自动切换就好了，这样既能检测到节点真实故障问题也检测了真实连接速率问题了。不管是节点更新或者是中转节点故障都能友好的切换最快的节点

PS：有没有一种曲线救国想法，在docker安装个专门检测节点的，定时检测更新，当docker里连接不上GOOGLE就自动检测节点，然后生成订阅节点触发SSR、passwall、vssr、openclash更新节点订阅

`带日志的 issues
带日志的 issues
带日志的 issues
Wed Aug 11 09:05:26 2021 daemon.warn `

VergilGao, 2021-08-11:
一两百包年还廉价么……我现在用的20块一个月的，openclash自动选择最优节点，我在订阅时把高倍率的屏蔽了，仍然很稳，如果你的机场经常出现不通的情况，最好还是赶紧换一家为妙……

suizhenmin, 2021-08-11:
真有需求，自建一个才是王道

WYC-2020, 2021-08-11:
都不用故障转移，直接搞负载均衡，还不信全部节点都挂哈哈

kiddin9, 2021-08-11:
passwall并不是用的tcpping,而是使用要检测的节点来开启socks5,通过走socks5代理来访问google判断节点是否通,所以并不存在你说的那个问题.

https://github.com/xiaorouji/openwrt-passwall/blob/15f4829ccf4c838f55ecaf0f2c08167660c9c567/luci-app-passwall/root/usr/share/passwall/test.sh#L63

lexin8, 2021-08-12:
> passwall并不是用的tcpping,而是使用要检测的节点来开启socks5,通过走socks5代理来访问google判断节点是否通,所以并不存在你说的那个问题.
> 
> 具体实现代码:
> https://github.com/xiaorouji/openwrt-passwall/blob/main/luci-app-passwall/root/usr/share/passwall/test.sh#L63

就像楼上说的passwall直接搞负载均衡，发现跟mwan3有冲突，很久没有用passwall了，SSR好像也可以自动切换，但是检测到主节点能ping通就切换回来了，这个问题之前有人反馈过，一直没有改进。你仓库的openwrt-bypass我也测试过了，也去反馈都没有得道回复！

kiddin9, 2021-08-12:
不要用passwall的haproxy负载均衡,用他自带的故障转移,bypass的故障转移也是和passwall用的同一原理,不存在你说的主节点能ping通就切换回来了,除非主节点能访问google才会切换.你可以再多测试下.

WYC-2020, 2021-08-13:
故障转移和负载均衡同时用不舒服吗？我只用负载均衡都稳定从来没断过，我是不知道你们咋用的哈哈

Micah365, 2021-08-17:
> 一两百包年还廉价么……我现在用的20块一个月的，openclash自动选择最优节点，我在订阅时把高倍率的屏蔽了，仍然很稳，如果你的机场经常出现不通的情况，最好还是赶紧换一家为妙……

求链接～

suizhenmin, 2021-08-17:
看来我是用得奢侈了，自建一个月100左右，原生ip，看网飞什么的，独享爽。负载均衡和故障转移一直就没设置过，以前用过一个机场，高峰期非常不稳定。弃了。

xwcoco, 2021-08-18:
> 看来我是用得奢侈了，自建一个月100左右，原生ip，看网飞什么的，独享爽。负载均衡和故障转移一直就没设置过，以前用过一个机场，高峰期非常不稳定。弃了。

求推荐

WYC-2020, 2021-08-19:
@suizhenmin 100块我买的别人的 一年30条线路，懒得自己搭建，一个月1T 完全够

Micah365, 2021-08-20:
> @suizhenmin 100块我买的别人的 一年30条线路，懒得自己搭建，一个月1T 完全够

求推荐，想换机场

suizhenmin, 2021-08-20:
> > @suizhenmin 100块我买的别人的 一年30条线路，懒得自己搭建，一个月1T 完全够
> 
> 求推荐，想换机场

买的是vps，需要自己部署。https://www.dmit.io/aff.php?aff=1139 14.9刀一个月，支付宝可以支付

madman43, 2021-09-09:
> 一两百包年还廉价么……我现在用的20块一个月的，openclash自动选择最优节点，我在订阅时把高倍率的屏蔽了，仍然很稳，如果你的机场经常出现不通的情况，最好还是赶紧换一家为妙……

你用openclash遇到过网页证书问题吗？我用小猫咪电脑打开youtube就提示证书问题，手机一点问题都没有。困扰很久了

coolsnowwolf, 2026-09-06:
这是代理节点健康检查和自动故障转移的功能/架构建议，不是有明确复现步骤的 LEDE 包缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf