coolsnowwolf/lede #7653: RAC2V1K (RT4230W-D187A Rev:9)  2.4G无线不正常

xraylf, 2021-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷新最新固件后，开始时2.4g和5g无线网络使用正常，但经过一段时间后，2.4g无线就无法搜索，只剩5g无线网络可以连接，请问这是无线驱动的原因，还是硬件出问题了？附上日志

Wed Aug 11 10:46:42 2021 kern.warn kernel: [   92.849856] WARNING: CPU: 1 PID: 59 at backports-5.10.42-1/net/mac80211/driver-ops.c:36 ieee80211_add_virtual_monitor+0x974/0xae8 [mac80211]
Wed Aug 11 10:46:42 2021 kern.warn kernel: [   92.854299] Modules linked in: ecm xt_FULLCONENAT iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT pl2303 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211 ipt_REJECT ftdi_sio ch341 cfg80211 xt_time xt_tcpudp xt_socket xt_quota xt_policy xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_esp xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG usbserial ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr ppp_mppe ppp_async nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan iptable_raw iptable_mangle
Wed Aug 11 10:46:42 2021 kern.warn kernel: [   92.854467]  iptable_filter ipt_ah ip_tables crc_ccitt compat asn1_decoder qca_nss_pppoe pppoe pppox ppp_generic slhc ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding qca_nss_drv ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 qca_nss_gmac tunnel6 tunnel4 tun xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 ghash_generic gf128mul gcm echainiv ecb des_generic libdes ctr cmac ccm cbc authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci
Wed Aug 11 10:46:42 2021 kern.warn kernel: [   92.932112]  xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_ipq806x_usb ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat(C) antfs crc32c_generic
Wed Aug 11 10:46:42 2021 kern.warn kernel: [   93.042626] CPU: 1 PID: 59 Comm: kworker/1:1 Tainted: G        WC        5.4.138 #0
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.064794] Hardware name: Generic DT based system
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.072374] Workqueue: events_freezable ieee80211_restart_work [mac80211]
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.077135] [<c030f954>] (unwind_backtrace) from [<c030b968>] (show_stack+0x14/0x20)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.083990] [<c030b968>] (show_stack) from [<c08f4b38>] (dump_stack+0x94/0xa8)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.091807] [<c08f4b38>] (dump_stack) from [<c031e85c>] (__warn+0xb4/0xd0)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.098829] [<c031e85c>] (__warn) from [<c031e8c8>] (warn_slowpath_fmt+0x50/0x90)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.105746] [<c031e8c8>] (warn_slowpath_fmt) from [<bf6d4584>] (ieee80211_add_virtual_monitor+0x974/0xae8 [mac80211])
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.113347] [<bf6d4584>] (ieee80211_add_virtual_monitor [mac80211]) from [<bf6d470c>] (ieee80211_stop+0x14/0x24 [mac80211])
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.123900] [<bf6d470c>] (ieee80211_stop [mac80211]) from [<c0782e88>] (__dev_close_many+0x98/0x108)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.134782] [<c0782e88>] (__dev_close_many) from [<c0782f70>] (dev_close_many+0x78/0x108)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.144147] [<c0782f70>] (dev_close_many) from [<c0785574>] (dev_close.part.17+0x2c/0x40)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.152285] [<c0785574>] (dev_close.part.17) from [<bf6560e8>] (cfg80211_shutdown_all_interfaces+0x7c/0xf4 [cfg80211])
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.160490] [<bf6560e8>] (cfg80211_shutdown_all_interfaces [cfg80211]) from [<bf6f82f8>] (ieee80211_reconfig+0x328/0xce4 [mac80211])
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.171086] [<bf6f82f8>] (ieee80211_reconfig [mac80211]) from [<bf6bb878>] (ieee80211_restart_work+0xb0/0xf8 [mac80211])
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.183104] [<bf6bb878>] (ieee80211_restart_work [mac80211]) from [<c0337970>] (process_one_work+0x234/0x4a0)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.193885] [<c0337970>] (process_one_work) from [<c0337c30>] (worker_thread+0x54/0x604)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.203695] [<c0337c30>] (worker_thread) from [<c033dbfc>] (kthread+0x160/0x164)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.211851] [<c033dbfc>] (kthread) from [<c03010e8>] (ret_from_fork+0x14/0x2c)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.219223] Exception stack(0xef20bfb0 to 0xef20bff8)
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.226259] bfa0:                                     00000000 00000000 00000000 00000000
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.231390] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.239547] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.247771] ---[ end trace 9eb240e94d91e803 ]---
Wed Aug 11 10:46:43 2021 daemon.notice netifd: Network device 'wlan1' link is down
Wed Aug 11 10:46:43 2021 kern.info kernel: [   93.254574] br-lan: port 2(wlan1) entered disabled state
Wed Aug 11 10:46:43 2021 daemon.err hostapd: Failed to set beacon parameters
Wed Aug 11 10:46:43 2021 kern.warn kernel: [   93.270176] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:46:43 2021 daemon.notice hostapd: wlan1: INTERFACE-DISABLED
Wed Aug 11 10:46:43 2021 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Wed Aug 11 10:46:44 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan6 (eth0.2)
Wed Aug 11 10:46:49 2021 kern.warn kernel: [   99.270848] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:46:55 2021 kern.warn kernel: [  105.271389] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:46:55 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Wed Aug 11 10:46:56 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (eth0.2)
Wed Aug 11 10:47:01 2021 kern.warn kernel: [  111.271924] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:47:07 2021 kern.warn kernel: [  117.272458] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:47:13 2021 kern.warn kernel: [  123.273304] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:47:19 2021 kern.warn kernel: [  129.273960] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:47:25 2021 kern.warn kernel: [  135.274674] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108
Wed Aug 11 10:47:31 2021 kern.warn kernel: [  141.276223] ath10k_pci 0001:01:00.0: failed to send pdev bss chan info request: -108

### (2) 路由器型号和固件版本

A：RT4230W-D187A Rev:9/ OpenWrt R21.8.6 / LuCI Master (git-21.211.30217-03784c3)--r3634-e80af1955

### (3) 详细日志

A：
[Askey RT4230W REV6 sys.txt](https://github.com/coolsnowwolf/lede/files/6965885/Askey.RT4230W.REV6.sys.txt)

asushugo, 2021-08-11:
不是固件有问题，是硬件问题

lanfenglin, 2021-08-15:
把网络加速关掉也许就正常了

xraylf, 2021-08-24:
> 把网络加速关掉也许就正常了

  请问把NSS去掉的话，要取消哪些选项?

closed 2024-12-22 by xraylf