coolsnowwolf/lede #7657: rt-acrh17固件ssr plus插件异常

yubanmeiqin9048, 2021-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：源码为8.11（号），一切编译操作皆按照仓库主页操作，编译固件时仅在默认配置的基础上删除和勾选了一些插件，linux系统为Ubuntu 20.04，编译后的固件ssrp插件功能在电脑上一切正常，但是在手机上不正常，表现为除了google以外的外网都可以使用，但纬度google不行

### (2) 路由器型号和固件版本

A：RT-ACRH17

### (3) 详细日志

A：

281677160, 2021-08-11:
这个问题一般应该都是DNS设置错误导致的吧，我以前试过，不过我忘记了是咋弄好了，反正就DNS问题

yubanmeiqin9048, 2021-08-11:
> 这个问题一般应该都是DNS设置错误导致的吧，我以前试过，不过我忘记了是咋弄好了，反正就DNS问题

我也没动关于DNS方面的设置，基本就是刷好了，然后订阅，就默认的配置启动

281677160, 2021-08-11:
你op的dns设置

yubanmeiqin9048, 2021-08-11:
> 你op的dns设置
> […](#)

[image]
[image]

4860575, 2021-08-11:
解决办法1、不使用pdnsd，改为dns2socks
或者：修改/etc/init.d/shadowsocksr 注释掉
			reject=::/0;
 			reject_policy=negate;
 			reject_recursively=on;

281677160, 2021-08-11:
自定义DNS那些你设置114.114.114.114 223.5.5.5这样两个测试下

yubanmeiqin9048, 2021-08-11:
> 解决办法1、不使用pdnsd，改为dns2socks
> 或者：修改/etc/init.d/shadowsocksr 注释掉
> reject=::/0;
> reject_policy=negate;
> reject_recursively=on;

通过法一解决了，请问这是什么问题呢？

wuiiled, 2021-08-11:
> > 解决办法1、不使用pdnsd，改为dns2socks
> > 或者：修改/etc/init.d/shadowsocksr 注释掉
> > reject=::/0;
> > reject_policy=negate;
> > reject_recursively=on;
> 
> 通过法一解决了，请问这是什么问题呢？我想不出是因为什么问题导致的这种情况，因为我看到群里的（acrh17的群）用pdnsd也没问题，基本也是默认配置在用

pdnsd DNS缓存问题

closed 2021-08-11 by yubanmeiqin9048