coolsnowwolf/lede #7660: 8月11日18：00本地编译出错

qqsir-dev, 2021-08-11:
反馈bug/问题模板，提建议请删除
昨晚没事，今天git pull之后就不行了
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
mkdir -p /home/qqsir/lede/dl
checking for stdint.h... SHELL= flock /home/qqsir/lede/tmp/.e2fsprogs-1.45.6.tar.xz.flock -c '          /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "e2fsprogs-1.45.6.tar.                                               xz" "ffa7ae6954395abdc50d0f8605d8be84736465afc53b8938ef473fcf7ff44256" "" "@KERNEL/linux/kernel/people/tytso/e2fsprogs/v1.45.6/"    '
yes
  CC [M]  /home/qqsir/lede/build_dir/target-x86_64_musl/linux-x86_64/r8125-9.005.06/r8125_n.o
mkdir -p /home/qqsir/lede/dl
SHELL= flock /home/qqsir/lede/tmp/.libubox-2021-05-16-b14c4688.tar.xz.flock -c '        /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "libubox-2021-05-16-b14c4688.tar.xz" "                                               7dd1db1e0074a9c7c722db654cce3111b3bd3cff0bfd791c4497cb0f6c22d3ca" "" || (       echo "Checking out files from the git repository..."; mkdir -p /home/qqsir/lede/tmp/dl && cd /home/qqsir/                                               lede/tmp/dl && rm -rf libubox-2021-05-16-b14c4688 && [ \! -d libubox-2021-05-16-b14c4688 ] && git clone  https://git.openwrt.org/project/libubox.git libubox-2021-05-16-b14c4688 && (cd l                                               ibubox-2021-05-16-b14c4688 && git checkout b14c4688612c05c78ce984d7bde633bce8703b1e && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd                                                libubox-2021-05-16-b14c4688 && git log -1 --format='\''@%ct'\''` && rm -rf libubox-2021-05-16-b14c4688/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${                                               TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libubox-2021-05-16-b14c4688 |       xz -zc -7e > /home/qqsir/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz && mv /home/qqsir/lede/tmp/dl/lib                                               ubox-2021-05-16-b14c4688.tar.xz /home/qqsir/lede/dl/ && rm -rf libubox-2021-05-16-b14c4688;  )    '
checking for unistd.h... yes
checking minix/config.h usability... make[3]: Leaving directory '/home/qqsir/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.18#0.03#0.34
make[3]: Leaving directory '/home/qqsir/lede/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/host-compile#0.21#0.08#0.39
no
checking minix/config.h presence... make[3]: Entering directory '/home/qqsir/lede/package/libs/libnl-tiny'
make[3]: Entering directory '/home/qqsir/lede/package/libs/libjson-c'
no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking for Minix Amsterdam compiler... no
checking for x86_64-openwrt-linux-ar... x86_64-openwrt-linux-musl-gcc-ar
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files...   CC [M]  /home/qqsir/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.049.01/src/r8168_n.o
no
checking how to print strings... printf
checking for a sed that does not truncate output... /home/qqsir/lede/staging_dir/host/bin/sed
checking for fgrep... /home/qqsir/lede/staging_dir/host/bin/grep -F
checking for ld used by x86_64-openwrt-linux-musl-gcc... x86_64-openwrt-linux-musl-ld
checking if the linker (x86_64-openwrt-linux-musl-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... x86_64-openwrt-linux-musl-gcc-nm
checking the name lister (x86_64-openwrt-linux-musl-gcc-nm) interface... mkdir -p /home/qqsir/lede/dl
SHELL= flock /home/qqsir/lede/tmp/.libnl-tiny-2020-08-05-c291088f.tar.xz.flock -c '     /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "libnl-tiny-2020-08-05-c291088f.tar.xz                                               " "99bcce12701bb34dadb39689d95c2c5cf1e27719d0ecfd645d3957a8947025ac" "" || (    echo "Checking out files from the git repository..."; mkdir -p /home/qqsir/lede/tmp/dl && cd /home/qqsir/                                               lede/tmp/dl && rm -rf libnl-tiny-2020-08-05-c291088f && [ \! -d libnl-tiny-2020-08-05-c291088f ] && git clone  https://git.openwrt.org/project/libnl-tiny.git libnl-tiny-2020-08-05-c2910                                               88f && (cd libnl-tiny-2020-08-05-c291088f && git checkout c291088f631d1694f7ba0444b59677b194348da8 && git submodule update --init --recursive) && echo "Packing checkout..." && export TA                                               R_TIMESTAMP=`cd libnl-tiny-2020-08-05-c291088f && git log -1 --format='\''@%ct'\''` && rm -rf libnl-tiny-2020-08-05-c291088f/.git &&    tar --numeric-owner --owner=0 --group=0 --mode=a-                                               s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libnl-tiny-2020-08-05-c291088f |    xz -zc -7e > /home/qqsir/lede/tmp/dl/libnl-tiny-2020-08-05-c291088f.tar.xz && mv /home/qq                                               sir/lede/tmp/dl/libnl-tiny-2020-08-05-c291088f.tar.xz /home/qqsir/lede/dl/ && rm -rf libnl-tiny-2020-08-05-c291088f;  )    '
BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
mkdir -p /home/qqsir/lede/dl
SHELL= flock /home/qqsir/lede/tmp/.json-c-0.15-nodoc.tar.gz.flock -c '          /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "json-c-0.15-nodoc.tar.gz" "99bca4f944b8ced8ae                                               0bbc6310d6a3528ca715e69541793a1ef51f8c5b4b0878" "" "https://s3.amazonaws.com/json-c_releases/releases/"    '
checking how to convert x86_64-pc-linux-gnu file names to x86_64-openwrt-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for x86_64-openwrt-linux-musl-ld option to reload object files... -r
checking for x86_64-openwrt-linux-objdump... x86_64-openwrt-linux-musl-objdump
checking how to recognize dependent libraries... pass_all
checking for x86_64-openwrt-linux-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for x86_64-openwrt-linux-ar... (cached) x86_64-openwrt-linux-musl-gcc-ar
checking for archiver @FILE support... @
checking for x86_64-openwrt-linux-strip... (cached) x86_64-openwrt-linux-musl-strip
checking for x86_64-openwrt-linux-ranlib... x86_64-openwrt-linux-musl-gcc-ranlib
checking command to parse x86_64-openwrt-linux-musl-gcc-nm output from x86_64-openwrt-linux-musl-gcc object... mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libnl-tiny.version || { echo '1' > /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libnl-tiny.version                                               ;  }
SHELL= flock /home/qqsir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/libnl-tiny-2020-08-05-c291088f/.pkgdir/libnl-tiny/. /home/qqsir/lede/staging                                               _dir/target-x86_64_musl/root-x86/'
touch /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libnl-tiny_installed
echo "libnl-tiny" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libnl-tiny.default.install
make[3]: Leaving directory '/home/qqsir/lede/package/libs/libnl-tiny'
time: package/libs/libnl-tiny/compile#0.14#0.06#0.34
[... 1602 lines cut ...]
cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/libiconv-1.16/ipkg-install/usr/lib/libiconv.so.* /home/qqsir/lede/build_dir/target-x86_64_musl/libiconv-1.16/.pkgdir/libiconv-full/                                               usr/lib/
touch /home/qqsir/lede/build_dir/target-x86_64_musl/libiconv-1.16/.pkgdir/libiconv-full.installed
mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '2' | cmp -s - /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libiconv-full.version || { echo '2' > /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libiconv-full.v                                               ersion;  }
[... cut ...]
echo "libiconv-full" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/libiconv-full.default.install
make[3]: Leaving directory '/home/qqsir/lede/package/libs/libiconv-full'
time: package/libs/libiconv-full/compile#13.38#2.39#17.07
make[2]: Leaving directory '/home/qqsir/lede'
make[1]: *** [package/Makefile:110: /home/qqsir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qqsir/lede'
make: *** [/home/qqsir/lede/include/toplevel.mk:230: world] Error 2

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WYC-2020, 2021-08-11:
我刚还编译通过了

zzsj0928, 2021-08-11:
错误在这，

五点多已经被修复了，

你重新跑一下

```
SHELL= flock /home/qqsir/lede/tmp/.xray-core-1.4.2.tar.gz.flock -c ' /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "xray-core-1.4.2.tar.gz" "565255d8c67b254f403d498b9152 fa7bc097d649c50cb318d278c2be644e92cc" "" "https://codeload.github.com/XTLS/xray-core/tar.gz/v1.4.2?" '
mkdir -p /home/qqsir/lede/dl
SHELL= flock /home/qqsir/lede/tmp/.xray-plugin-1.4.2.tar.gz.flock -c ' /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "xray-plugin-1.4.2.tar.gz" "4a178a2bacffcc2fd3 74c57e47b71eb0cb5667bfe747690a16501092c0618707" "" "https://codeload.github.com/teddysun/xray-plugin/tar.gz/v1.4.2?" '
Hash of the local file xray-plugin-1.4.2.tar.gz does not match (file: ae7ff14a34ca7e226969ebeae20317adb2514ab0d1d78b18e2ee6fd731eb0e89, requested: 4a178a2bacffcc2fd374c57e47b71eb0cb5667 bfe747690a16501092c0618707) - deleting download.
make[3]: *** [Makefile:85: /home/qqsir/lede/dl/xray-plugin-1.4.2.tar.gz] Error 2
make[3]: Leaving directory '/home/qqsir/lede/feeds/helloworld/xray-plugin'
time: package/feeds/helloworld/xray-plugin/compile#0.13#0.02#0.27
ERROR: package/feeds/helloworld/xray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/xray-plugin/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
```

qqsir-dev, 2021-08-11:
跑不过去，还是有错误
SHELL= flock /home/qqsir/lede/tmp/.xray-core-1.4.2.tar.gz.flock -c ' /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "xray-core-1.4.2.tar.gz" "565255d8c67b254f403d498b9152 fa7bc097d649c50cb318d278c2be644e92cc" "" "https://codeload.github.com/XTLS/xray-core/tar.gz/v1.4.2?" '
mkdir -p /home/qqsir/lede/dl
SHELL= flock /home/qqsir/lede/tmp/.xray-plugin-1.4.2.tar.gz.flock -c ' /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "xray-plugin-1.4.2.tar.gz" "4a178a2bacffcc2fd3 74c57e47b71eb0cb5667bfe747690a16501092c0618707" "" "https://codeload.github.com/teddysun/xray-plugin/tar.gz/v1.4.2?" '
Hash of the local file xray-plugin-1.4.2.tar.gz does not match (file: ae7ff14a34ca7e226969ebeae20317adb2514ab0d1d78b18e2ee6fd731eb0e89, requested: 4a178a2bacffcc2fd374c57e47b71eb0cb5667 bfe747690a16501092c0618707) - deleting download.
make[3]: *** [Makefile:85: /home/qqsir/lede/dl/xray-plugin-1.4.2.tar.gz] Error 2
make[3]: Leaving directory '/home/qqsir/lede/feeds/helloworld/xray-plugin'
time: package/feeds/helloworld/xray-plugin/compile#0.13#0.02#0.27
ERROR: package/feeds/helloworld/xray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/xray-plugin/compile] Error 1
make[2]: *** Waiting for unfinished jobs....

qqsir-dev, 2021-08-12:
不知道为什么，git pull 和更新feeds都没有效果，最后全部删除重新来过就没问题了

closed 2021-08-12 by qqsir-dev