coolsnowwolf/lede #7661: RaspberryPI4B安装编译后的OpenWrt WIFI一直无法开启

joesdu, 2021-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

使用master分支最新代码编译树莓派4B 8G版的固件,刷上后无法开启WiFi,虽然页面显示开启了,但是无法搜索到信号,Windows11系统,WIFI6网卡,包括IPhone12Pro也无法搜索到.2.4G网卡也无法搜索到信号.

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：RaspberryPI4B安装编译后的OpenWrt WIFI一直无法开启

### (2) 路由器型号和固件版本

A：使用master分支最新代码编译

### (3) 详细日志

A：

EnnawYang, 2021-08-12:
因为cypress的驱动限制，请避免使用DFS信道，使用以下信道，设置完最好重启下：  
```
Channel 01 : 2.412 GHz
Channel 02 : 2.417 GHz
Channel 03 : 2.422 GHz
Channel 04 : 2.427 GHz
Channel 05 : 2.432 GHz
Channel 06 : 2.437 GHz
Channel 07 : 2.442 GHz
Channel 08 : 2.447 GHz
Channel 09 : 2.452 GHz
Channel 10 : 2.457 GHz
Channel 11 : 2.462 GHz
Channel 12 : 2.467 GHz
Channel 13 : 2.472 GHz
Channel 36 : 5.18 GHz
Channel 40 : 5.2 GHz
Channel 44 : 5.22 GHz
Channel 48 : 5.24 GHz
Channel 149 : 5.745 GHz
Channel 153 : 5.765 GHz
Channel 157 : 5.785 GHz
Channel 161 : 5.805 GHz
Channel 165 : 5.825 GHz
```

joesdu, 2021-08-12:
> 因为cypress的驱动限制，请避免使用DFS信道，使用以下信道，设置完最好重启下：
>
> ```
> Channel 01 : 2.412 GHz
> Channel 02 : 2.417 GHz
> Channel 03 : 2.422 GHz
> Channel 04 : 2.427 GHz
> Channel 05 : 2.432 GHz
> Channel 06 : 2.437 GHz
> Channel 07 : [...]

joesdu, 2021-08-12:
> 因为cypress的驱动限制，请避免使用DFS信道，使用以下信道，设置完最好重启下：
>
> ```
> Channel 01 : 2.412 GHz
> Channel 02 : 2.417 GHz
> Channel 03 : 2.422 GHz
> Channel 04 : 2.427 GHz
> Channel 05 : 2.432 GHz
> Channel 06 : 2.437 GHz
> Channel 07 : [...]

测试了,改信道没有效果,而且默认的就是36信道.

EnnawYang, 2021-08-12:
> 
> 测试了,改信道没有效果,而且默认的就是36信道.

emmm，摸出了我正在吃灰的π试了下，好像是有问题，可以先切到N模式用用

joesdu, 2021-08-12:
> > 测试了,改信道没有效果,而且默认的就是36信道.
> 
> emmm，摸出了我正在吃灰的π试了下，好像是有问题，可以先切到N模式用用

N模式是能搜到了,但是我加了下密码就又搜不到了🤣

joesdu, 2021-08-12:
> 有个3B+，做客户端时对op的无线加密甚至选项都有要求，WPA3不支持，WPA2/WPA3混合也不支持，最高只能WPA2 CCMP 802.11w MFP也不支持，就是op里MFP设成optional都连不上，必须disable，不知道你这个当AP会不会也是类似的原因。

[image]
最终设置成这样才行,国家码一定要改成US,用CN不行.

EnnawYang, 2021-08-12:
[image]

joesdu, 2021-08-13:
树莓派搞这个太难了.还是继续用R2S+路由器吧😂

closed 2021-08-13 by joesdu