coolsnowwolf/lede #7665: mt7620 看视频 流量大时 就无法上网

Ho-Lay, 2021-08-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：看视频 流量大时 就无法上网 断流  然后一会儿又自动连上，有时候断网了不能自动连上 必须拔电重启才行，查看日志，都是出现 “WARNING: CPU: 0 PID: 0” 之后就会出现这种情况，没出现这条日志时 看视频是没问题的，不确定是看视频流量大导致内核报错不能上网   还是内核报错导致视频流量大不能上网，5.4和5.10内核都编译过  都会出现同样问题  5.4内核试过编译5.4.132和138     5.10内核试过编译5.10.46  50  51  56
  我编译的插件有 ssrplus xray内核  SQM  Flow Offload  samba  hd-idle  zerotier  upnp  autoreboot  ramfree  webadmin
我试过把 ssrplus  SQM    Flow Offload  zerotier   upnp逐一关闭排查  都一样出现上面我所说的一模一样的状况
### (2) 路由器型号和固件版本

A：R21.7  R21.8 的固件都是这样  小米路由mini的

### (3) 详细日志

A：Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.715155] ------------[ cut here ]------------
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.719955] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:468 dev_watchdog+0x26c/0x274
Thu Aug 12 21:07:53 2021 kern.info kernel: [  559.728416] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.735791] Modules linked in: xt_FULLCONENAT xt_connlimit rt2800soc rt2800mmio rt2800lib pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.736260]  slhc sch_cake nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat asn1_decoder fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb tun vfat fat nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.825907]  drbg hmac ecb cmac authenc arc4 uas usb_storage leds_gpio uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common crc16 crc32c_generic
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.938462] CPU: 0 PID: 0 Comm: swapper Not tainted 5.10.56 #0
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.944409] Stack : 806e0000 805eda00 00000000 00000000 80c09e0c 80820000 806601c0 8065fd87
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.952983]         805c512c 00000000 808233c4 ffffffff 806e0000 00000001 80c09dc0 655877d1
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.961559]         00000000 00000000 805c512c 80c09c60 ffffefff 806cb01c 00000000 00000000
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.970135]         00000000 0000001b 00000000 000e51de 00000000 80c09ea0 00000009 803ecd64
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.978711]         00000009 ffffffff 806e0000 00000000 00000018 80313040 00000000 80820000
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.987285]         ...
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.989789] Call Trace:
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.992304] [<80008474>] show_stack+0x30/0x100
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  559.996869] [<80026130>] __warn+0xc0/0xe8
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.000966] [<800261e4>] warn_slowpath_fmt+0x8c/0xac
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.006065] [<803ecd64>] dev_watchdog+0x26c/0x274
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.010880] [<8006ead4>] call_timer_fn.isra.33+0x20/0x90
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.016318] [<8006ee14>] run_timer_softirq+0x2d0/0x340
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.021570] [<80532c90>] __do_softirq+0x110/0x298
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.026387] [<80002fb0>] except_vec_vi_end+0xb8/0xc4
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.031478] [<80532a34>] r4k_wait_irqoff+0x18/0x24
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.036399] [<8004e7f4>] do_idle+0xd0/0x104
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.040672] [<8004ea34>] cpu_startup_entry+0x18/0x28
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.045759] [<806f0d30>] start_kernel+0x584/0x5ac
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.050556]
Thu Aug 12 21:07:53 2021 kern.warn kernel: [  560.052075] ---[ end trace 6b0e11fd9a66bc3d ]---
Thu Aug 12 21:07:53 2021 kern.err kernel: [  560.056813] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Aug 12 21:07:53 2021 kern.info kernel: [  560.063391] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Aug 12 21:07:53 2021 kern.info kernel: [  560.069814] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=576, dtx=576, fdx=539, next=576
Thu Aug 12 21:07:53 2021 kern.info kernel: [  560.081110] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0290c000, max=1024, calc=224, drx=226
Thu Aug 12 21:07:58 2021 kern.err kernel: [  562.294984] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Aug 12 21:07:58 2021 kern.info kernel: [  562.301590] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Aug 12 21:07:58 2021 kern.info kernel: [  562.308023] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=439, dtx=439, fdx=368, next=439
Thu Aug 12 21:07:58 2021 kern.info kernel: [  562.319322] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=029c8000, max=1024, calc=375, drx=376
Thu Aug 12 21:08:08 2021 kern.err kernel: [  566.230029] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Aug 12 21:08:08 2021 kern.info kernel: [  566.236635] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Aug 12 21:08:08 2021 kern.info kernel: [  566.243067] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=250, dtx=250, fdx=160, next=250
Thu Aug 12 21:08:08 2021 kern.info kernel: [  566.254369] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0290c000, max=1024, calc=424, drx=425
Thu Aug 12 21:08:18 2021 kern.err kernel: [  570.014435] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Aug 12 21:08:18 2021 kern.info kernel: [  570.021057] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Aug 12 21:08:18 2021 kern.info kernel: [  570.027493] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=253, dtx=253, fdx=144, next=253
Thu Aug 12 21:08:18 2021 kern.info kernel: [  570.038792] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=029c8000, max=1024, calc=310, drx=311
Thu Aug 12 21:08:23 2021 kern.err kernel: [  571.940320] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Aug 12 21:08:23 2021 kern.info kernel: [  571.946928] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Aug 12 21:08:23 2021 kern.info kernel: [  571.953364] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=70, dtx=70, fdx=16, next=70
Thu Aug 12 21:08:23 2021 kern.info kernel: [  571.964304] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0290c000, max=1024, calc=131, drx=132
Thu Aug 12 21:08:33 2021 kern.err kernel: [  575.848626] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[... cut ...]
Thu Aug 12 21:08:53 2021 kern.info kernel: [  583.429280] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=02994000, max=1024, ctx=307, dtx=307, fdx=112, next=307
Thu Aug 12 21:08:53 2021 kern.info kernel: [  583.440574] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=029c8000, max=1024, calc=261, drx=262
Thu Aug 12 21:11:28 2021 daemon.err rmmod: module is not loaded
Thu Aug 12 21:11:29 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Aug 12 21:11:29 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Aug 12 21:11:35 2021 daemon.info dnsmasq[6332]: Connected to system UBus

Her0R0cket, 2021-08-13:
这cpu干这活有点吃力

Ho-Lay, 2021-08-13:
> 这cpu干这活有点吃力

不是，这个路由器不是重度使用，只是偶尔有人用而已，而且用的人一般就一两个人，两个人不会同时看视频的，4月份之前编译的固件 相同的插件  都很稳定  就是上个月编译了5.10内核更新就出现这个问题了，因为没保留到前几个月的固件，现在有点烦恼

Neverends4, 2021-08-13:
现在已经又回到5.4内核了，可以再重新编译试试

Her0R0cket, 2021-08-14:
你可以去掉那些不必要的内核模块或者建个新配置试下,基本上就是内核的问题了

AndyLin2020, 2021-08-14:
一样的，之前版本也有这个问题，好了一阵子又出现了
ralink_soc_eth 10100000.ethernet eth0: transmit timed out

AndyLin2020, 2021-08-16:
`Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.113935] ------------[ cut here ]------------
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.118711] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:468 dev_watchdog+0x26c/0x274
Mon Aug 16 15:13:47 2021 kern.info kernel: [  886.127186] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.134565] Modules linked in: xt_FULLCONENAT rt2800soc rt2800mmio rt2800lib pppoe ppp_async rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_bbr slhc sch_cake nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.135011]  iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat asn1_decoder sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel tun ntfs nls_utf8 nls_iso8859_1 nls_cp437 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.224397]  sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg hmac ecb cmac authenc arc4 mmc_block usb_storage mtk_sd mmc_core leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc16 crc32c_generic
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.342100] CPU: 0 PID: 0 Comm: swapper Not tainted 5.10.57 #0
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.348061] Stack : 806d0000 805d9880 00000000 00000000 80c09e0c 80810000 8064c21c 8064bde7
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.356633]         805b1710 00000000 808133c4 ffffffff 806c0000 00000001 80c09dc0 25459022
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.365202]         00000000 00000000 805b1710 80c09c60 ffffefff 806b6d7c 00000000 00000000
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.373760]         00000000 0000002c 00000000 00053854 00000000 80c09ea0 00000009 803d9e44
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.382329]         00000009 ffffffff 806c0000 00000000 00000018 80300940 00000000 80810000
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.390897]         ...
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.393399] Call Trace:
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.395927] [<80008464>] show_stack+0x30/0x100
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.400480] [<80026188>] __warn+0xc0/0xe8
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.404594] [<8002623c>] warn_slowpath_fmt+0x8c/0xac
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.409678] [<803d9e44>] dev_watchdog+0x26c/0x274
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.414503] [<8006c93c>] call_timer_fn.isra.33+0x20/0x90
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.419926] [<8006cc7c>] run_timer_softirq+0x2d0/0x340
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.425189] [<80520a38>] __do_softirq+0x110/0x298
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.429994] [<80002fb0>] except_vec_vi_end+0xb8/0xc4
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.435076] [<805207dc>] r4k_wait_irqoff+0x18/0x24
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.439984] [<8004daec>] do_idle+0xd0/0x104
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.444273] [<8004dd34>] cpu_startup_entry+0x20/0x28
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.449347] [<806dcd28>] start_kernel+0x57c/0x5a4
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.454162]
Mon Aug 16 15:13:47 2021 kern.warn kernel: [  886.455682] ---[ end trace 60af423319cb0c91 ]---
Mon Aug 16 15:13:47 2021 kern.err kernel: [  886.460404] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Mon Aug 16 15:13:47 2021 kern.info kernel: [  886.467002] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Mon Aug 16 15:13:47 2021 kern.info kernel: [  886.473409] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=025fc000, max=1024, ctx=858, dtx=858, fdx=785, next=858
[... cut ...]
Mon Aug 16 15:13:53 2021 kern.info kernel: [  888.820189] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=025fc000, max=1024, ctx=270, dtx=270, fdx=32, next=270
Mon Aug 16 15:13:53 2021 kern.info kernel: [  888.831395] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=025c8000, max=1024, calc=115, drx=119
Mon Aug 16 15:13:53 2021 kern.info kernel: [  888.851569] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
Mon Aug 16 15:13:58 2021 kern.info kernel: [  890.420058] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (1000Mbps/Full duplex)`

AndyLin2020, 2021-08-20:
试过了最新的5.4和5.10内核，都是一样问题，流量一大就挂，今天换了5.4.132内核重新编译就好了，希望之后能解决

固件版本	OpenWrt R21.7.15 / LuCI Master (git-21.211.30217-03784c3)
内核版本	5.4.132

lunatickochiya, 2021-08-23:
呃呃呃，mtk断流，老毛病了…

WYC-2020, 2021-10-10:
怕是性能不够，cpu炸了吧

yulinsoft, 2021-11-07:
7620还是用4.14的内核比较稳定，建议恢复到R21.6.1,这个版本可以编译4.14的内核，我的youku-L1刷新的内核负载高时会重启，用回4.14内核非常稳定。

Ho-Lay, 2021-12-17:
我替换了mt7621 mt7620的驱动，mt7621通过了编译，mt7620正在编译，目前测试中

Ho-Lay, 2022-01-09:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ ]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：看视频 流量大时 就无法上网 断流 然后一会儿又自动连上，有时候断网了不能自动连上 必须拔电重启才行，查看日志，都是出现 “WARNING: CPU: 0 PID: 0” [...]

closed 2022-01-09 by Ho-Lay

Y32Gcnte8z, 2022-03-03:
> 我替换了mt7621 mt7620的驱动，mt7621通过了编译，mt7620正在编译，目前测试中

请问大佬怎么解决的呢？需要怎么替换驱动呢？