coolsnowwolf/lede #7684: 编译多次出现相同错误

CJM88888, 2021-08-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译多次出现同样错误，期间使用过clean命令后无效，重新编译还是现实同样的饿错误，请问应该如何处理，谢谢。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：-- Build files have been written to: /home/cjm/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688
touch /home/cjm/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688/.configured
MAKEFLAGS="" /home/cjm/lede/staging_dir/host/bin/ninja  -j1 -C /home/cjm/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688 
ninja: Entering directory `/home/cjm/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688'
[2/42] Building C object CMakeFiles/jshn.dir/jshn.c.o
FAILED: CMakeFiles/jshn.dir/jshn.c.o 
/home/cjm/lede/staging_dir/host/bin/gcc   -O2 -I/home/cjm/lede/staging_dir/host/include -I/home/cjm/lede/staging_dir/hostpkg/include -I/home/cjm/lede/staging_dir/target-arm_cortex-a9_musl_eabi/host/include -DNDEBUG   -Wall -Werror -Wextra -Werror=implicit-function-declaration -Wformat -Werror=format-security -Werror=format-nonliteral -Os -std=gnu99 -g3 -Wmissing-declarations -Wno-unused-parameter -MD -MT CMakeFiles/jshn.dir/jshn.c.o -MF CMakeFiles/jshn.dir/jshn.c.o.d -o CMakeFiles/jshn.dir/jshn.c.o -c jshn.c
jshn.c:19:18: fatal error: json/json.h: No such file or directory
   19 |         #include <json/json.h>
      |                  ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:116: /home/cjm/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688/.built] Error 1
make[3]: Leaving directory '/home/cjm/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#1.77#0.98#2.99
    ERROR: package/libs/libubox [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/libubox/host/compile] Error 1
make[2]: Leaving directory '/home/cjm/lede'
make[1]: *** [package/Makefile:110: /home/cjm/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cjm/lede'
make: *** [/home/cjm/lede/include/toplevel.mk:230：world] 错误 2

miansan, 2021-08-16:
同样的错误，不知道什么原因。

9l, 2021-08-17:
改成单线程编译看看错在哪，你这日志没给出错误点

jshn.c:19:18: fatal error: json/json.h: No such file or directory

可能是在前面哪里有下载不成功的地方

showme2005, 2021-08-17:
我这里老是出现passwall 的错误..

closed 2022-02-12 by Beginner-Go