coolsnowwolf/lede #7686: kernel 5.10.58编译失败

cuiyf5516, 2021-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kernel 5.10.58编译失败

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
Applying /home/sky/build/lede/target/linux/generic/pending-5.10/683-of_net-add-mac-address-to-of-tree.patch using plaintext: 
patching file drivers/of/of_net.c
Hunk #1 FAILED at 141.
Hunk #2 FAILED at 216.
2 out of 2 hunks FAILED -- saving rejects to file drivers/of/of_net.c.rej
Patch failed!  Please fix /home/sky/build/lede/target/linux/generic/pending-5.10/683-of_net-add-mac-address-to-of-tree.patch!
make[3]: *** [Makefile:122: /home/sky/build/lede/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/linux-5.10.58/.prepared] Error 1
make[3]: Leaving directory '/home/sky/build/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#9.07#8.45#10.95
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/sky/build/lede'
make[1]: *** [toolchain/Makefile:93: /home/sky/build/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/sky/build/lede'
make: *** [/home/sky/build/lede/include/toplevel.mk:230：world] 错误 2

tianiue, 2021-08-16:
同样问题

cuiyf5516, 2021-08-16:
#7687 该版本已回退，close了

closed 2021-08-16 by cuiyf5516