coolsnowwolf/lede #7689: 660x出现重启的情况，突然就WiFi都不见了，过一会又链接上了

286700234, 2021-08-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：不知道是断网还是路由器重启，WiFi突然断掉了

### (2) 路由器型号和固件版本

A：660X

### (3) 详细日志

Mon Aug 16 22:53:41 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Mon Aug 16 22:53:41 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Aug 16 22:53:41 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Aug 16 22:53:41 2021 user.err vssr-rules[6598]: Start failed!
Mon Aug 16 22:53:42 2021 user.info mwan3rtmon[6556]: Detect rtchange event.
Mon Aug 16 22:53:45 2021 daemon.info dnsmasq[7330]: Connected to system UBus
Mon Aug 16 22:53:49 2021 daemon.info pdnsd[7411]: pdnsd-1.2.9b-par starting.
Mon Aug 16 22:53:50 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Aug 16 22:53:51 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Aug 16 22:53:55 2021 daemon.info dnsmasq[7632]: Connected to system UBus
Mon Aug 16 22:54:07 2021 daemon.info hostapd: wlan1: STA e8:6f:38:cf:c7:df IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Aug 16 22:54:15 2021 daemon.info hostapd: wlan0: STA e0:1f:88:f0:69:5a IEEE 802.11: authenticated
Mon Aug 16 22:54:15 2021 daemon.info hostapd: wlan0: STA e0:1f:88:f0:69:5a IEEE 802.11: associated (aid 3)
Mon Aug 16 22:54:15 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e0:1f:88:f0:69:5a
Mon Aug 16 22:54:15 2021 daemon.info hostapd: wlan0: STA e0:1f:88:f0:69:5a RADIUS: starting accounting session 66F1CAC9DDDBDFD3
Mon Aug 16 22:54:15 2021 daemon.info hostapd: wlan0: STA e0:1f:88:f0:69:5a WPA: pairwise key handshake completed (RSN)
Mon Aug 16 22:54:15 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED e0:1f:88:f0:69:5a
Mon Aug 16 22:54:28 2021 cron.err crond[5277]: time disparity of 1132 minutes detected
Mon Aug 16 22:55:02 2021 daemon.info hostapd: wlan1: STA 70:bb:e9:ac:9f:63 IEEE 802.11: authenticated
Mon Aug 16 22:55:02 2021 daemon.info hostapd: wlan1: STA 70:bb:e9:ac:9f:63 IEEE 802.11: associated (aid 1)
Mon Aug 16 22:55:02 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 70:bb:e9:ac:9f:63
Mon Aug 16 22:55:02 2021 daemon.info hostapd: wlan1: STA 70:bb:e9:ac:9f:63 RADIUS: starting accounting session 85E52B6E9EE5F920
Mon Aug 16 22:55:02 2021 daemon.info hostapd: wlan1: STA 70:bb:e9:ac:9f:63 WPA: pairwise key handshake completed (RSN)
Mon Aug 16 22:55:02 2021 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 70:bb:e9:ac:9f:63
Mon Aug 16 23:01:00 2021 cron.err crond[5277]: USER root pid 7990 cmd sh /usr/share/jd-dailybonus/newapp.sh -u
Mon Aug 16 23:01:30 2021 daemon.info hostapd: wlan0: STA 70:bb:e9:ac:9f:63 IEEE 802.11: authenticated
Mon Aug 16 23:01:30 2021 daemon.info hostapd: wlan0: STA 70:bb:e9:ac:9f:63 IEEE 802.11: associated (aid 4)
Mon Aug 16 23:01:30 2021 daemon.notice hostapd: wlan1: Prune association for 70:bb:e9:ac:9f:63
Mon Aug 16 23:01:30 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 70:bb:e9:ac:9f:63
Mon Aug 16 23:01:32 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 70:bb:e9:ac:9f:63
Mon Aug 16 23:01:32 2021 daemon.info hostapd: wlan0: STA 70:bb:e9:ac:9f:63 RADIUS: starting accounting session 62B62722E741D384
Mon Aug 16 23:01:32 2021 daemon.info hostapd: wlan0: STA 70:bb:e9:ac:9f:63 WPA: pairwise key handshake completed (RSN)
Mon Aug 16 23:01:32 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 70:bb:e9:ac:9f:63
Mon Aug 16 23:02:00 2021 daemon.info hostapd: wlan1: STA 70:bb:e9:ac:9f:63 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Aug 16 23:08:08 2021 daemon.info hostapd: wlan1: STA 14:3c:c3:56:1d:2d IEEE 802.11: authenticated
Mon Aug 16 23:08:08 2021 daemon.info hostapd: wlan1: STA 14:3c:c3:56:1d:2d IEEE 802.11: associated (aid 1)
Mon Aug 16 23:08:08 2021 daemon.notice hostapd: wlan0: Prune association for 14:3c:c3:56:1d:2d
Mon Aug 16 23:08:08 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 14:3c:c3:56:1d:2d
Mon Aug 16 23:08:10 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 14:3c:c3:56:1d:2d
Mon Aug 16 23:08:10 2021 daemon.info hostapd: wlan1: STA 14:3c:c3:56:1d:2d RADIUS: starting accounting session 6DD99BBE23C91F65
Mon Aug 16 23:08:10 2021 daemon.info hostapd: wlan1: STA 14:3c:c3:56:1d:2d WPA: pairwise key handshake completed (RSN)
Mon Aug 16 23:08:10 2021 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 14:3c:c3:56:1d:2d
Mon Aug 16 23:08:38 2021 daemon.info hostapd: wlan0: STA 14:3c:c3:56:1d:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

junexie-op, 2021-08-19:
一样的问题，刷了padavan也会自动重启。感觉是pd-boot的问题。

bilibilistack, 2021-08-30:
AX6 有类似的情况，我看日志并没有重启报错，只有WPA连接断开连接成功连接断开连接成功的日志。
小米手机上表现为断流后掉线，重新连接后卡在正在获取IP地址。

hzjixiang, 2021-09-01:
> 一样的问题，刷了padavan也会自动重启。感觉是pd-boot的问题。
https://github.com/coolsnowwolf/lede/issues/7627
源码的问题，LEDE源码之前的5.10.46内核我测了一周，5.10内核非常稳

junexie-op, 2021-09-01:
> > 一样的问题，刷了padavan也会自动重启。感觉是PD-启动的问题。
> > [＃7627](https://github.com/coolsnowwolf/lede/issues/7627)
> > 源码的问题，LEDE源码之前的46年5月10日内核我测了一周，5.10内核非常稳

还是会重启的，5.10.50.我二级路由挂了PCDN。负载很大。换K2P就稳如狗。

hzjixiang, 2021-09-01:
> > > 一样的问题，刷了padavan也会自动重启。感觉是PD-启动的问题。
> > > [＃7627](https://github.com/coolsnowwolf/lede/issues/7627)
> > > 源码的问题，LEDE源码之前的46年5月10日内核我测了一周，5.10内核非常稳
> 
> 还是会重启的，5.10.50.我二级路由挂了PCDN。负载很大。换K2P就稳如狗。

我这群居房，7部手机，三个笔记本，目前一周了没问题。。。5.4.132重启概率几乎是1天1到两次。

junexie-op, 2021-09-01:
> > > > 一样的问题，刷了padavan也会自动重启。感觉是PD-启动的问题。
> > > > [＃7627](https://github.com/coolsnowwolf/lede/issues/7627)
> > > > 源码的问题，LEDE源码之前的46年5月10日内核我测了一周，5.10内核非常稳
> > 
> > 
> > 还是会重启的，5.10.50。我超级还原挂了PCDN。很有看头。换K2P就稳如狗。
> 
> 我这群居房，7部手机，三本笔记本，现在一周没了。。。5.4.132重启几乎是1天1到曲面。

负载不够大，我就挂了一个京东云PCDN连接数就2000+了，1天-2天必定重启。之前K2P挂100天都稳如狗。

coolsnowwolf, 2026-09-06:
报告只有无线断开前后的服务日志，没有 watchdog、完整内核崩溃或重启原因；同时包含旧版 vssr/mwan3 环境，无法定位当前无线或系统包缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf