coolsnowwolf/lede #7705: lede x86重启掉配置

Dimitri2020007, 2021-08-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86设置好lede之后重启就还原成默认配置了，在hyper-v 1代虚拟机中运行，而且在“备份/升级”中没有执行重置按钮。hyper-v二代也是一样的。用的是最新的源码。

2021.7.30版本的源码没有出现重启掉配置的问题，这是为啥啊？

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

启动的：

[    0.000000] Linux version 5.4.140 (runner@fv-az224-155) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3690-5668f7462)) #0 SMP Wed Aug 18 11:59:29 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=86969391-509a-5f55-dd84-f07367077e02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000f7feffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f7ff0000-0x00000000f7ffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000f7fff000-0x00000000f7ffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000107ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.3 present.
[    0.000000] DMI: Microsoft Corporation Virtual Machine/Virtual Machine, BIOS 090007  05/18/2018
[    0.000000] Hypervisor detected: Microsoft Hyper-V
[    0.000000] Hyper-V: features 0x2e7f, hints 0x60e24, misc 0xbed7b2
[    0.000000] Hyper-V Host Build:17763-10.0-1-0.1339
[    0.000000] Hyper-V: LAPIC Timer Frequency: 0xc3500
[    0.000000] tsc: Marking TSC unstable due to running on Hyper-V
[    0.000000] Hyper-V: Using hypercall for remote TLB flush
[    0.000000] clocksource: hyperv_clocksource_tsc_page: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000003] tsc: Detected 2903.998 MHz processor
[    0.001256] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001257] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001260] last_pfn = 0x108000 max_arch_pfn = 0x400000000
[    0.001295] MTRR default type: uncachable
[    0.001296] MTRR fixed ranges enabled:
[    0.001297]   00000-9FFFF write-back
[    0.001298]   A0000-DFFFF uncachable
[    0.001298]   E0000-FFFFF write-back
[    0.001299] MTRR variable ranges enabled:
[    0.001300]   0 base 0000000000 mask 7F00000000 write-back
[    0.001300]   1 base 0100000000 mask 7000000000 write-back
[    0.001301]   2 disabled
[    0.001301]   3 disabled
[    0.001301]   4 disabled
[    0.001302]   5 disabled
[... 1388 lines cut ...]
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: uc_update: OpenWrt/interface-br-lan/if_octets: ds[1] = 1550.435452
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: rrd_cache_insert: file = /tmp/rrd/OpenWrt/interface-br-lan/if_packets.rrd; values_num = 1; age = 0.000;
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: plugin: plugin_write: Writing values via rrdtool.
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: uc_update: OpenWrt/interface-br-lan/if_errors: ds[0] = 0.000000
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: rrd_cache_insert: file = /tmp/rrd/OpenWrt/interface-br-lan/if_octets.rrd; values_num = 1; age = 0.000;
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: uc_update: OpenWrt/interface-br-lan/if_errors: ds[1] = 0.000000
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: plugin: plugin_write: Writing values via rrdtool.
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: rrd_cache_insert: file = /tmp/rrd/OpenWrt/interface-br-lan/if_errors.rrd; values_num = 1; age = 0.000;
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: queue thread: Wrote 1 value to /tmp/rrd/OpenWrt/interface-br-lan/if_packets.rrd
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: queue thread: Wrote 1 value to /tmp/rrd/OpenWrt/interface-br-lan/if_octets.rrd
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: queue thread: Wrote 1 value to /tmp/rrd/OpenWrt/interface-br-lan/if_errors.rrd
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: cu_rrd_create_file: Successfully created RRD file "/tmp/rrd/OpenWrt/interface-br-lan/if_dropped.rrd".
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: rrd_cache_insert: file = /tmp/rrd/OpenWrt/interface-br-lan/if_dropped.rrd; values_num = 1; age = 0.000;
Fri Aug 20 00:17:48 2021 daemon.err collectd[10183]: rrdtool plugin: queue thread: Wrote 1 value to /tmp/rrd/OpenWrt/interface-br-lan/if_dropped.rrd
Fri Aug 20 00:17:51 2021 daemon.warn Tor[7916]: Problem bootstrapping. Stuck at 5% (conn): Connecting to a relay. (Host is unreachable; NOROUTE; count 12; recommendation warn; host 2736CFC07D1FA472FD42518A46BC720A62E85C38 at 195.154.106.60:443)
Fri Aug 20 00:17:51 2021 daemon.warn Tor[7916]: 11 connections have failed:
Fri Aug 20 00:17:51 2021 daemon.warn Tor[7916]:  11 connections died in state connect()ing with SSL state (No SSL object)
Fri Aug 20 00:17:54 2021 daemon.warn Tor[7916]: Problem bootstrapping. Stuck at 5% (conn): Connecting to a relay. (Host is unreachable; NOROUTE; count 13; recommendation warn; host 2C4E15CD40EE3D2D6F062F04ADFE9B85C8C3C52B at 184.105.220.24:9001)
Fri Aug 20 00:17:54 2021 daemon.warn Tor[7916]: 12 connections have failed:
Fri Aug 20 00:17:54 2021 daemon.warn Tor[7916]:  12 connections died in state connect()ing with SSL state (No SSL object)
Fri Aug 20 00:17:57 2021 daemon.warn Tor[7916]: Problem bootstrapping. Stuck at 5% (conn): Connecting to a relay. (Host is unreachable; NOROUTE; count 14; recommendation warn; host F2044413DAC2E02E3D6BCF4735A19BCA1DE97281 at 131.188.40.189:443)
Fri Aug 20 00:17:57 2021 daemon.warn Tor[7916]: 13 connections have failed:
Fri Aug 20 00:17:57 2021 daemon.warn Tor[7916]:  13 connections died in state connect()ing with SSL state (No SSL object)
Fri Aug 20 00:17:57 2021 user.info mwan3rtmon[9258]: Detect rtchange event.
Fri Aug 20 00:17:57 2021 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)

kisaragychihaya, 2021-08-22:
掉配置先试试看编译小一点的包

Chandler-Lu, 2021-08-22:
某一个版本确实出现过重启以后所有配置重置的问题。。但是重启完之后就好了。
你都虚拟化了，别啥都往里塞，openwrt 留几个必要的东西就好了，把包弄小一点。有些插件服务完全可以上 ubuntu 去跑

Dimitri2020007, 2021-08-22:
> 某一个版本确实出现过重启以后所有配置重置的问题。。但是重启完之后就好了。
> 你都虚拟化了，别啥都往里塞，openwrt 留几个必要的东西就好了，把包弄小一点。有些插件服务完全可以上 ubuntu 去跑

这回的版本是手动重启多次所有配置重置，吓的我也不敢继续编译了。

以前是没有这个问题的。

在hyper-v上运行的。今年7.30号之后的源码出现了掉配置的问题。

turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢？是不是这个bug出现了以后重启就掉配置了？

Chandler-Lu, 2021-08-22:
> turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢

我也想知道，我明明没编译但就是有。但是好像问题不大，我直接手动把 sfe 关了，也出国问题

Chandler-Lu, 2021-08-22:
https://github.com/Chandler-Lu/action-lede/actions 你用我固件试试？

Dimitri2020007, 2021-08-22:
> https://github.com/Chandler-Lu/action-lede/actions 你用我固件试试？

我来了

Chandler-Lu, 2021-08-22:
> > https://github.com/Chandler-Lu/action-lede/actions 你用我固件试试？
> 
> 我来了

我跑 esxi 上没问题

Neverends4, 2021-08-23:
> > 
> turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢？是不是这个bug出现了以后重启就掉配置了？

进menuconfig 点help可以看到sfe是被什么选中的。

Chandler-Lu, 2021-08-23:
> > turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢？是不是这个bug出现了以后重启就掉配置了？
> 
> 进menuconfig 点help可以看到sfe是被什么选中的。

啊似乎找到了，谢谢

Chandler-Lu, 2021-08-23:
> > 某一个版本确实出现过重启以后所有配置重置的问题。。但是重启完之后就好了。
> > 你都虚拟化了，别啥都往里塞，openwrt 留几个必要的东西就好了，把包弄小一点。有些插件服务完全可以上 ubuntu 去跑
> 
> 这回的版本是手动重启多次所有配置重置，吓的我也不敢继续编译了。
> 
> 以前是没有这个问题的。
> 
> 在hyper-v上运行的。今年7.30号之后的源码出现了掉配置的问题。
> 
> turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢？是不是这个bug出现了以后重启就掉配置了？

`kmod-fast-classifier`强制选中了`kmod-shortcut-fe`

Dimitri2020007, 2021-08-23:
> > > 某一个版本确实出现过重启以后所有配置重置的问题。。但是重启完之后就好了。
> > > 你都虚拟化了，别啥都往里塞，openwrt 留几个必要的东西就好了，把包弄小一点。有些插件服务完全可以上 ubuntu 去跑
> > 
> > 
> > 这回的版本是手动重启多次所有配置重置，吓的我也不敢继续编译了。
> > 以前是没有这个问题的。
> > 在hyper-v上运行的。今年7.30号之后的源码出现了掉配置的问题。
> > turboacc上只选了flowoffload没选sfe咋还会有sfe的选项呢？是不是这个bug出现了以后重启就掉配置了？
> 
> `kmod-fast-classifier`强制选中了`kmod-shortcut-fe`

谢谢你的提醒，我这就处理掉这玩意

Chandler-Lu, 2021-08-23:
> 谢谢你的提醒，我这就处理掉这玩意

我发现我这里更新内核有变动的固件之后，所有配置全丢

closed 2022-02-12 by Beginner-Go