coolsnowwolf/lede #7707: 编译入docker后,DDNS内网无法访问

jt8823, 2021-08-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
我试过docker ce   和拉取dockerman
分别编译了两个套固件
都会导致,内网无法通过DDNS访问,外网到可以
如果在不编译docker的情况下,就是正常的
搜了一下,各大站点相关问题, 描述为NAT回流问题
但是没找到有用的解决办法

在安装其它人编译带有docker的固件,又可以正常使用
所以我不知道到底哪一环节出了问题.

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：j1900

固件版本 | OpenWrt R21.8.6 / LuCI Master (git-21.211.30217-03784c3)

内核版本 | 5.4.140

### (3) 详细日志

A：

NagaseKouichi, 2021-08-20:
https://github.com/coolsnowwolf/lede/issues/6149

jt8823, 2021-08-20:
> #6149
> 试试我的解决方法

哟   可以了....  折腾了好多天,  还是得在这儿来找办法
谢谢哈

这防火墙关了
危险吗??

NagaseKouichi, 2021-08-21:
不清楚，但是既然大佬们的代码默认就关闭了网桥防火墙，应该没问题吧

jt8823, 2021-08-21:
> 不清楚，但是既然大佬们的代码默认就关闭了网桥防火墙，应该没问题吧

好的    可以使用了  谢谢
还有一事向请问下
我在配置好两个容器后,   不管选的host还是macvlan模式,   点击创建就变成bridge模式了
这是什么原因啊

lisankai93, 2021-09-20:
> #6149
> 试试我的解决方法

我是lede，试了还是不行

lisankai93, 2021-09-20:
vim /etc/sysctl.conf后
是：
# Defaults are configured in /etc/sysctl.d/* and can be customized in this file
#Enable the TCP BBR congestion control algorithm
net.core.default_qdisc=fq
net.ipv4.tcp_congestion_control=bbr
net.netfilter.nf_conntrack_max=100000
net.bridge.bridge-nf-call-arptables = 0
net.bridge.bridge-nf-call-ip6tables = 0
net.bridge.bridge-nf-call-iptables = 0

然后进行重启，不知道是不是你说 sysctl -p我没有执行的原因？我没法执行命令 sysctl -p？
我运行后显示
root@Openwrt:~# sysctl -p
net.core.default_qdisc = fq
sysctl: write error: No such file or directory

重启不开docker的时候都ok，就是开了docker后不行

coolsnowwolf, 2026-09-06:
Docker 启用后 NAT 回流失败没有提供 firewall、Docker 网段、iptables/nft 规则或路由信息，无法确认是当前 Docker/防火墙包缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf