coolsnowwolf/lede #7711: R4S编译错误有大佬能帮忙解决一下不

JSZMonkey, 2021-08-21:
`make[5]: Entering directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.137'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      kernel/sched/fair.o
kernel/sched/fair.c:10549:12: error: static declaration of 'newidle_balance' follows non-static declaration
 static int newidle_balance(struct rq *this_rq, struct rq_flags *rf)
            ^~~~~~~~~~~~~~~
In file included from kernel/sched/fair.c:27:0:
kernel/sched/sched.h:1483:12: note: previous declaration of 'newidle_balance' was here
 extern int newidle_balance(struct rq *this_rq, struct rq_flags *rf);
            ^~~~~~~~~~~~~~~
kernel/sched/fair.c:10549:12: warning: 'newidle_balance' defined but not used [-Wunused-function]
 static int newidle_balance(struct rq *this_rq, struct rq_flags *rf)
            ^~~~~~~~~~~~~~~
kernel/sched/fair.c:6631:12: warning: 'wake_cap' defined but not used [-Wunused-function]
 static int wake_cap(struct task_struct *p, int cpu, int prev_cpu)
            ^~~~~~~~
make[7]: *** [scripts/Makefile.build:262: kernel/sched/fair.o] Error 1
make[6]: *** [scripts/Makefile.build:497: kernel/sched] Error 2
make[5]: *** [Makefile:1734: kernel] Error 2
make[5]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.137'
make[4]: *** [Makefile:26: /home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.137/.image] Error 2
make[3]: *** [Makefile:11: install] Error 2
make[4]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/target/linux/rockchip'
make[3]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/target/linux'
time: target/linux/install#1.99#0.66#2.62
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt'
make[1]: *** [target/Makefile:19: /home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt'
make: *** [/home/runner/work/R4S-OpenWrt/R4S-OpenWrt/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`

ricksuzade-maker, 2021-08-21:
你这个都不是最新版源码了，重新同步一下。

1715173329, 2021-08-21:
You're __NOT__ using Lean's source tree.
Instead, you tried to mix your own hacks, so resolve anything by yourself.

closed 2021-08-22 by JSZMonkey