coolsnowwolf/lede #7717: 运营商的 DNS 域名劫持

tanst, 2021-08-22:
电信用户，最近发现 cloudflare.com 无法访问，检查了一下，发现 ping 返回 127.0.0.1
用 223.5.5.5 查询了一下，返回正常，应该是默认运营商的 DNS 服务器劫持了，如图

[image]

之前尝试过用 turbo ACC 里面那个 DNS 缓存，结果勾选后就无法出国，索性直接在编译里去掉了。
请问不影响出国的情况下，正确设置自定义上游 DNS 服务器的方法是什么呢？

Neverends4, 2021-08-23:
那要看你主DNS用的什么，turbo ACC好像用的pdnsd，去看下man page，把223.5.5.5设为上游就可以了
默认不做设置确实是用ISP的DNS。

tanst, 2021-08-23:
> 那要看你主DNS用的什么，turbo ACC好像用的pdnsd，去看下man page，把223.5.5.5设为上游就可以了
> 默认不做设置确实是用ISP的DNS。

man page?
是不是 wan 接口设置，是这里吗？我这样改了之后无法出国了。
[image]

Neverends4, 2021-08-23:
你之前说了用了turbo ACC，我记得是使用的pdnsd对网站进行分流，然后OX的网站是采用上游DOH代理的方式解析。
man page是说明页面。pdnsd那里应该分别设置了国内网站和GFW网站的DNS，你应该去那里修改，这里改了就变成全部网站都用腾讯DNS走，被OX的网站自然无法解析或者地址被污染
访问OX网站分为两个部分，一部分是DNS解析正确，一部分是可以访问网站地址的资源。
你需要补课，了解一下DNS和墙的原理。

zhangguanzhang, 2021-09-05:
adh 上游直接doh完事

coolsnowwolf, 2026-09-06:
这是运营商 DNS 污染和自定义上游 DNS 的配置咨询，不是 LEDE 仓库缺陷；应通过 DHCP/DNS 转发配置指定可信上游，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf