coolsnowwolf/lede #7725: 运营商强制断线重连PPPOE拨号后获取IPV4不自动获取IPV6并导致内网设备无法访问外网

Suwipe, 2021-08-23:
问题：运营商每2天的凌晨5 6点强制断线重连PPPOE，自动重新拨号后成功获取IPV4，不自动获取IPV6并会导致内网设备完全无法访问外网，V4的也不行，WAN接口配置高级配置里使用内置的 IPv6 管理勾选，获取 IPv6 地址为自动。
出现不能访问外网问题后进路由器网络接口节点直接点WAN的连接手动重新拨号的就正常，从未出现手动点连接不获取IPV6地址的情况。
此问题一直以来都有，以前出现次数不频繁，最近用的8月初自行构建的版本现在是基本每次被运营商断线都会出现这个问题
现在暂时通过搜索网络说的方法自行添加加了一个接口@wan DHCPV6客户端试试还会不会出现问题

### (2) 路由器型号和固件版本

X86软路由，git-21.211.30217-03784c3

### (3) 详细日志

Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: LCP terminated by peer
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Connect time 2886.8 minutes.
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Sent 1507638168 bytes, received 2035941322 bytes.
Mon Aug 23 05:49:46 2021 daemon.err miniupnpd[31440]: ioctl(s, SIOCGIFADDR, ...): Address not available
Mon Aug 23 05:49:46 2021 daemon.err miniupnpd[31440]: Failed to get IP for interface pppoe-wan
Mon Aug 23 05:49:46 2021 daemon.warn miniupnpd[31440]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Mon Aug 23 05:49:46 2021 daemon.err odhcp6c[20337]: Failed to send RS (Permission denied)
Mon Aug 23 05:49:46 2021 daemon.err odhcp6c[20337]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Mon Aug 23 05:49:46 2021 daemon.notice pppd[19888]: Modem hangup
Mon Aug 23 05:49:46 2021 daemon.notice pppd[19888]: Connection terminated.
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Connect time 2886.8 minutes.
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Sent 1507638168 bytes, received 2035941322 bytes.
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' has lost the connection
Mon Aug 23 05:49:46 2021 daemon.notice netifd: wan_6 (20337): Command failed: Permission denied
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Sent PADT
Mon Aug 23 05:49:46 2021 daemon.info pppd[19888]: Exit.
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' is now down
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' is disabled
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' is enabled
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' is setting up now
Mon Aug 23 05:49:46 2021 daemon.err insmod: module is already loaded - slhc
Mon Aug 23 05:49:46 2021 daemon.err insmod: module is already loaded - ppp_generic
Mon Aug 23 05:49:46 2021 daemon.err insmod: module is already loaded - pppox
Mon Aug 23 05:49:46 2021 daemon.err insmod: module is already loaded - pppoe
Mon Aug 23 05:49:46 2021 daemon.notice netifd: wan (17463): uci: Entry not found
Mon Aug 23 05:49:46 2021 daemon.notice netifd: wan (17463): sh: out of range
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: Plugin rp-pppoe.so loaded.
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: pppd 2.4.8 started by root, uid 0
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: PPP session is 40122
Mon Aug 23 05:49:46 2021 daemon.warn pppd[17492]: Connected to 00:18:82:xx:xx:xx via interface eth4.41
Mon Aug 23 05:49:46 2021 kern.info kernel: [1686677.908130] pppoe-wan: renamed from ppp0
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: Renamed interface ppp0 to pppoe-wan
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: Using interface pppoe-wan
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: Connect: pppoe-wan <--> eth4.41
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: syncppp not active
Mon Aug 23 05:49:46 2021 daemon.info pppd[17492]: CHAP authentication succeeded: Authentication success,Welcome!
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: CHAP authentication succeeded
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: peer from calling number 00:18:82:xx:xx:xx authorized
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: local  LL address fe80::c147:4158:xxxx:xxxx
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: remote LL address fe80::0218:82ff:xxxx:xxxx
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: local  IP address 58.x.x.x
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: remote IP address 58.x.x.x
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: primary   DNS address 59.51.78.210
Mon Aug 23 05:49:46 2021 daemon.notice pppd[17492]: secondary DNS address 222.246.129.80
Mon Aug 23 05:49:46 2021 daemon.notice netifd: Interface 'wan' is now up
Mon Aug 23 05:49:46 2021 user.notice SQM: ERROR: State file does not exist; SQM was not running on interface eth0
Mon Aug 23 05:49:46 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Mon Aug 23 05:49:47 2021 daemon.err odhcp6c[20337]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Mon Aug 23 05:49:47 2021 daemon.notice netifd: Interface 'wan_6' is now down
Mon Aug 23 05:49:47 2021 daemon.notice netifd: Interface 'wan_6' is disabled
Mon Aug 23 05:49:47 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Mon Aug 23 05:49:47 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
[... 31 lines cut ...]
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: Connect: pppoe-wan <--> eth4.41
Mon Aug 23 07:05:42 2021 daemon.info pppd[22866]: syncppp not active
Mon Aug 23 07:05:42 2021 daemon.info pppd[22866]: CHAP authentication succeeded: Authentication success,Welcome!
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: CHAP authentication succeeded
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: peer from calling number 00:18:82:xx:xx:xx authorized
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: local  LL address fe80::1958:37e9:xxxx:xxxx
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: remote LL address fe80::0218:82ff:xxxx:xxxx
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Interface 'wan' is now up
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Interface 'wan_6' is enabled
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity
Mon Aug 23 07:05:42 2021 daemon.notice netifd: Interface 'wan_6' is setting up now
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: local  IP address x.x.x.x
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: remote IP address x.x.x.x
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: primary   DNS address 59.51.78.210
Mon Aug 23 07:05:42 2021 daemon.notice pppd[22866]: secondary DNS address 222.246.129.80
Mon Aug 23 07:05:42 2021 user.notice SQM: ERROR: State file does not exist; SQM was not running on interface eth0
Mon Aug 23 07:05:43 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Mon Aug 23 07:05:43 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Mon Aug 23 07:05:43 2021 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Mon Aug 23 07:05:45 2021 daemon.notice netifd: Interface 'wan_6' is now up
Mon Aug 23 07:05:45 2021 user.notice SQM: ERROR: State file does not exist; SQM was not running on interface eth0
Mon Aug 23 07:05:45 2021 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Mon Aug 23 07:05:45 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)

abclai, 2021-08-23:
这个问题好久啦！最后我的解决方案就是关闭IPv6，毕竟我不可能随时在家！而且就算设置每天重启也是一样！因为你不知道ipv6分配IP！

Neverends4, 2021-08-23:
我这边联通好一些，7天/10800分钟重拨，但IPV4V6都可以自动获取新地址呀
之前自己编译，现在用官方op RC4都没有这个问题。

592732377, 2021-08-24:
一样的问题， PPPOE每2天强制下线一次，拨号成功后无法上网。目前使用watchcat自动重启，临时解决此问题。
(R21.8.6/5.4.138，X86，我取消了turboACC插件，偶尔可以坚持到第2次拨号，也就是路由器最长运行4天)

asushugo, 2021-08-24:
这个问题我也被苦恼过，我用的是联通，我的解决方案是在/etc/hotplug.d/iface中加入了一个脚本，重新拨号后自动检测是否能使用ipv6，否则就重拨，代码如下
`#!/bin/sh

[ "$ACTION" = ifup ] || exit 0
sleep 20s

ipv6Addr=$(curl -s -6 icanhazip.com)
if [ "$ipv6Addr" == "" ]; then
 echo 'no ipv6, now reboot wan interface'
 echo `date '+%Y-%m-%d %H:%M:%S'` no ipv6  >> /tmp/ipv6Check.log
 ifup wan
else
 echo 'have ipv6'
 echo `date '+%Y-%m-%d %H:%M:%S'` have ipv6  >> /tmp/ipv6Check.log
 echo 'you ipv6 address is' $ipv6Addr
fi`

asushugo, 2021-08-24:
想了一下，也许下面这个更好
```
#!/bin/sh

[ "$ACTION" = ifup ] || exit 0
sleep 20s

ipv6Addr=$(ubus call network.interface.wan_6 status | jsonfilter -e '@["ipv6-address"][0].address')
if [ "$ipv6Addr" == "" ]; then
 echo 'no ipv6, now reboot wan interface'
 ifup wan
else
 echo 'have ipv6'
 echo 'you ipv6 address is' $ipv6Addr
fi
```

laaArmy, 2021-08-25:
我想问问，你问题解决没

Suwipe, 2021-08-28:
使用了上面那个脚本，也改了一下只重连wan_6，可以在获取不到IPV6的时候自动重连了，这次再缩短等待时间其实最多5秒就够了
Sat Aug 28 02:55:03 2021 daemon.notice netifd: Interface 'wan_6' is now down
Sat Aug 28 02:55:03 2021 daemon.notice netifd: Interface 'wan_6' is disabled
Sat Aug 28 02:55:03 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Sat Aug 28 02:55:03 2021 user.notice SQM: ERROR: State file does not exist; SQM was not running on interface eth0
Sat Aug 28 02:55:03 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat Aug 28 02:55:03 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Sat Aug 28 02:55:23 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat Aug 28 02:55:23 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat Aug 28 02:55:23 2021 daemon.notice netifd: Interface 'wan_6' is enabled
Sat Aug 28 02:55:23 2021 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat Aug 28 02:55:24 2021 daemon.notice netifd: Interface 'wan_6' is now down
Sat Aug 28 02:55:24 2021 daemon.notice netifd: Interface 'wan_6' is disabled
Sat Aug 28 02:55:24 2021 daemon.notice netifd: Interface 'wan_6' is enabled
Sat Aug 28 02:55:24 2021 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat Aug 28 02:55:24 2021 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sat Aug 28 02:55:26 2021 daemon.notice netifd: Interface 'wan_6' is now up
Sat Aug 28 02:55:26 2021 user.notice SQM: ERROR: State file does not exist; SQM was not running on interface eth0
Sat Aug 28 02:55:26 2021 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat Aug 28 02:55:27 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)

laaArmy, 2021-08-28:
> 使用了上面那个脚本，也改了一下只重连wan_6，可以在获取不到IPV6的时候自动重连了，这次再缩短等待时间其实最多5秒就够了
> Sat Aug 28 02:55:03 2021 daemon.notice netifd: Interface 'wan_6' is now down
> Sat Aug 28 02:55:03 2021 daemon.notice netifd: Interface [...]

在/etc/hotplug.d/iface中加入一个脚本具体是怎么用，能说说吗

592732377, 2021-08-30:
[image]
如图所示，在指定目录下按格式新建一个脚本。可以百度一下hotplug.d、ifup、ifdown运行机制。

uosX, 2021-09-12:
我的办法就是设置每天另存3点自动重新获取ip地址，这样子直接不超过24小时，等到白天直接可用，不影响白天使用

closed 2022-10-31 by aiamadeus