coolsnowwolf/lede #7728: BUSYBOX一直报错

emcsong, 2021-08-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
  登录路由器后发现内存使用涨了很多，正常使用都是稳定在170M左右，这次涨到220M左右，查看系统日志，有很多
 cron.err crond[15323]: crond (busybox 1.33.1) started, log level 8 这种信息输出，查看了系统->计划任务也是空的
 目前还不知道有什么影响，另外还有一些 daemon.err nlbwmon[5560]: Netlink receive failure: Out of memory 的错误，这个已经搜索了相关issue，调了配置目前在看效果
### (1) 具体问题

A：busybox 调度报错

### (2) 路由器型号和固件版本

A：R7800 R21.7.26

### (3) 详细日志

A：Tue Aug 24 07:07:51 2021 cron.err crond[15323]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:08:24 2021 cron.err crond[16319]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:08:25 2021 cron.err crond[16768]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:08:57 2021 cron.err crond[18679]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:08:58 2021 cron.err crond[19128]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:09:31 2021 cron.err crond[20089]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:09:32 2021 cron.err crond[20538]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:10:04 2021 cron.err crond[21506]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:10:05 2021 cron.err crond[21956]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:10:38 2021 cron.err crond[22916]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:10:39 2021 cron.err crond[23365]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:11:11 2021 cron.err crond[24322]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:11:12 2021 cron.err crond[24772]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:11:45 2021 cron.err crond[25732]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:11:46 2021 cron.err crond[26185]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:12:18 2021 cron.err crond[27149]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:12:19 2021 cron.err crond[27598]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:12:52 2021 cron.err crond[28561]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:12:53 2021 cron.err crond[29011]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:13:25 2021 cron.err crond[29968]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:13:26 2021 cron.err crond[30417]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:13:58 2021 cron.err crond[32139]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:14:00 2021 cron.err crond[32588]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:14:32 2021 cron.err crond[1298]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:14:33 2021 cron.err crond[1755]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:15:05 2021 cron.err crond[2719]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:15:07 2021 cron.err crond[3175]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:15:39 2021 cron.err crond[4149]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:15:40 2021 cron.err crond[4598]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:16:12 2021 cron.err crond[5570]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:16:14 2021 cron.err crond[6019]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:16:46 2021 cron.err crond[6976]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:16:47 2021 cron.err crond[7426]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:17:19 2021 cron.err crond[8386]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:17:20 2021 cron.err crond[8837]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:17:53 2021 cron.err crond[9798]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:17:54 2021 cron.err crond[10248]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:18:26 2021 cron.err crond[11207]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:18:27 2021 cron.err crond[11656]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:19:00 2021 cron.err crond[12622]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:19:01 2021 cron.err crond[13071]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:19:34 2021 cron.err crond[14991]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:19:35 2021 cron.err crond[15440]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:20:07 2021 cron.err crond[16397]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:20:08 2021 cron.err crond[16846]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:20:40 2021 cron.err crond[17800]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:20:42 2021 cron.err crond[18249]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 07:21:00 2021 daemon.err nlbwmon[5560]: Netlink receive failure: Out of memory
Tue Aug 24 07:21:00 2021 daemon.err nlbwmon[5560]: Unable to dump conntrack: No buffer space available
[... 527 lines cut ...]
Tue Aug 24 09:49:48 2021 cron.err crond[1112]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:50:20 2021 cron.err crond[3009]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:50:22 2021 cron.err crond[3458]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:50:54 2021 cron.err crond[4840]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:50:55 2021 cron.err crond[5298]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:51:28 2021 cron.err crond[6454]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:51:29 2021 cron.err crond[6921]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:52:01 2021 cron.err crond[8081]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:52:02 2021 cron.err crond[8531]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:52:35 2021 cron.err crond[9841]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:52:36 2021 cron.err crond[10290]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:53:08 2021 cron.err crond[11473]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:53:09 2021 cron.err crond[11933]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:53:42 2021 cron.err crond[13080]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:53:43 2021 cron.err crond[13542]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:54:15 2021 cron.err crond[14733]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:54:16 2021 cron.err crond[15182]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:54:49 2021 cron.err crond[16374]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:54:50 2021 cron.err crond[16823]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:55:22 2021 cron.err crond[17980]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:55:23 2021 cron.err crond[18488]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:55:56 2021 cron.err crond[20566]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:55:57 2021 cron.err crond[21015]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:56:30 2021 cron.err crond[22076]: crond (busybox 1.33.1) started, log level 8
Tue Aug 24 09:56:31 2021 cron.err crond[22525]: crond (busybox 1.33.1) started, log level 8

zlfzy, 2021-08-24:
和你一样，暂时找不到缘由

zlfzy, 2021-08-24:
> 和你一样，暂时找不到缘由

我的是x86

zhangguanzhang, 2021-09-05:
日志看就是 nlbwmon 出问题了

farmerhanxin, 2021-09-10:
确实 k2p也是报这个错 直接断网了

Kisoul, 2021-09-11:
你们解决了吗？

emcsong, 2021-09-15:
刚看了，最近提这个问题的比较多，有人说是解锁网易云音乐nodejs版有bug, 大家可以切一下其他的试试

coolsnowwolf, 2026-09-06:
日志显示的是 crond 被反复启动以及 nlbwmon 内存不足警告，没有计划任务配置、进程树或内存诊断，无法确认 BusyBox 缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf