coolsnowwolf/lede #7730: 5.4.142编译不通过

Eay5, 2021-08-24:
[image]
编译报错

EnnawYang, 2021-08-24:
咱编译过了，请提供详细日志

ZZMHA227, 2021-08-24:
我也有同样的错误。actions编译的无法提供日志。

Beginner-Go, 2021-08-24:
没有详细日志 麻烦不要提issue

jiatianxa, 2021-08-24:
> [image]
> 编译提示
我的同步后编译也是卡这里,可惜是小白不会解决!

Boos4721, 2021-08-24:
[image]

Boos4721, 2021-08-24:
[image]

waynon, 2021-08-24:
我也有这个问题

waynon, 2021-08-25:
云编译的错误日志如下：
```
echo timestamp > s-specs
rm -f tmp-all-tree.def
echo '#include "tree.def"' > tmp-all-tree.def
echo 'END_OF_BASE_TREE_CODES' >> tmp-all-tree.def
echo '#include "c-family/c-common.def"' >> tmp-all-tree.def
ltf="/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/ada/gcc-interface/ada-tree.def /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/cp/cp-tree.def /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/objc/objc-tree.def"; for f in $ltf; do \
  echo "#include \"$f\""; \
done | sed 's|/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/||' >> tmp-all-tree.def
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-all-tree.def all-tree.def
echo timestamp > s-alltree
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/gencheck.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/gencheck.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/gencheck \
    build/gencheck.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/gencheck > tmp-check.h
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-check.h tree-check.h
echo timestamp > s-check
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/gcov-iov.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/gcov-iov.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie \
	build/gcov-iov.o -o build/gcov-iov
build/gcov-iov '8.4.0' '' \
    > tmp-gcov-iov.h
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-gcov-iov.h gcov-iov.h
echo timestamp > s-iov
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/genhooks.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c: In function 'void emit_documentation(const char*)':
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:130:17: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  130 |   while (fscanf (f, "%*[^@]"), buf[0] = '\0',
      |          ~~~~~~~^~~~~~~~~~~~~
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:139:14: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  139 |       fscanf (f, "%999s", buf);
      |       ~~~~~~~^~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:192:14: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  192 |       fscanf (f, "%5[^ \n]", buf);
      |       ~~~~~~~^~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Terminated
make[4]: *** [Makefile:4250: all-gcc] Terminated
make[5]: *** [Makefile:1110: insn-automata.o] Terminated
make[3]: *** [Makefile:37: /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Terminated
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Terminated
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Terminated
Error: The operation was canceled.
```

WYC-2020, 2021-08-25:
咋都那么喜欢云编译，搞不懂，我都是本地编译

waynon, 2021-08-25:
> 我也有同样的错误。actions编译的无法提供日志。

告诉你提供详细日志的方法：修改.github/workflows中的workflow文件（如build-openwrt.yml），将编译固件的命令加上参数 
`-V=s`
形如：`make -j1 V=s`

如果你用的https://github.com/P3TERX/Actions-OpenWrt 的脚本，那个命令在Compile the firmware部分

closed 2021-08-25 by coolsnowwolf