coolsnowwolf/lede #7732: 在树莓派4中用引脚控制关机，不起作用

hyzgit, 2021-08-24:
直接断电很容易出问题，每次ssh连接用命令关机又很麻烦，在树莓派官方系统中，只要在/boot/config中增加"dtoverlay=gpio-shutdown,gpio_pin=3",即可用此引脚执行关机，但是在openwrt中不起作用，有什么办法可以让它起作用吗？

`Name:   gpio-shutdown
Info:   Initiates a shutdown when GPIO pin changes. The given GPIO pin
        is configured as an input key that generates KEY_POWER events.

        This event is handled by systemd-logind by initiating a
        shutdown. Systemd versions older than 225 need an udev rule
        enable listening to the input device:

                ACTION!="REMOVE", SUBSYSTEM=="input", KERNEL=="event*", \
                        SUBSYSTEMS=="platform", DRIVERS=="gpio-keys", \
                        ATTRS{keys}=="116", TAG+="power-switch"

        Alternatively this event can be handled also on systems without
        systemd, just by traditional SysV init daemon. KEY_POWER event
        (keycode 116) needs to be mapped to KeyboardSignal on console
        and then kb::kbrequest inittab action which is triggered by
        KeyboardSignal from console can be configured to issue system
        shutdown. Steps for this configuration are:

            Add following lines to the /etc/console-setup/remap.inc file:

                # Key Power as special keypress
                keycode 116 = KeyboardSignal

            Then add following lines to /etc/inittab file:

                # Action on special keypress (Key Power)
                kb::kbrequest:/sbin/shutdown -t1 -a -h -P now

            And finally reload configuration by calling following commands:

                # dpkg-reconfigure console-setup
                # service console-setup reload
                # init q
        This overlay only handles shutdown. After shutdown, the system
        can be powered up again by driving GPIO3 low. The default
        configuration uses GPIO3 with a pullup, so if you connect a
        button between GPIO3 and GND (pin 5 and 6 on the 40-pin header),
        you get a shutdown and power-up button. Please note that
        Raspberry Pi 1 Model B rev 1 uses GPIO1 instead of GPIO3.
Load:   dtoverlay=gpio-shutdown,<param>=<val>
Params: gpio_pin                GPIO pin to trigger on (default 3)
                                For Raspberry Pi 1 Model B rev 1 set this
                                explicitly to value 1, e.g.:

                                    dtoverlay=gpio-shutdown,gpio_pin=1

        active_low              When this is 1 (active low), a falling
                                edge generates a key down event and a
                                rising edge generates a key up event.
                                When this is 0 (active high), this is
                                reversed. The default is 1 (active low).

        gpio_pull               Desired pull-up/down state (off, down, up)
                                Default is "up".

                                Note that the default pin (GPIO3) has an
                                external pullup. Same applies for GPIO1
                                on Raspberry Pi 1 Model B rev 1.

        debounce                Specify the debounce interval in milliseconds
                                (default 100)

`

还有下面这个，是很实用的功能

`Name:   gpio-poweroff
Info:   Drives a GPIO high or low on poweroff (including halt). Enabling this
        overlay will prevent the ability to boot by driving GPIO3 low.
Load:   dtoverlay=gpio-poweroff,<param>=<val>
Params: gpiopin                 GPIO for signalling (default 26)

        active_low              Set if the power control device requires a
                                high->low transition to trigger a power-down.
                                Note that this will require the support of a
                                custom dt-blob.bin to prevent a power-down
                                during the boot process, and that a reboot
                                will also cause the pin to go low.
        input                   Set if the gpio pin should be configured as
                                an input.
        export                  Set to export the configured pin to sysfs
        timeout_ms              Specify (in ms) how long the kernel waits for
                                power-down before issuing a WARN (default 3000).`

acooler15, 2021-08-24:
[How to make gpio-shutdown overlay work?](https://forum.openwrt.org/t/how-to-make-gpio-shutdown-overlay-work/64088/4)安装`kmod-gpio-button-hotplug`试试

hyzgit, 2021-08-25:
> [How to make gpio-shutdown overlay work?](https://forum.openwrt.org/t/how-to-make-gpio-shutdown-overlay-work/64088/4)安装`kmod-gpio-button-hotplug`试试

谢谢，装上这个后，gpio-shutdown功能可以正常使用了，另一个gpio-poweroff，还是不能用，不知道有没有知道怎么解决这个问题。

acooler15, 2021-08-25:
> > [How to make gpio-shutdown overlay work?](https://forum.openwrt.org/t/how-to-make-gpio-shutdown-overlay-work/64088/4)安装`kmod-gpio-button-hotplug`试试
> 
> 谢谢，装上这个后，gpio-shutdown功能可以正常使用了，另一个gpio-poweroff，还是不能用，不知道有没有知道怎么解决这个问题。

弱弱的问下，`gpio-powroff`一般是用来干嘛的？

coolsnowwolf, 2026-09-06:
当前 bcm27xx 目标已包含 gpio-shutdown 设备树 overlay，gpio-keys 事件也由系统 gpio-button-hotplug 支持；可使用对应内核版本的 overlay 和按键配置实现关机，关闭为 completed。

closed 2026-09-06 by coolsnowwolf