coolsnowwolf/lede #7739: 使用Github Actions编译x86_64的OpenWRT报错

waynon, 2021-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
https://github.com/coolsnowwolf/lede/issues/7730
https://github.com/coolsnowwolf/lede/issues/6987
## 2. 详细叙述

### (1) 具体问题

A：使用https://github.com/P3TERX/Actions-OpenWrt提供的Github Actions脚本云编译x86_64的OpenWRT报错（所有.config中的设置保持默认值）。而使用OpenWRT官方的repo（https://github.com/openwrt/openwrt）进行编译可以成功。

### (2) 路由器型号和固件版本

A：云编译x86_64的OpenWRT，所有.config中的设置保持默认值

### (3) 详细日志

A：
```
echo timestamp > s-specs
rm -f tmp-all-tree.def
echo '#include "tree.def"' > tmp-all-tree.def
echo 'END_OF_BASE_TREE_CODES' >> tmp-all-tree.def
echo '#include "c-family/c-common.def"' >> tmp-all-tree.def
ltf="/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/ada/gcc-interface/ada-tree.def /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/cp/cp-tree.def /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/objc/objc-tree.def"; for f in $ltf; do \
  echo "#include \"$f\""; \
done | sed 's|/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/||' >> tmp-all-tree.def
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-all-tree.def all-tree.def
echo timestamp > s-alltree
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/gencheck.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/gencheck.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/gencheck \
    build/gencheck.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/gencheck > tmp-check.h
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-check.h tree-check.h
echo timestamp > s-check
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/gcov-iov.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/gcov-iov.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie \
	build/gcov-iov.o -o build/gcov-iov
build/gcov-iov '8.4.0' '' \
    > tmp-gcov-iov.h
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-gcov-iov.h gcov-iov.h
echo timestamp > s-iov
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
	-o build/genhooks.o /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c: In function 'void emit_documentation(const char*)':
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:130:17: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  130 |   while (fscanf (f, "%*[^@]"), buf[0] = '\0',
      |          ~~~~~~~^~~~~~~~~~~~~
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:139:14: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  139 |       fscanf (f, "%999s", buf);
      |       ~~~~~~~^~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:192:14: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  192 |       fscanf (f, "%5[^ \n]", buf);
      |       ~~~~~~~^~~~~~~~~~~~~~~~~~~~
/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/genhooks.c:198:14: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
  198 |       fscanf (f, "%999s", buf);
      |       ~~~~~~~^~~~~~~~~~~~~~~~~
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/genhooks \
    build/genhooks.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/genhooks "Target Hook" \
				     > tmp-target-hooks-def.h
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-target-hooks-def.h \
				     target-hooks-def.h
echo timestamp > s-target-hooks-def-h
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/build -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include  -I/workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include  \
[... 102 lines cut ...]
    build/genattrtab.o build/rtl.o build/read-rtl.o build/ggc-none.o build/vec.o build/min-insn-modes.o build/gensupport.o build/print-rtl.o build/hash-table.o build/read-md.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/genattrtab /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/common.md /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/config/i386/i386.md insn-conditions.md \
	-Atmp-attrtab.c -Dtmp-dfatab.c -Ltmp-latencytab.c
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-attrtab.c    insn-attrtab.c
/bin/bash /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/../move-if-change tmp-dfatab.c     insn-dfatab.c
[... cut ...]
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Terminated
make[4]: *** [Makefile:4250: all-gcc] Terminated
make[5]: *** [Makefile:1110: insn-automata.o] Terminated
make[3]: *** [Makefile:37: /workdir/openwrt/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Terminated
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Terminated
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Terminated
Error: The operation was canceled.
```

EnnawYang, 2021-08-25:
`The operation was canceled.`  
说白了，Github Actions白嫖过度

waynon, 2021-08-25:
> `The operation was canceled.`
> 说白了，Github Actions白嫖过度

这是结果，不是原因。
你没有看到上面有其他错误信息吗？是先有了错误，然后才被迫终止的。

不是白嫖过度，我可以用官方OpenWRT仓库编译，3个多小时都没有问题。用lean的源码1个小时就会报错

EnnawYang, 2021-08-25:
> 
> 你没有看到上面有其他错误信息吗？是先有了错误，然后才被迫终止的。

看过了，不是源码错误，源码编译错误会显示Error 1 or 2
而你们这些都是 `Terminated` `The operation was canceled` ，被取消的人还不少。但我自己的Actions就没被取消，你这里俩机型我都有，当然也有其他人编译成功

waynon, 2021-08-25:
https://github.com/coolsnowwolf/lede/issues/7740 贴的日志的倒数第3行：
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2

是Error 2

EnnawYang, 2021-08-25:
> #7740 贴的日志的倒数第3行：
> make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
> 
> 是Error 2

不也是一样吗，`Terminated` `The operation was canceled`，意思就是：不是你自己取消的就是被github取消的，并不是源码错误导致的**被取消**

closed 2021-08-25 by coolsnowwolf

waynon, 2021-08-25:
感谢EnnawYang。

根据https://github.com/actions/virtual-environments/issues/1918 ，操作被取消 可能是因为内存不足。

madman43, 2021-08-25:
我也是100%编译一小时被莫名取消