coolsnowwolf/lede #7740: 使用Github Actions编译 新路由3 的固件报错（报错信息不同于https://github.com/coolsnowwolf/lede/issues/7739 中的）

waynon, 2021-08-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
https://github.com/coolsnowwolf/lede/issues/7739
https://github.com/coolsnowwolf/lede/issues/7730
https://github.com/coolsnowwolf/lede/issues/6987
## 2. 详细叙述

### (1) 具体问题

A：使用https://github.com/P3TERX/Actions-OpenWrt 提供的Github Actions脚本云编译新路由3的固件报错（除型号外所有.config中的设置保持默认值）。**而使用OpenWRT官方的repo（https://github.com/openwrt/openwrt ）进行编译可以成功。**

**注：报错信息不同于https://github.com/coolsnowwolf/lede/issues/7739 中的报错信息**

### (2) 路由器型号和固件版本

A：云编译 新路由3 的固件

### (3) 详细日志

A：
```
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o tree-ssa-tail-merge.o -MT tree-ssa-tail-merge.o -MMD -MP -MF ./.deps/tree-ssa-tail-merge.TPo /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-tail-merge.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o tree-ssa-ter.o -MT tree-ssa-ter.o -MMD -MP -MF ./.deps/tree-ssa-ter.TPo /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-ter.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o tree-ssa-threadbackward.o -MT tree-ssa-threadbackward.o -MMD -MP -MF ./.deps/tree-ssa-threadbackward.TPo /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-threadbackward.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o tree-ssa-threadedge.o -MT tree-ssa-threadedge.o -MMD -MP -MF ./.deps/tree-ssa-threadedge.TPo /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-threadedge.c
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
make[2]: *** wait: No child processes.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** wait: No child processes.  Stop.
make[4]: *** wait: No child processes.  Stop.
make[4]: *** Waiting for unfinished jobs....
make[4]: *** wait: No child processes.  Stop.
make[5]: *** [Makefile:1110: tree-ssa-threadedge.o] Terminated
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[3]: *** [Makefile:37: /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Terminated
Error: The operation was canceled.
```

Boos4721, 2021-08-25:
`Error: The operation was canceled.` 自己翻译

waynon, 2021-08-25:
> `Error: The operation was canceled.` 自己翻译

这是因为之前发生了错误，所以任务被取消（不是我手动取消的）。
错误应该在这一段：
```
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
make[2]: *** wait: No child processes.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** wait: No child processes.  Stop.
make[4]: *** wait: No child processes.  Stop.
make[4]: *** Waiting for unfinished jobs....
make[4]: *** wait: No child processes.  Stop.
make[5]: *** [Makefile:1110: tree-ssa-threadedge.o] Terminated
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[3]: *** [Makefile:37: /workdir/openwrt/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Terminated
```

waynon, 2021-08-25:
我在网上看这是由于使用多线程编译导致的，但是我已经加上了 -j1 参数了。

Boos4721, 2021-08-25:
> 这是因为之前发生了错误，所以任务被取消（不是我手动取消的）。
> 错误应该在这一段：
>
> ```
> make: *** wait: No child processes.  Stop.
> make: *** Waiting for unfinished jobs....
> make: *** wait: No child processes.  Stop.
> make[2]: *** wait: No [...]

没有任何有用的错误信息

closed 2021-08-25 by coolsnowwolf

leopardciaw, 2021-09-13:
> 我在网上看这是由于使用多线程编译导致的，但是我已经加上了 -j1 参数了。

和你一样的错误，已经尝试3次了，依然是这个错误，不知道错在哪儿。
你的问题现在解决了吧。

waynon, 2021-09-13:
🙃没🈶解决