coolsnowwolf/lede #7770: ERROR: package/feeds/packages/samba4 failed to build

zhangguanzhang, 2021-08-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [+] 没有类似的issue

## 2. 详细叙述

**编译 N1 固件的时候，到了 samba4 失败，提示缺少依赖 `python3-markdown`，但是安装了后还是报**

### (1) 具体问题

<details>
<summary>ERROR: package/feeds/packages/samba4 failed to build. 详细日志点击此处展开</summary>

```
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/samba4'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.samba-4.14.6.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "samba-4.14.6.tar.gz" "86760692dd74a04705c0f6d11b31965a477265a50e79eb15838184476146f4b0" "" "https://ftp.gwdg.de/pub/samba/stable/" "https://ftp.heanet.ie/mirrors/ftp.samba.org/stable/" "https://ftp.riken.jp/net/samba/samba/stable/" "http://www.nic.funet.fi/index/samba/pub/samba/stable/" "http://samba.mirror.bit.nl/samba/ftp/stable/" "https://download.samba.org/pub/samba/stable/"    '
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/host-compile#0.39#0.16#0.45
make[3]: Entering directory '/workdir/openwrt/package/libs/jansson'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.jansson-2.13.1.tar.bz2.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "jansson-2.13.1.tar.bz2" "ee90a0f879d2b7b7159124ff22b937a2a9a8c36d3bb65d1da7dd3f04370a10bd" "" "http://www.digip.org/jansson/releases/"    '
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp
echo '4' | cmp -s - /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/jansson.version || { echo '4' > /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/jansson.version;  } 
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/jansson-2.13.1/.pkgdir/jansson/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-armvirt/stamp/.jansson_installed
echo "jansson" >> /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/jansson.default.install
make[3]: Leaving directory '/workdir/openwrt/package/libs/jansson'
time: package/libs/jansson/compile#0.18#0.06#0.22
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/samba4'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.samba-4.14.6.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "samba-4.14.6.tar.gz" "86760692dd74a04705c0f6d11b31965a477265a50e79eb15838184476146f4b0" "" "https://ftp.gwdg.de/pub/samba/stable/" "https://ftp.heanet.ie/mirrors/ftp.samba.org/stable/" "https://ftp.riken.jp/net/samba/samba/stable/" "http://www.nic.funet.fi/index/samba/pub/samba/stable/" "http://samba.mirror.bit.nl/samba/ftp/stable/" "https://download.samba.org/pub/samba/stable/"    '
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/.configured_*
rm -f /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.samba4_installed
cp -fpR ./waf-cross-answers/aarch64.txt /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
echo 'Checking uname sysname type: "OpenWrt"' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
echo 'Checking uname machine type: "aarch64"' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
echo 'Checking uname release type: "5.4.142"' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
echo 'Checking uname version type: "OpenWrt Linux-5.4.142 2021-08-27"' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
# NOTE: special answers for freeBSD/CircleCI
echo 'checking for clnt_create(): OK' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/cross-answers.txt
(cd /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/./; if [ -x ./buildtools/bin/waf ]; then find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6=samba-4.14.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -ffunction-sections -fdata-sections" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6=samba-4.14.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -ffunction-sections -fdata-sections " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6=samba-4.14.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -ffunction-sections -fdata-sections " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed "  CPP="aarch64-openwrt-linux-musl-cpp" XSLTPROC=false WAF_NO_PREFORK=1 PATH="/workdir/openwrt/staging_dir/hostpkg/bin:/workdir/openwrt/staging_dir/hostpkg/usr/bin:/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin:/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin:/workdir/openwrt/staging_dir/host/bin:/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin:/workdir/openwrt/staging_dir/host/bin:/workdir/openwrt/staging_dir/host/bin:/home/linuxbrew/.linuxbrew/bin:/home/linuxbrew/.linuxbrew/sbin:/home/runner/.local/bin:/opt/pipx_bin:/usr/share/rust/.cargo/bin:/home/runner/.config/composer/vendor/bin:/usr/local/.ghcup/bin:/home/runner/.dotnet/tools:/snap/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PYTHON_CONFIG="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/bin/python3.9-config"  ./buildtools/bin/waf configure --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info _python_sysroot="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr" --hostcc="gcc" --cross-compile --cross-answers=cross-answers.txt --disable-cups --disable-iprint --disable-cephfs --disable-fault-handling --disable-glusterfs --disable-spotlight --enable-fhs --without-automount --without-iconv --without-lttng --without-ntvfs-fileserver --without-pam --without-systemd --without-utmp --without-dmapi --without-fam --without-gettext --without-regedit --without-gpgme --accel-aes=none --with-lockdir=/var/lock --with-logfilebase=/var/log --with-piddir=/va [... cut ...]
time: package/feeds/packages/samba4/compile#30.91#20.25#41.04
    ERROR: package/feeds/packages/samba4 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/samba4/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```
</details>

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zhangguanzhang, 2021-08-28:
ref: https://github.com/coolsnowwolf/lede/issues/7486

zhangguanzhang, 2021-08-30:
我同样的步骤回退了后可以编译了
```
if grep -Eq '^PKG_VERSION:=4.14.6' package/feeds/packages/samba4/Makefile; then
    sed -ri -e '/^PKG_VERSION:=/s#4.14.6#4.13.9#' \
        -e '/^PKG_HASH:=/s#:=.+#:=b97a773ed3b4dae6d5ebd3e09337c897ae898b65f38abad550f852b594d4e07f#' \
    package/feeds/packages/samba4/Makefile 
fi
```

不知道为啥新版本依赖检查一直过不去

Vohrt, 2021-11-21:
同样问题
退回4.13.9后解决 #7486

closed 2022-02-12 by Beginner-Go