coolsnowwolf/lede #7774: 新路由3 (newifi D2) 编译错误

waynon, 2021-08-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译错误

### (2) 路由器型号和固件版本

A：newifi D2 (新路由3) 

### (3) 详细日志

A：
```
make[5]: Leaving directory '/home/y/桌面/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.142'
touch /home/y/桌面/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.142/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/home/y/桌面/lede/target/linux/ramips/image'
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/home/y/桌面/lede/target/linux/ramips/image'
make[4]: Leaving directory '/home/y/桌面/lede/target/linux/ramips'
make[3]: Leaving directory '/home/y/桌面/lede/target/linux'
time: target/linux/compile#15.35#4.20#18.73
make[2]: Leaving directory '/home/y/桌面/lede'
export MAKEFLAGS= ;make -w -r buildinfo
make[2]: Entering directory '/home/y/桌面/lede'
 make[3] diffconfig buildversion feedsversion
make[2]: Leaving directory '/home/y/桌面/lede'
make[2]: Entering directory '/home/y/桌面/lede'
rm -rf /home/y/桌面/lede/staging_dir/target-mipsel_24kc_musl/root-ramips
make[2]: Leaving directory '/home/y/桌面/lede'
make[2]: Entering directory '/home/y/桌面/lede'
make[3]: Entering directory '/home/y/桌面/lede/package/libs/libjson-c'
mkdir -p /home/y/桌面/lede/dl
SHELL= flock /home/y/桌面/lede/tmp/.json-c-0.15-nodoc.tar.gz.flock -c '  	/home/y/桌面/lede/scripts/download.pl "/home/y/桌面/lede/dl" "json-c-0.15-nodoc.tar.gz" "99bca4f944b8ced8ae0bbc6310d6a3528ca715e69541793a1ef51f8c5b4b0878" "" "https://s3.amazonaws.com/json-c_releases/releases/"    '
make[3]: Leaving directory '/home/y/桌面/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.14#0.02#0.19
make[3]: Entering directory '/home/y/桌面/lede/package/libs/libubox'
mkdir -p /home/y/桌面/lede/dl
SHELL= flock /home/y/桌面/lede/tmp/.libubox-2021-05-16-b14c4688.tar.xz.flock -c '  	/home/y/桌面/lede/scripts/download.pl "/home/y/桌面/lede/dl" "libubox-2021-05-16-b14c4688.tar.xz" "7dd1db1e0074a9c7c722db654cce3111b3bd3cff0bfd791c4497cb0f6c22d3ca" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/y/桌面/lede/tmp/dl && cd /home/y/桌面/lede/tmp/dl && rm -rf libubox-2021-05-16-b14c4688 && [ \! -d libubox-2021-05-16-b14c4688 ] && git clone  https://git.openwrt.org/project/libubox.git libubox-2021-05-16-b14c4688 && (cd libubox-2021-05-16-b14c4688 && git checkout b14c4688612c05c78ce984d7bde633bce8703b1e && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd libubox-2021-05-16-b14c4688 && git log -1 --format='\''@%ct'\''` && rm -rf libubox-2021-05-16-b14c4688/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libubox-2021-05-16-b14c4688 | 	xz -zc -7e > /home/y/桌面/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz && mv /home/y/桌面/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz /home/y/桌面/lede/dl/ && rm -rf libubox-2021-05-16-b14c4688;  )    '
MAKEFLAGS="" /home/y/桌面/lede/staging_dir/host/bin/ninja  -j1 -C /home/y/桌面/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688 
ninja: Entering directory `/home/y/桌面/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688'
[1/41] Building C object CMakeFiles/jshn.dir/jshn.c.o
FAILED: CMakeFiles/jshn.dir/jshn.c.o 
/home/y/桌面/lede/staging_dir/host/bin/gcc -DJSONC -I/home/y/�/�/�/�/�/�/lede/staging_dir/host/include -I/home/y/�/�/�/�/�/�/lede/staging_dir/host/include/json-c -O2 -I/home/y/桌面/lede/staging_dir/host/include -I/home/y/桌面/lede/staging_dir/hostpkg/include -I/home/y/桌面/lede/staging_dir/target-mipsel_24kc_musl/host/include -DNDEBUG   -Wall -Werror -Wextra -Werror=implicit-function-declaration -Wformat -Werror=format-security -Werror=format-nonliteral -Os -std=gnu99 -g3 -Wmissing-declarations -Wno-unused-parameter -MD -MT CMakeFiles/jshn.dir/jshn.c.o -MF CMakeFiles/jshn.dir/jshn.c.o.d -o CMakeFiles/jshn.dir/jshn.c.o -c jshn.c
jshn.c:17:18: fatal error: json.h: No such file or directory
   17 |         #include <json.h>
      |                  ^~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:116: /home/y/桌面/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688/.built] Error 1
make[3]: Leaving directory '/home/y/桌面/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.11#0.05#0.15
    ERROR: package/libs/libubox [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/libubox/host/compile] Error 1
make[2]: Leaving directory '/home/y/桌面/lede'
make[1]: *** [package/Makefile:110: /home/y/桌面/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
```

EnnawYang, 2021-08-29:
能不能换全英文的环境？

waynon, 2021-08-29:
改为纯英文路径后仍然报错。
```
CMake Deprecation Warning at lua/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking for one of the modules 'lua5.1;lua-5.1'
CMake Deprecation Warning at examples/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_FIND_USE_PACKAGE_REGISTRY
    CMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY
    CMAKE_INSTALL_LIBDIR
    CMAKE_MODULE_LINKER_FLAGS

-- Build files have been written to: /home/y/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688
touch /home/y/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688/.configured
MAKEFLAGS="" /home/y/lede/staging_dir/host/bin/ninja  -j1 -C /home/y/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688 
ninja: Entering directory `/home/y/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688'
[2/42] Building C object CMakeFiles/jshn.dir/jshn.c.o
FAILED: CMakeFiles/jshn.dir/jshn.c.o 
/home/y/lede/staging_dir/host/bin/gcc   -O2 -I/home/y/lede/staging_dir/host/include -I/home/y/lede/staging_dir/hostpkg/include -I/home/y/lede/staging_dir/target-mipsel_24kc_musl/host/include -DNDEBUG   -Wall -Werror -Wextra -Werror=implicit-function-declaration -Wformat -Werror=format-security -Werror=format-nonliteral -Os -std=gnu99 -g3 -Wmissing-declarations -Wno-unused-parameter -MD -MT CMakeFiles/jshn.dir/jshn.c.o -MF CMakeFiles/jshn.dir/jshn.c.o.d -o CMakeFiles/jshn.dir/jshn.c.o -c jshn.c
jshn.c:19:18: fatal error: json/json.h: No such file or directory
   19 |         #include <json/json.h>
      |                  ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:116: /home/y/lede/build_dir/hostpkg/libubox-2021-05-16-b14c4688/.built] Error 1
make[3]: Leaving directory '/home/y/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.55#0.11#0.63
    ERROR: package/libs/libubox [host] failed to build.
make[2]: *** [package/Makefile:114: package/libs/libubox/host/compile] Error 1
make[2]: Leaving directory '/home/y/lede'
make[1]: *** [package/Makefile:110: /home/y/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/y/lede'
make: *** [/home/y/lede/include/toplevel.mk:230：world] 错误 2
```

XiaoliChan, 2021-08-30:
Try “make dirclean".

closed 2021-08-30 by waynon