coolsnowwolf/lede #7783: 无法编译开源WiFi驱动

Luastonsu, 2021-08-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
最新源码编译RM2100路由器固件，只能使用mtwifi闭源驱动，无法编译使用开源兼容无线驱动
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
7月编译的固件取消选择luci-app-mtwifi就会编译开源兼容无线驱动，现在使用最新版本源码如果取消选择的话就没有无线驱动，导致无法使用无线。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
Xiaomi Redmi Router AC2100
A：

### (3) 详细日志

A：

KelvinSoh8, 2021-08-31:
红米路由器AC2100，无线性能，速度上不去，老毛子固件160Mhz可以跑到980Mbps但是Openwrt只有500Mbps不到

我自己使用Lean源码自己编译后，觉得无线性能有点不稳定2.4Ghz有时候，链接不上去，还在研究中，感觉驱动不稳定呢！

Neverends4, 2021-09-01:
[驱动切换](https://github.com/coolsnowwolf/lede/issues/7379#issuecomment-881787811)
说的是切闭源，反过来就是开源。只取消Luci是掩耳盗铃，驱动包都要全部取消，否则两者会冲突，固件怎么知道你要用什么

Ozzy-hzy, 2021-09-02:
开源驱动信号速度差了很多。

Luastonsu, 2021-09-05:
> [驱动切换](https://github.com/coolsnowwolf/lede/issues/7379#issuecomment-881787811)
> 说的是切闭源，反过来就是开源。只取消Luci是掩耳盗铃，驱动包都要全部取消，否则两者会冲突，固件怎么知道你要用什么

自己研究解决了 谢谢

closed 2021-09-05 by Luastonsu