coolsnowwolf/lede #7785: j4105 4口211网卡断网

ramondelee, 2021-08-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：j4105 4口211网卡断网报错，会自动恢复

### (2) 路由器型号和固件版本

A：J4105 -OpenWrt R21.7.15 / LuCI Master (git-21.197.21199-cd0ddb8)-5.10.46

### (3) 详细日志

A：
Mon Aug 30 20:07:03 2021 kern.info kernel: [118460.222776] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Mon Aug 30 20:07:03 2021 kern.info kernel: [118460.352479] br-lan: port 1(eth0) entered blocking state
Mon Aug 30 20:07:03 2021 kern.info kernel: [118460.358196] br-lan: port 1(eth0) entered forwarding state
Mon Aug 30 20:07:03 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Aug 30 20:07:03 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Aug 30 20:07:03 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.229303] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.247777] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.269537] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.322751] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.411610] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.490758] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.564931] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.602239] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.676416] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:04 2021 kern.warn kernel: [118461.702261] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:09 2021 kern.warn kernel: [118466.252593] net_ratelimit: 188 callbacks suppressed
Mon Aug 30 20:07:09 2021 kern.warn kernel: [118466.252600] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:09 2021 kern.warn kernel: [118466.269192] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.948360] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.948442] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.966611] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.970428] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.994808] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118468.994891] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118469.013981] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:12 2021 kern.warn kernel: [118469.016879] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.268455] net_ratelimit: 816 callbacks suppressed
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.268462] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.284819] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.295925] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.306919] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.317889] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.328856] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.339877] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.350964] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.362018] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)
Mon Aug 30 20:07:14 2021 kern.warn kernel: [118471.372997] br-lan: received packet on eth0 with own address as source address (addr:2c:f0:5d:8c:b7:86, vlan:0)

shenyz, 2021-08-31:
内网mac地址冲突

coolsnowwolf, 2026-09-06:
J4105 igb 日志只显示链路重协商及网桥收到本机源地址，没有当前内核、完整网卡信息或可复现配置，无法定位现行驱动缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf