coolsnowwolf/lede #7788: 使用GitHub Actions云编译R2S的OPENWRT固件时一直出现错误。而选择X86的就成功

rejake, 2021-08-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每次使用GitHub Actions云编译R2S的OPENWRT固件时一直出现以下错误(默认插件也会出现），但选择X86的固件就能正常编译完成，请问这是什么原因？

### (2) 路由器型号和固件版本

A：R2S 

### (3) 详细日志

A：2021-08-31T02:29:29.2659284Z 2 thread compile
2021-08-31T02:29:38.3007628Z  make[1] world
2021-08-31T02:29:40.5554416Z  make[2] tools/compile
2021-08-31T02:29:40.6199704Z  make[2] package/cleanup
2021-08-31T02:29:42.8678145Z  make[3] -C tools/flock compile
2021-08-31T02:29:42.9307217Z  make[3] -C tools/xz compile
2021-08-31T02:30:00.0528779Z  make[3] -C tools/sed compile
2021-08-31T02:30:29.2236342Z  make[3] -C tools/patch compile
2021-08-31T02:30:29.2407406Z  make[3] -C tools/tar compile
2021-08-31T02:31:12.3056802Z  make[3] -C tools/m4 compile
2021-08-31T02:31:12.3062997Z  make[3] -C tools/autoconf-archive compile
2021-08-31T02:31:14.1333247Z  make[3] -C tools/pkgconf compile
2021-08-31T02:31:23.9456707Z  make[3] -C tools/ninja compile
2021-08-31T02:31:42.5644979Z  make[3] -C tools/cpio compile
2021-08-31T02:32:03.4719633Z  make[3] -C tools/lzma compile
2021-08-31T02:32:11.7505411Z  make[3] -C tools/mtools compile
2021-08-31T02:32:27.8768521Z  make[3] -C tools/patch-image compile
2021-08-31T02:32:28.0953737Z  make[3] -C tools/sstrip compile
2021-08-31T02:32:29.8309040Z  make[3] -C tools/ucl compile
2021-08-31T02:32:33.6061953Z  make[3] -C tools/xxd compile
2021-08-31T02:32:33.9335700Z  make[3] -C tools/zip compile
2021-08-31T02:32:37.6131333Z  make[3] -C tools/expat compile
2021-08-31T02:32:52.4608386Z  make[3] -C tools/autoconf compile
2021-08-31T02:32:55.8997521Z  make[3] -C tools/libressl compile
2021-08-31T02:33:00.4548762Z  make[3] -C tools/automake compile
2021-08-31T02:33:05.9541707Z  make[3] -C tools/missing-macros compile
2021-08-31T02:33:06.0331064Z  make[3] -C tools/dosfstools compile
2021-08-31T02:33:13.7447533Z  make[3] -C tools/upx compile
2021-08-31T02:33:43.3103460Z  make[3] -C tools/libtool compile
2021-08-31T02:35:12.5020700Z  make[3] -C tools/cmake compile
2021-08-31T02:35:19.9535243Z  make[3] -C tools/mkimage compile
2021-08-31T02:35:36.0933548Z  make[3] -C tools/mklibs compile
2021-08-31T02:35:46.0987331Z  make[3] -C tools/mm-macros compile
2021-08-31T02:35:50.5230512Z  make[3] -C tools/patchelf compile
2021-08-31T02:36:03.2464705Z  make[3] -C tools/gmp compile
2021-08-31T02:37:19.4097736Z  make[3] -C tools/flex compile
2021-08-31T02:37:51.8690611Z  make[3] -C tools/e2fsprogs compile
2021-08-31T02:38:43.9106503Z  make[3] -C tools/fakeroot compile
2021-08-31T02:39:06.5727991Z  make[3] -C tools/gengetopt compile
2021-08-31T02:40:47.2659886Z  make[3] -C tools/mpfr compile
2021-08-31T02:42:15.4340111Z  make[3] -C tools/bison compile
2021-08-31T02:43:14.3580162Z  make[3] -C tools/mpc compile
2021-08-31T02:43:28.2935354Z  make[3] -C tools/bc compile
2021-08-31T02:43:37.9045755Z  make[3] -C tools/findutils compile
2021-08-31T02:44:27.8468493Z  make[3] -C tools/padjffs2 compile
2021-08-31T02:44:27.9874954Z  make[3] -C tools/quilt compile
2021-08-31T02:50:04.6562244Z  make[3] -C tools/zlib compile
2021-08-31T02:50:04.6566711Z  make[3] -C tools/zstd compile
2021-08-31T02:50:11.1146406Z  make[3] -C tools/firmware-utils compile
2021-08-31T02:50:23.8709950Z  make[3] -C tools/make-ext4fs compile
2021-08-31T02:50:26.2365856Z  make[3] -C tools/mtd-utils compile
2021-08-31T02:50:49.1111207Z  make[3] -C tools/squashfskit4 compile
2021-08-31T02:50:55.5074550Z  make[2] toolchain/compile
2021-08-31T02:50:57.7996598Z  make[3] -C toolchain/binutils compile
2021-08-31T02:50:57.7998002Z  make[3] -C toolchain/gdb compile
2021-08-31T02:53:36.8659619Z  make[3] -C toolchain/fortify-headers compile
2021-08-31T02:53:36.9287758Z  make[3] -C toolchain/nasm compile
2021-08-31T02:54:05.4327134Z  make[3] -C toolchain/gcc/initial compile
2021-08-31T03:09:07.2817189Z  make[3] -C toolchain/musl compile
2021-08-31T03:09:07.2846350Z  make[3] -C toolchain/kernel-headers compile
2021-08-31T03:10:11.1626295Z  make[3] -C toolchain/gcc/final compile
2021-08-31T03:12:53.2913463Z make: *** wait: No child processes.  Stop.
2021-08-31T03:12:53.2939235Z make: *** Waiting for unfinished jobs....
2021-08-31T03:12:53.2940449Z make: *** wait: No child processes.  Stop.
2021-08-31T03:12:53.3049937Z ##[error]The runner has received a shutdown signal. This can happen when the runner service is stopped, or a manually started runner is canceled.
2021-08-31T03:12:53.3723726Z ##[error]The operation was canceled.
2021-08-31T03:12:53.3785534Z Cleaning up orphan processes

EnnawYang, 2021-08-31:
`The runner has received a shutdown signal. This can happen when the runner service is stopped, or a manually started runner is canceled.`  
非源码问题

deplives, 2021-08-31:
https://github.com/coolsnowwolf/lede/issues/7757

closed 2021-08-31 by AmadeusGhost