coolsnowwolf/lede #7790: Newifi-D2使用最新源码无法获取到LAN口IP

signigelchan, 2021-08-31:
## 1.关于你要提交的问题

最新源码编译，刷入启动后无法获取到LAN口IP。

- 内核版本

```
5.4.143
```

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新源码[f5cb3a0af40bdf21f96ee899fd075b2beca5232c]，固件编译完成。

- 通过 ``` sysupgrade -v /var/openwrt-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin ``` 方式升级或BREED方式重刷固件，启动后都无法获取到LAN口IP。

- WIFI2G/5G灯不亮，但是PC网络可用列表可以搜索出 ``` OpwnWrt ```，手机连接WIFI会分配到IPv6，但没有IPv4。

> 重刷回之前编译的版本[25369be255dbdebc6168c45cc432f73d726e3468]，功能正常。

### (2) 路由器型号和固件版本

A：Newifi-D2

### (3) 详细日志

A： 因无法获取LAN口IP，所以无法抓取LOG。

AmadeusGhost, 2021-09-01:
不要保留配置升级，建议进 breed 重刷固件。

signigelchan, 2021-09-02:
> 不要保留配置升级，建议进 breed 重刷固件。

问题说明里，使用升级的方式不行后，尝试使用BREED，刷入固件前恢复了出厂设置，依旧不行。

AmadeusGhost, 2021-09-02:
那可能是用旧配置编译的问题，删除 .config 重新编译吧。

signigelchan, 2021-09-02:
> 那可能是用旧配置编译的问题，删除 .config 重新编译吧。

有可能，准备周末重新试试，感谢回复。

closed 2021-09-02 by signigelchan