coolsnowwolf/lede #7797: netdata编译https版本缺少依赖

Dimitri2020007, 2021-09-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：去除netdata disable https参数后编译报错，提示缺少依赖libcrypto.so.1.1和libssl.so.1.1，没有在源码目录中找到libcrypto和libssl

### (2) 路由器型号和固件版本

A：x86_64-efi

### (3) 详细日志

A：

2021-09-01T18:31:22.0883725Z SHELL= flock /workdir/openwrt/tmp/.netdata-v1.30.1.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "netdata-v1.30.1.tar.gz" "e05f8b59d283fb2844280455b9481a2f9104730fd77f535312ff2fec40a6bc11" "" "https://codeload.github.com/netdata/netdata/tar.gz/v1.30.1?"    '
2021-09-01T18:31:22.3001860Z mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2021-09-01T18:31:22.3030971Z SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/.pkgdir/netdata/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2021-09-01T18:31:22.4432714Z touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.netdata_installed
2021-09-01T18:31:22.4573049Z mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/CONTROL /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo
2021-09-01T18:31:22.4593819Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc/netdata/custom-plugins.d
2021-09-01T18:31:22.4615452Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-install/etc/netdata /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc
2021-09-01T18:31:22.4642734Z cp -fpR ./files/netdata.conf /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc/netdata
2021-09-01T18:31:22.4671269Z touch /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc/netdata/.opt-out-from-anonymous-statistics
2021-09-01T18:31:22.4688493Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/lib
2021-09-01T18:31:22.4713663Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-install/usr/lib/netdata /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/lib
2021-09-01T18:31:22.5653016Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/lib/netdata/conf.d/fping.conf /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc
2021-09-01T18:31:22.5685958Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/lib/netdata/conf.d/health_alarm_notify.conf /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc
2021-09-01T18:31:22.5709333Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/lib/netdata/plugins.d/tc-qos-helper.sh /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc
2021-09-01T18:31:22.5737847Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/sbin
2021-09-01T18:31:22.5759212Z install -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-install/usr/sbin/netdata /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/sbin
2021-09-01T18:31:22.5820891Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata
2021-09-01T18:31:22.5846264Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-install/usr/share/netdata /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share
2021-09-01T18:31:22.6238738Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/demo*html
2021-09-01T18:31:22.6266712Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/fonts/*.svg
2021-09-01T18:31:22.6284768Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/fonts/*.ttf
2021-09-01T18:31:22.6306696Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/fonts/*.woff
2021-09-01T18:31:22.6324868Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/images/*.png
2021-09-01T18:31:22.6352389Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/images/*.gif
2021-09-01T18:31:22.6370413Z rm /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/images/*.ico
2021-09-01T18:31:22.6388947Z rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/usr/share/netdata/web/old
2021-09-01T18:31:22.6411280Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc/init.d
2021-09-01T18:31:22.6432012Z install -m0755 ./files/netdata.init /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata/etc/init.d/netdata
2021-09-01T18:31:22.6459195Z find /workdir/openwrt/build_dir/target-x86_64_musl/netdata-1.30.1/ipkg-x86_64/netdata -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
2021-09-01T18:31:24.1840911Z Package netdata is missing dependencies for the following libraries:
2021-09-01T18:31:24.1845803Z libcrypto.so.1.1
2021-09-01T18:31:24.1846515Z libssl.so.1.1
2021-09-01T18:31:24.1852868Z make[3]: *** [Makefile:126: /workdir/openwrt/bin/packages/x86_64/packages/netdata_1.30.1-1_x86_64.ipk] Error 1
2021-09-01T18:31:24.1856013Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/admin/netdata'
2021-09-01T18:31:24.1856885Z time: package/feeds/packages/netdata/compile#1.43#0.55#2.22
2021-09-01T18:31:24.1863699Z     ERROR: package/feeds/packages/netdata failed to build.
2021-09-01T18:31:24.1891573Z make[2]: *** [package/Makefile:116: package/feeds/packages/netdata/compile] Error 1

zhangguanzhang, 2021-09-05:
你编译的机器上要有这些依赖

Dimitri2020007, 2021-09-05:
> 你编译的机器上要有这些依赖

看下update

依赖的问题我已经修改makefile解决了

firstep, 2021-11-16:
`sed -i '/disable-https/d' package/feeds/packages/netdata/Makefile
sed -i '/DEPENDS.*$/s//& +libopenssl/' package/feeds/packages/netdata/Makefile
sed -i '/allow dashboard/a\\tssl certificate = /etc/nginx/conf.d/_lan.crt\n\tssl key = /etc/nginx/conf.d/_lan.key' package/feeds/packages/netdata/files/netdata.conf
sed -i 's/\"http:\/\/\"/window.location.protocol + \"\/\/\"/' package/lean/luci-app-netdata/luasrc/view/netdata/netdata.htm`

以上是我编译时netdata支持https的方法，你是不是没有改netdata.conf去设置ssl证书

Dimitri2020007, 2021-11-16:
> `sed -i '/disable-https/d' package/feeds/packages/netdata/Makefile` `sed -i '/DEPENDS.*$/s//& +libopenssl/' package/feeds/packages/netdata/Makefile` `sed -i '/allow dashboard/a\\tssl certificate = [...]

这个我还没有注意到

Dimitri2020007, 2021-12-10:
> `sed -i '/disable-https/d' package/feeds/packages/netdata/Makefile` `sed -i '/DEPENDS.*$/s//& +libopenssl/' package/feeds/packages/netdata/Makefile` `sed -i '/allow dashboard/a\\tssl certificate = [...]

谢谢你提供的办法，但是还是没有用啊

closed 2022-02-12 by Beginner-Go