coolsnowwolf/lede #7798: 2021.8.25日以后的版本“手机投屏”功能不能使用（附测试方法和有问题的commit_id）

ericwang2006, 2021-09-02:
## 1.  没有类似的issue

## 2. 详细叙述

### (1) 具体问题

2021.8.25日以后的版本手机投屏功能不能使用。
测试方法：路由器（红米AC2100）做主路由，电视盒子和手机连接到路由器的5G WIFI，电视盒子启动乐播投屏（或者其他投屏服务），安卓手机启动优酷视频，进行投屏操作，不能搜索到DLNA服务端（乐播投屏），使用8月25日前的版本没有问题。

涉及到 [#7734](https://github.com/coolsnowwolf/lede/pull/7734) 和 [#7742](https://github.com/coolsnowwolf/lede/pull/7742) 两个commit

### (2) 路由器型号和固件版本

红米AC2100，固件采用8月25号的版本（或者这个日期之后的），我编译的源码版本是**2021-08-25 commit id 634663fe5663c40893ce6527e48af170c80d5626**，无线驱动使用开源驱动，打开以下三项
**CONFIG_PACKAGE_kmod-mt7603
CONFIG_PACKAGE_kmod-mt7615-common
CONFIG_PACKAGE_kmod-mt7615-firmware**
附件中有config文件
[AC2100.config.txt](https://github.com/coolsnowwolf/lede/files/7096250/AC2100.config.txt)

### (3) 详细日志
[kernellog.txt](https://github.com/coolsnowwolf/lede/files/7096245/kernellog.txt)
[syslog.txt](https://github.com/coolsnowwolf/lede/files/7096246/syslog.txt)

### (4) 经我尝试，取消7734和7742编译可以解决问题
[#7734](https://github.com/coolsnowwolf/lede/pull/7734)  [#7742](https://github.com/coolsnowwolf/lede/pull/7742) 
```
git revert --no-edit 634663fe5663c40893ce6527e48af170c80d5626
git revert --no-edit a74c473f42394c02695f909f1cbf9283b10dfc5d
```
### (5) 我分析可能是上游的mac80211代码有问题
其他型号路由器没有测试过，目前只在红米AC2100上复现成功。

z826540272, 2021-09-03:
应该是因为他把那个patch删除的原因.
使用官方op没这个问题.还有就是,你切换成2.4G就能投屏了. 5g不行

ericwang2006, 2021-09-03:
> 应该是因为他把那个patch删除的原因.
> 使用官方op没这个问题.还有就是,你切换成2.4G就能投屏了. 5g不行

[image]

就是因为删除了这个patch？

coolsnowwolf, 2021-09-04:
先 revert 吧，之前的确实没问题

closed 2021-09-06 by coolsnowwolf

Luastonsu, 2021-09-23:
> 先 revert 吧，之前的确实没问题

已经影响到局域网内无线打印机了，现在打印只能连电脑，很不方便
有没有解决方案？