coolsnowwolf/lede #7801: 网络选项里没有"无线"选项

bingxueqingzhi, 2021-09-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络选项里没有"无线"选项

### (2) 路由器型号和固件版本

A：MobiPromo CM520-79F   5.4.143内核

### (3) 详细日志

A：[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.143 (lw@lw) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3703-21642b107)) #0 SMP Thu Sep 2 23:41:22 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: MobiPromo CM520-79F
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.block=0,1 root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 506736K/522240K available (5753K kernel code, 188K rwdata, 1496K rodata, 1024K init, 235K bss, 15504K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x49c with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000220] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000237] pid_max: default: 32768 minimum: 301
[    0.000434] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000453] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001465] CPU: Testing write buffer coherency: ok
[    0.002476] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002613] rcu: Hierarchical SRCU implementation.
[    0.003051] smp: Bringing up secondary CPUs ...
[    0.006123] smp: Brought up 1 node, 4 CPUs
[    0.006142] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006150] CPU: All CPU(s) started in SVC mode.
[    0.010857] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.011014] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.011041] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011249] pinctrl core: initialized pinctrl subsystem
[    0.012269] NET: Registered protocol family 16
[    0.012680] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013931] cpuidle: using governor ladder
[    0.013977] cpuidle: using governor menu
[    0.014704] pstore: Registered ramoops as persistent store backend
[    0.014721] ramoops: using 0x80000@0x83f00000, ecc: 0
[    0.035496] cryptd: max_cpu_qlen set to 1000
[... 397 lines cut ...]
[  105.292488]  slab_reclaimable:2737 slab_unreclaimable:6137
[  105.292488]  mapped:2291 shmem:39 pagetables:138 bounce:0
[  105.292488]  free:96022 free_pcp:605 free_cma:0
[  105.304979] Node 0 active_anon:3152kB inactive_anon:64kB active_file:39408kB inactive_file:16152kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:9164kB dirty:44kB writeback:0kB shmem:156kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[  105.327740] Normal free:383992kB min:16384kB low:20480kB high:24576kB active_anon:3152kB inactive_anon:64kB active_file:39408kB inactive_file:16152kB unevictable:0kB writepending:44kB present:522240kB managed:507760kB mlocked:0kB kernel_stack:1040kB pagetables:552kB bounce:0kB free_pcp:2416kB local_pcp:344kB free_cma:0kB
[  105.355922] lowmem_reserve[]: 0 0 0
[  105.378153] Normal: 72*4kB (ME) 34*8kB (UME) 18*16kB (UME) 19*32kB (UME) 5*64kB (UME) 4*128kB (UME) 1*256kB (U) 1*512kB (M) 2*1024kB (ME) 1*2048kB (U) 92*4096kB (M) = 383984kB
[  105.381563] 13929 total pagecache pages
[  105.397159] 0 pages in swap cache
[  105.400946] Swap cache stats: add 0, delete 0, find 0/0
[  105.404440] Free swap  = 0kB
[  105.409454] Total swap = 0kB
[  105.412611] 130560 pages RAM
[  105.415441] 0 pages HighMem/MovableOnly
[  105.418306] 3620 pages reserved
[  161.096610] mac format error: :
[  161.159518] recv user msg
[  161.159553] oaf msg handle, action = 0
[  161.161210] module init.........
[  161.164786] reset  mac=b4:2e:99:36:59:0e report num to 0
[  161.168198] reset  mac=b0:68:b6:ff:a6:bf report num to 0
[  161.549068] ksmbd: kill command received
[  171.441002] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  171.441167] NFSD: Using legacy client tracking operations.
[  171.448628] NFSD: starting 10-second grace period (net f0000038)

AmadeusGhost, 2021-09-03:
选的内核模块太多了，自己精简一下。

ggllioio, 2021-09-05:
没有无线有可能是art异常，把备份的art重刷下试试

closed 2021-09-10 by bingxueqingzhi