coolsnowwolf/lede #7811: 中兴MF269开启lan的ipv6后经常自动重启

Rhystar, 2021-09-05:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：MF269开启lan ipv6的路由通告和DHCPv6后经常自动重启。关闭lan ipv6的路由通告和DHCPv6后可正常使用。

### (2) 路由器型号和固件版本

A：ZTE MF269 固件版本 2021年9月4日 commit a3c941d1267229a71da2a5cde5055698eadc0fd0

### (3) 详细日志

A：15:27和15:32时重启

Sun Sep  5 15:17:40 2021 daemon.notice procd: /etc/rc.d/S16qca-ssdk: /etc/rc.common: line 1: ssdk_sh: not found
Sun Sep  5 15:17:40 2021 daemon.notice procd: /etc/rc.d/S16qca-ssdk: starting
Sun Sep  5 15:17:40 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Sep  5 15:17:40 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Sep  5 15:17:41 2021 daemon.info dnsmasq[1797]: Connected to system UBus
Sun Sep  5 15:17:42 2021 kern.info kernel: [   18.768468] ECM init
Sun Sep  5 15:17:42 2021 kern.info kernel: [   18.768525] ECM database jhash random seed: 0x655b2d71
Sun Sep  5 15:17:42 2021 kern.info kernel: [   18.770915] ECM init complete
Sun Sep  5 15:17:42 2021 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: net.bridge.bridge-nf-call-ip6tables = 1
Sun Sep  5 15:17:42 2021 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: net.bridge.bridge-nf-call-iptables = 1
Sun Sep  5 15:17:42 2021 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: dev.nss.general.redirect = 1
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: 8021ad
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: 8021q
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: macvlan
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: veth
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: bridge
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: Network device
Sun Sep  5 15:17:43 2021 user.notice : Added device handler type: tunnel
Sun Sep  5 15:17:43 2021 daemon.notice wpa_supplicant[2018]: Successfully initialized wpa_supplicant
Sun Sep  5 15:17:44 2021 kern.info kernel: [   20.367520] br-lan: port 1(eth0) entered blocking state
Sun Sep  5 15:17:44 2021 kern.info kernel: [   20.367555] br-lan: port 1(eth0) entered disabled state
Sun Sep  5 15:17:44 2021 kern.info kernel: [   20.371897] device eth0 entered promiscuous mode
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'lan' is enabled
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'lan' is setting up now
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'lan' is now up
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'loopback' is enabled
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'loopback' is setting up now
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'loopback' is now up
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'wan' is enabled
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'wan6' is enabled
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Network device 'lo' link is up
Sun Sep  5 15:17:44 2021 daemon.notice netifd: Interface 'loopback' has link connectivity
Sun Sep  5 15:17:44 2021 daemon.err block: No "mount.ubi" utility available
Sun Sep  5 15:17:44 2021 daemon.err block: mounting /dev/mtdblock25 (ubi) as /mnt/mtdblock25 failed (25) - Not a tty
Sun Sep  5 15:17:44 2021 daemon.err block: No "mount.ubi" utility available
Sun Sep  5 15:17:44 2021 daemon.err block: mounting /dev/mtdblock26 (ubi) as /mnt/mtdblock26 failed (2) - No such file or directory
Sun Sep  5 15:17:44 2021 daemon.err block: No "mount.ubi" utility available
Sun Sep  5 15:17:44 2021 daemon.err block: mounting /dev/mtdblock27 (ubi) as /mnt/mtdblock27 failed (2) - No such file or directory
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.131771] wlan0: Destroyed NSS virtual interface
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.131806] wlan0: RX-Q purged.
Sun Sep  5 15:17:45 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.199861] wlan1: Destroyed NSS virtual interface
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.199894] wlan1: RX-Q purged.
Sun Sep  5 15:17:45 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Sun Sep  5 15:17:45 2021 cron.err crond[2396]: crond (busybox 1.33.1) started, log level 8
Sun Sep  5 15:17:45 2021 user.notice SQM: WARNING: Unable to get run lock - already held by
Sun Sep  5 15:17:45 2021 kern.err kernel: [   21.485895] debugfs: File 'virt_if' in directory 'stats' already present!
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.486141] wlan1: Created a NSS virtual interface.
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.491730] wlan1: RX-Q and tasklet initialized.
Sun Sep  5 15:17:45 2021 kern.err kernel: [   21.499225] debugfs: File 'virt_if' in directory 'stats' already present!
Sun Sep  5 15:17:45 2021 kern.info kernel: [   21.502654] wlan0: Created a NSS virtual interface.
[... 807 lines cut ...]
Sun Sep  5 15:19:08 2021 daemon.info pppd[6016]: Plugin rp-pppoe.so loaded.
Sun Sep  5 15:19:08 2021 daemon.info pppd[6016]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Sep  5 15:19:08 2021 daemon.notice pppd[6016]: pppd 2.4.8 started by root, uid 0
Sun Sep  5 15:19:11 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED e0:d4:e8:a1:fa:9f
Sun Sep  5 15:19:14 2021 daemon.info hostapd: wlan0: STA e0:d4:e8:a1:fa:9f IEEE 802.11: authenticated
Sun Sep  5 15:19:14 2021 daemon.info hostapd: wlan0: STA e0:d4:e8:a1:fa:9f IEEE 802.11: associated (aid 1)
Sun Sep  5 15:19:14 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e0:d4:e8:a1:fa:9f
Sun Sep  5 15:19:14 2021 daemon.info hostapd: wlan0: STA e0:d4:e8:a1:fa:9f RADIUS: starting accounting session CF26842A51F90AE9
Sun Sep  5 15:19:14 2021 daemon.notice hostapd: wlan0: STA-OPMODE-SMPS-MODE-CHANGED e0:d4:e8:a1:fa:9f off
Sun Sep  5 15:19:23 2021 daemon.warn pppd[6016]: Timeout waiting for PADO packets
Sun Sep  5 15:19:23 2021 daemon.err pppd[6016]: Unable to complete PPPoE Discovery
Sun Sep  5 15:19:23 2021 daemon.info pppd[6016]: Exit.
Sun Sep  5 15:19:23 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Sep  5 15:19:23 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Sep  5 15:19:24 2021 daemon.err insmod: module is already loaded - slhc
Sun Sep  5 15:19:24 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Sep  5 15:19:24 2021 daemon.err insmod: module is already loaded - pppox
Sun Sep  5 15:19:24 2021 daemon.err insmod: module is already loaded - pppoe
Sun Sep  5 15:19:24 2021 daemon.notice netifd: wan (6167): uci: Entry not found
Sun Sep  5 15:19:24 2021 daemon.notice netifd: wan (6167): sh: out of range
Sun Sep  5 15:19:24 2021 daemon.info pppd[6200]: Plugin rp-pppoe.so loaded.
Sun Sep  5 15:19:24 2021 daemon.info pppd[6200]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Sep  5 15:19:24 2021 daemon.notice pppd[6200]: pppd 2.4.8 started by root, uid 0
Sun Sep  5 15:19:26 2021 authpriv.info dropbear[6336]: Child connection from 192.168.0.216:1528
Sun Sep  5 15:19:29 2021 authpriv.notice dropbear[6336]: Password auth succeeded for 'root' from 192.168.0.216:1528

Boos4721, 2021-09-05:
[自己看](https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=5240925&pid=13224761)

Rhystar, 2021-09-05:
> [自己看](https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=5240925&pid=13224761)

好的谢谢，我修改试试

Rhystar, 2021-09-05:
> [自己看](https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=5240925&pid=13224761)

没有效果，改过之后隔几分钟还是重启，重启前内存剩余150MB以上，应该不是内存满了的问题

asushugo, 2021-09-06:
应该是OOM了，你留意一下内存看看，可能是无线部分内存泄露导致的

Rhystar, 2021-09-06:
> 应该是OOM了，你留意一下内存看看，可能是无线部分内存泄露导致的

好的谢谢，晚上回去我再测试一下

Rhystar, 2021-09-07:
> 应该是OOM了，你留意一下内存看看，可能是无线部分内存泄露导致的

上次测试重启时内存剩余184MB。而且关闭ipv6后一切正常，可以稳定运行48小时以上，可能不是无线内存泄漏导致的问题。开启ipv6后重启频率很高，大概5~10分钟重启一次。

Rhystar, 2021-09-07:
> 应该是OOM了，你留意一下内存看看，可能是无线部分内存泄露导致的

关闭无线，开启ipv6的状态下也会重启

1582130940, 2021-09-11:
AX3600出现相同情况，在禁用LAN-IPv6设置中和路由通告服务和DHCPv6 服务后重启现象消失。不确定是否是WAN链接方式为DHCP导致（WAN-PPPoE下无此问题）

closed 2021-11-30 by AmadeusGhost