coolsnowwolf/lede #7816: 5.10内核编译quantenna驱动报错

Dimitri2020007, 2021-09-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64-efi

### (3) 详细日志

A：

2021-09-05T21:09:00.0737723Z make[3]: Entering directory '/workdir/openwrt/package/firmware/quantenna'
2021-09-05T21:09:00.2045693Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/.built
2021-09-05T21:09:00.2069778Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/.built_check
2021-09-05T21:09:00.2099224Z CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44=quantenna-37.3.2.44 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include/tirpc  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44=quantenna-37.3.2.44 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include/tirpc  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -ltirpc " make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44 AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44=quantenna-37.3.2.44 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include/tirpc" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" qcsapi_sockrpc
2021-09-05T21:09:00.2113103Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44'
2021-09-05T21:09:00.2202121Z make[4]: 'qcsapi_sockrpc' is up to date.
2021-09-05T21:09:00.2203059Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44'
2021-09-05T21:09:00.2206920Z echo "######################################### PCIE2 DRIVER BUILD #####################################"
2021-09-05T21:09:00.2218076Z ######################################### PCIE2 DRIVER BUILD #####################################
2021-09-05T21:09:00.2218991Z cd /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm
2021-09-05T21:09:00.2237800Z make  -j1 -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.61" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Sep  5 05:36:13 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/netatop.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/rtpengine.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/ksmbd.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/xtables-addons.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/antfs.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/ntfs3.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/siit.symvers" KERNELRELEASE=5.10.61 M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm" NOSTDINC_FLAGS="" modules
2021-09-05T21:09:00.2275160Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.61'
2021-09-05T21:09:00.8245782Z   CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.o
2021-09-05T21:09:01.2713183Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.c: In function 'qdpc_map_bar':
2021-09-05T21:09:01.2716444Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.c:192:10: error: implicit declaration of function 'ioremap_nocache'; did you mean 'ioremap_cache'? [-Werror=implicit-function-declaration]
2021-09-05T21:09:01.2718273Z   vaddr = ioremap_nocache(busaddr, len);
2021-09-05T21:09:01.2719019Z           ^~~~~~~~~~~~~~~
2021-09-05T21:09:01.2719644Z           ioremap_cache
2021-09-05T21:09:01.2721483Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.c:192:8: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
2021-09-05T21:09:01.2723022Z   vaddr = ioremap_nocache(busaddr, len);
2021-09-05T21:09:01.2723699Z         ^
2021-09-05T21:09:01.2725182Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.c: In function 'qdpc_map_pciemem':
2021-09-05T21:09:01.2727635Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.c:339:9: warning: returning 'int' from a function with return type 'void *' makes pointer from integer without a cast [-Wint-conversion]
2021-09-05T21:09:01.2729254Z   return ioremap_nocache(busaddr, len);
2021-09-05T21:09:01.2729967Z          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2021-09-05T21:09:01.3263838Z cc1: some warnings being treated as errors
2021-09-05T21:09:01.3308671Z make[5]: *** [scripts/Makefile.build:280: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm/../common/qdpc_pcie.o] Error 1
2021-09-05T21:09:01.3314901Z make[4]: *** [Makefile:1822: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/drivers/pcie2/host/arm] Error 2
2021-09-05T21:09:01.3316067Z make[3]: *** [Makefile:128: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/quantenna-37.3.2.44/.built] Error 2
2021-09-05T21:09:01.3317256Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.61'
2021-09-05T21:09:01.3318146Z make[3]: Leaving directory '/workdir/openwrt/package/firmware/quantenna'
2021-09-05T21:09:01.3320330Z time: package/firmware/quantenna/compile#0.80#0.27#1.26
2021-09-05T21:09:01.3325271Z     ERROR: package/firmware/quantenna failed to build.
2021-09-05T21:09:01.3340578Z make[2]: *** [package/Makefile:116: package/firmware/quantenna/compile] Error 1

zhangguanzhang, 2021-09-09:
俺也一样

Dimitri2020007, 2021-09-09:
> 俺也一样

目前只能取消这个驱动的编译了，在5.4内核下一切正常

zhangguanzhang, 2021-09-09:
> > 俺也一样
> 
> 目前只能取消这个驱动的编译了，在5.4内核下一切正常

怎么取消，我在config里搜不到这个东西

Dimitri2020007, 2021-09-09:
在.config中找到这2个条目
```
CONFIG_PACKAGE_qtn-utils=y
CONFIG_PACKAGE_kmod-qtn-pcie2=y
```
给它改成这样
```
# CONFIG_PACKAGE_qtn-utils is not set
# CONFIG_PACKAGE_kmod-qtn-pcie2 is not set
```
就取消编译了

两个有依赖关系的
先取消kmod才能取消另一个

zhangguanzhang, 2021-09-09:
怎么切回5.4的内核，我看config里没地方选择内核版本

Dimitri2020007, 2021-09-09:
在global build option下有using tested kernel version

去掉那个就能用回5.4

zhangguanzhang, 2021-09-13:
> 在global build settings下有using the testing kernel version
> 
> 去掉那个就能用回5.4

```
# CONFIG_HAS_TESTING_KERNEL is not set
```