coolsnowwolf/lede #7824: 概况页不显示ipv6 wan状态了。。。

drw2005, 2021-09-07:
最近编译的几个版本，概况页都不显示ipv6 wan状态了。但是功能是正常的，也能正常获得ipv6地址。

请教大佬这是何原因？是编译时哪个插件没有选上？
[image]

4月份时编译的版本还能正常显示：
[image]

hcym, 2021-09-07:
6有点问题，只有一个早期的ipv6非常好用，大概是两个星期左右的，还是手机版很少更新比较安逸

drw2005, 2021-09-08:
哪位大佬知道原因，指点指点啊，谢谢了！

lexin8, 2021-09-09:
If there is no such problem, reinstall the system

LXY631, 2021-11-25:
遇到同样问题，占个坑，希望大神解决。新编译的版本 R21.11.11 ，状态-概览 无法展示 ipv6 概况了，以前一直使用 isp 下发的 pd 动态解析。

WYC-2020, 2022-02-09:
我虚拟机测试没问题
[image]

drw2005, 2022-10-10:
extra packages这里选上就可以了：
[image]

firstep, 2023-01-30:
> 我是拨号后可以显示一段时间ipv6 wan状态，过一会就变成未连接了

和你一样，过一会儿ipv6 wan状态就未连接， 需要重启op才能用v6，现在重启接口没用，很烦

coolsnowwolf, 2026-09-05:
当前 LuCI 状态页已通过 `getWAN6Networks()` 和 DHCPv6 状态渲染 WAN6 信息（与 #10929 相同代码证据）；该 issue 的 R21.8.6 旧版本问题已被当前实现覆盖，关闭为 completed。

closed 2026-09-05 by coolsnowwolf