coolsnowwolf/lede #7827: RedMi AX6活动连接数最大限制不能修改

idelame, 2021-09-07:
### (1) 具体问题

A：活动连接数最大限制默认已经改为65535，但系统依然是8192，因为连接数的增长，造成不断的丢包。

### (2) 路由器型号和固件版本

A：RedMi AX6 
B：OpenWrt R21.8.6

### (3) 详细日志
Tue Sep  7 22:56:25 2021 kern.warn kernel: [ 2382.046650] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:25 2021 kern.warn kernel: [ 2382.049884] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:25 2021 kern.warn kernel: [ 2382.052409] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.002371] net_ratelimit: 508 callbacks suppressed
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.002375] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.005276] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.008862] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.018789] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.019071] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.032815] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.044779] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.047416] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.047419] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:30 2021 kern.warn kernel: [ 2387.047422] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.005707] net_ratelimit: 499 callbacks suppressed
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.005712] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.013334] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.013414] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.015976] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.016173] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.016456] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.034526] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.038974] nf_conntrack: nf_conntrack: table full, dropping packet
Tue Sep  7 22:56:35 2021 kern.warn kernel: [ 2392.040773] nf_conntrack: nf_conntrack: table full, dropping packet

Boos4721, 2021-09-07:
https://github.com/coolsnowwolf/lede/blob/1214d8dc98f76aea8050f999c428e7c61361949d/package/qca/nss/qca-nss-ecm-64/Makefile#L50

idelame, 2021-09-08:
> https://github.com/coolsnowwolf/lede/blob/1214d8dc98f76aea8050f999c428e7c61361949d/package/qca/nss/qca-nss-ecm-64/Makefile#L50

感谢，已经解决。

closed 2021-09-08 by idelame