coolsnowwolf/lede #7830: 额外的HOSTS不起作用了。。。

drw2005, 2021-09-08:
之前用的4月份编译的版本，一直启用“额外的HOSTS”，拦截广告之类的效果很好，一切正常。

最近编译的版本，这个似乎不起作用了。（重新恢复4月份的版本验证过，正常起作用）

请教大佬这是哪里问题？

[image]

Neverends4, 2021-09-08:
看看 `/etc/dnsmasq.conf` 里面有没有你这个`/etc/hosts_ad`
或者手动添加到`/etc/config/dhcp`然后重启dnsmasq试试

drw2005, 2021-09-08:
> 看看 /etc/dnsmasq.conf 里面有没有你这个/etc/hosts_ad
> 或者手动添加到 etc/config/dhcp然后重启dnsmasq试试

谢谢指点，看了下，dnsmasq.conf里确实没有/etc/hosts_ad，dhcp里倒是有一行：list addnhosts '/etc/hosts_ad'

这样的情况是对的还是不对？应该如何处理？

再次感谢指点！

Neverends4, 2021-09-09:
看起来配置没问题，不清楚了，我没用addnhosts这个参数，能确定的是conf-dir 参数有效，
我在/etc/dnsmasq.conf里加了一行
`conf-dir=/etc/dnsmasq.d`
然后把几个自定义host文件放在了/etc/dnsmasq.d文件夹下面，重启后确认是生效的，具体固件版本没记，不过应该是0722号的源码，你可以试试。现在用官方op的RC4也没问题，

ddd-zero, 2022-02-19:
我的也是不生效。今天才发现的。
list addnhosts '/etc/hosts_ad'
这个命令有。直接修改主hosts也不行。路由器能识别，但电脑识别不到

coolsnowwolf, 2026-09-06:
旧版“额外 Hosts”不生效没有当前 dnsmasq 配置、解析结果或缓存日志，且基于 R21.8.6，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf