coolsnowwolf/lede #7833: 达人请帮忙看一下，本地和去编译都报错

sclsyin, 2021-09-08:
日志如下：
make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/ninja compile
 make[3] -C tools/cpio compile
 make[3] -C tools/lzma compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/ucl compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/expat compile
 make[3] -C tools/kernel2minor compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/libressl compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/upx compile
 make[3] -C tools/libtool compile
 make[3] -C tools/cmake compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpc compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/padjffs2 compile
 make[3] -C tools/quilt compile
 make[3] -C tools/zstd compile
 make[3] -C tools/zlib compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/squashfskit4 compile
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/musl compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/gcc/final compile
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[2] package/compile
 make[3] diffconfig buildversion feedsversion
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/ncurses host-compile
 make[3] -C package/system/fwtool host-compile
[... 1372 lines cut ...]
Thread model: posix
InstalledDir: ../../third_party/llvm-build/Release+Asserts/bin
clang: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: ../../tools/clang/crashreports/xdgmimemagic-88a69b.c
clang: note: diagnostic msg: ../../tools/clang/crashreports/xdgmimemagic-88a69b.sh
clang: note: diagnostic msg: 

********************
ninja: build stopped: subcommand failed.
Makefile:135: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/naiveproxy-93.0.4577.63-1/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#1.03#0.32#0.95
    ERROR: package/feeds/helloworld/naiveproxy failed to build.
package/Makefile:114: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

Dimitri2020007, 2021-09-08:
naiveproxy编译出错了，可能是上游版本又更新了。

Dimitri2020007, 2021-09-11:
可能patch没跟上吧