coolsnowwolf/lede #7838: R4s最新提交pull后编译失败，请大佬看看日志什么原因

GreatMichaelFu, 2021-09-10:
此前一直都OK的，pull了最新提交后编译失败。谢谢！

/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/scripts/dtc/dtc -O dtb -i/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-nanopi-r2s.dtb /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-nanopi-r2s.dtb.tmp
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts:488.3-13: Warning (reg_format): /usb@ff600000/dwc3@ff600000/usb-eth@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-nanopi-r2s.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-nanopi-r2s.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts:486.12-491.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #address-cells value
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts:486.12-491.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #size-cells value
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1067.25-1078.5: Warning (interrupt_provider): /pinctrl/gpio0@ff210000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1080.25-1091.5: Warning (interrupt_provider): /pinctrl/gpio1@ff220000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1093.25-1104.5: Warning (interrupt_provider): /pinctrl/gpio2@ff230000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1106.25-1117.5: Warning (interrupt_provider): /pinctrl/gpio3@ff240000: Missing #address-cells in interrupt provider
rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-nanopi-r2s.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/include -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/include/ -undef -D__DTS__  -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb.tmp /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/scripts/dtc/dtc -O dtb -i/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb.tmp
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts:91.4-22: Warning (reg_format): /i2c@ff120000/eeprom@51/mac-address@fa:reg: property has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts:90.31-92.5: Warning (avoid_default_addr_size): /i2c@ff120000/eeprom@51/mac-address@fa: Relying on default #address-cells value
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts:90.31-92.5: Warning (avoid_default_addr_size): /i2c@ff120000/eeprom@51/mac-address@fa: Relying on default #size-cells value
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1939.25-1950.5: Warning (interrupt_provider): /pinctrl/gpio0@ff720000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1952.25-1963.5: Warning (interrupt_provider): /pinctrl/gpio1@ff730000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1965.25-1976.5: Warning (interrupt_provider): /pinctrl/gpio2@ff780000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1978.25-1989.5: Warning (interrupt_provider): /pinctrl/gpio3@ff788000: Missing #address-cells in interrupt provider
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399.dtsi:1991.25-2002.5: Warning (interrupt_provider): /pinctrl/gpio4@ff790000: Missing #address-cells in interrupt provider
rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-nanopi-r4s.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/include -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/include/ -undef -D__DTS__  -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb.tmp /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3399-rockpro64.dts
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/scripts/dtc/dtc -O dtb -i/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-rockpro64.dtb.tmp
[... cut ...]
make[3]: Leaving directory '/home/builder/lede_r4s/target/linux'
time: target/linux/install#8.68#7.47#26.51
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/builder/lede_r4s'
make[1]: *** [target/Makefile:19: /home/builder/lede_r4s/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/builder/lede_r4s'
make: *** [/home/builder/lede_r4s/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_r4s$

KLT007, 2021-09-11:
是补丁没有生效
aarch64-openwrt-linux-musl-cpp: error: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.61/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts: No such file or directory
此已提交补丁https://github.com/coolsnowwolf/lede/pull/7846
等待合并

galaxyskyknight, 2021-09-11:
最新提交已解决

closed 2021-09-11 by GreatMichaelLee