coolsnowwolf/lede #7840: turboacc的sfe运行问题

baby195lxl, 2021-09-10:
反馈bug/问题模板，提建议请删除

 [#1.关于你要提交的问题]

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
 A：采用最新源码编译的turboacc，选择sfe后流量分载显示未运行

### (2) 路由器型号和固件版本

A：斐讯n1盒子

### (3) 详细日志

A：
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.000000] Linux version 5.4.144-flippy-63+o (root@gtking-pro) (gcc version 11.2.1 20210816 [revision 056e324ce46a7924b5cf10f61010cf9dd2ca10e9] (SUSE Linux)) #318 SMP PREEMPT Fri Sep 3 23:04:42 CST 2021
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Machine model: Phicomm N1
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] efi: Getting EFI parameters from FDT:
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] efi: UEFI not found.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x0000000063800000, size 256 MiB
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 478464
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 7616 pages used for memmap
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 478464 pages, LIFO batch:63
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] psci: PSCIv0.2 detected in firmware.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] psci: Trusted OS migration not required
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s47576 r8192 d30248 u86016
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s47576 r8192 d30248 u86016 alloc=21*4096
Fri Sep 10 23:10:30 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 470848
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.000000] Kernel command line: root=UUID=94c9c65d-588e-49a3-9435-aed7c60ad81a rootfstype=btrfs rootflags=compress=zstd console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 mac=FC:7C:02:89:E4:7C
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] Memory: 1582300K/1913856K available (12476K kernel code, 1446K rwdata, 7412K rodata, 896K init, 851K bss, 69412K reserved, 262144K cma-reserved)
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.000000] random: get_random_u64 called from __kmem_cache_create+0x38/0x570 with crng_init=0
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] rcu:  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000]       Tasks RCU enabled.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] irq_meson_gpio: 110 to 8 gpio interrupt mux initialized
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000003] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000264] Console: colour dummy device 80x25
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000552] printk: console [tty0] enabled
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000589] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=24000)
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000609] pid_max: default: 32768 minimum: 301
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000730] LSM: Security Framework initializing
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000776] SELinux:  Initializing.
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.000848] *** VALIDATE SELinux ***
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000905] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
Fri Sep 10 23:10:30 2021 kern.info kernel: [    0.000959] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.001018] *** VALIDATE tmpfs ***
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.001412] *** VALIDATE proc ***
Fri Sep 10 23:10:30 2021 kern.notice kernel: [    0.001618] *** VALIDATE cgroup1 ***
[... 529 lines cut ...]
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: Joining mDNS multicast group on interface vethbc250a3.IPv6 with address fe80::f84a:24ff:fe63:1459.
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: New relevant interface vethbc250a3.IPv6 for mDNS.
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: Registering new address record for fe80::f84a:24ff:fe63:1459 on vethbc250a3.*.
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: Joining mDNS multicast group on interface docker0.IPv6 with address fe80::42:a4ff:fe9c:a527.
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: New relevant interface docker0.IPv6 for mDNS.
Fri Sep 10 23:25:29 2021 daemon.info avahi-daemon[1635]: Registering new address record for fe80::42:a4ff:fe9c:a527 on docker0.*.
Fri Sep 10 23:25:35 2021 daemon.err dockerd[2569]: time="2021-09-10T23:25:35.150632194+08:00" level=error msg="copy shim log" error="read /proc/self/fd/13: file already closed"
Fri Sep 10 23:25:35 2021 kern.info kernel: [  895.738501] docker0: port 1(vethbc250a3) entered disabled state
Fri Sep 10 23:25:35 2021 kern.info kernel: [  895.739095] vethcb5cb9d: renamed from eth0
Fri Sep 10 23:25:35 2021 daemon.info avahi-daemon[1635]: Interface vethbc250a3.IPv6 no longer relevant for mDNS.
Fri Sep 10 23:25:35 2021 daemon.info avahi-daemon[1635]: Leaving mDNS multicast group on interface vethbc250a3.IPv6 with address fe80::f84a:24ff:fe63:1459.
Fri Sep 10 23:25:35 2021 kern.info kernel: [  895.781710] docker0: port 1(vethbc250a3) entered disabled state
Fri Sep 10 23:25:35 2021 kern.info kernel: [  895.783588] device vethbc250a3 left promiscuous mode
Fri Sep 10 23:25:35 2021 kern.info kernel: [  895.786961] docker0: port 1(vethbc250a3) entered disabled state
Fri Sep 10 23:25:35 2021 daemon.info avahi-daemon[1635]: Withdrawing address record for fe80::f84a:24ff:fe63:1459 on vethbc250a3.
Fri Sep 10 23:26:36 2021 daemon.notice ttyd[2733]: WS closed from 192.168.123.226, clients: 0
Fri Sep 10 23:26:36 2021 daemon.notice ttyd[2733]: killing process 29804 with signal: 1 (SIGHUP)
Fri Sep 10 23:26:36 2021 daemon.notice ttyd[2733]: process killed with signal 1 (SIGHUP), pid: 29804
Fri Sep 10 23:37:33 2021 daemon.notice ttyd[2733]: HTTP / - 192.168.123.226
Fri Sep 10 23:37:33 2021 daemon.notice ttyd[2733]: HTTP /token - 192.168.123.226
Fri Sep 10 23:37:33 2021 daemon.notice ttyd[2733]: WS   /ws - 192.168.123.226, clients: 1
Fri Sep 10 23:37:33 2021 daemon.notice ttyd[2733]: started process, pid: 6676
Fri Sep 10 23:37:40 2021 auth.info login[6677]: root login on 'pts/0'
Fri Sep 10 23:44:45 2021 auth.info sshd[9479]: Accepted password for root from 192.168.123.226 port 63101 ssh2
root@OpenWrt:~#

lxhao61, 2021-09-10:
此模块不光运行问题，[最近此部分源码升级后，根本无法编译了。](https://github.com/lxhao61/Actions-Lean-OpenWrt/runs/3556978214?check_suite_focus=true)
[另外还有其它问题。](https://github.com/coolsnowwolf/lede/commit/4e1354dc490e59476bf605ab4d9b01000a1f1d90#comments)

baby195lxl, 2021-09-11:
> 此应用相关模块不光运行问题，[最近shortcut-fe升级后，根本无法编译了。](https://github.com/lxhao61/Actions-Lean-OpenWrt/runs/3556978214?check_suite_focus=true)
> [另外还有其它问题。](https://github.com/coolsnowwolf/lede/commit/4e1354dc490e59476bf605ab4d9b01000a1f1d90#comments)

谢谢回复，我还以为是我的编译问题

Ailick, 2021-09-11:
Fri Sep 10 23:10:32 2021 daemon.notice procd: /etc/rc.d/S90turboacc: insmod: ERROR: could not load module shortcut-fe: No such file or directory
/etc/rc.d/S90turboacc: insmod: ERROR: could not load module shortcut-fe-ipv6: No such file or directory
不确定你是否成功编译了sfe，我想看看你的`/lib/modules/内核版本号/`中是否有这个两个文件
其他的几个问题今天下午会被解决。

baby195lxl, 2021-09-11:
> Fri Sep 10 23:10:32 2021 daemon.notice procd: /etc/rc.d/S90turboacc: insmod: ERROR: could not load module shortcut-fe: No such file or directory
> /etc/rc.d/S90turboacc: insmod: ERROR: could not load [...]

[image]
存在shortcut-fe.ko和shortcut-fe-ipv6.ko文件

baby195lxl, 2021-09-11:
> Fri Sep 10 23:10:32 2021 daemon.notice procd: /etc/rc.d/S90turboacc: insmod: ERROR: could not load module shortcut-fe: No such file or directory
> /etc/rc.d/S90turboacc: insmod: ERROR: could not load [...]

[image]
存在shortcut-fe.ko，shortcut-fe-cm.ko和shortcut-fe-ipv6.ko文件

Ailick, 2021-09-11:
> [image]
> 存在shortcut-fe.ko，shortcut-fe-cm.ko和shortcut-fe-ipv6.ko文件

ok等我pr

baby195lxl, 2021-09-11:
> ok等我pr

谢谢开发者

Ailick, 2021-09-11:
> 谢谢开发者

请更新，已经解决此问题

baby195lxl, 2021-09-11:
> 请更新，已经解决此问题

好的，谢谢

closed 2021-09-11 by baby195lxl