coolsnowwolf/lede #7848: cron err 计划任务失效

Kisoul, 2021-09-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1、系统日志中出现如下大量错误日志
      2、计划任务貌似都失效了

### (2) 路由器型号和固件版本

A：cpu: i5 7200U  
      版本：2021.09.08 编译的OpenWrt R21.8.6   x86 系统 
      运行环境 ：esxi虚拟机
      

### (3) 详细日志

A：Sat Sep 11 22:38:50 2021 cron.err crond[31868]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:39:20 2021 cron.err crond[641]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:39:20 2021 cron.err crond[880]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:39:23 2021 user.err ddns-scripts[10490]: ileyee: wget 错误代码: '8'
Sat Sep 11 22:39:24 2021 user.warn ddns-scripts[10490]: ileyee: 传输失败 - 1/0 在 60 秒后重试
Sat Sep 11 22:39:49 2021 cron.err crond[4147]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:39:49 2021 cron.err crond[4390]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:40:19 2021 cron.err crond[5624]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:40:19 2021 cron.err crond[5858]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:40:49 2021 cron.err crond[8427]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:40:49 2021 cron.err crond[8661]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:41:18 2021 cron.err crond[11938]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:41:19 2021 cron.err crond[12171]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:41:44 2021 cron.err crond[13234]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:41:48 2021 cron.err crond[13589]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:41:48 2021 cron.err crond[13838]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:42:18 2021 cron.err crond[15226]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:42:18 2021 cron.err crond[15460]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:42:21 2021 authpriv.info dropbear[8592]: Exit (root) from <192.168.2.55:10187>: Exited normally
Sat Sep 11 22:42:21 2021 authpriv.info dropbear[14553]: Exit (root) from <192.168.2.55:38894>: Exited normally
Sat Sep 11 22:42:48 2021 cron.err crond[16633]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:42:48 2021 cron.err crond[16866]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:43:09 2021 cron.err crond[17715]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:43:17 2021 cron.err crond[18234]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:43:17 2021 cron.err crond[18468]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:43:47 2021 cron.err crond[19657]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:43:47 2021 cron.err crond[19893]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:44:17 2021 cron.err crond[21120]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:44:17 2021 cron.err crond[21384]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:44:46 2021 cron.err crond[22560]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:44:46 2021 cron.err crond[22799]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:45:16 2021 cron.err crond[24020]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:45:16 2021 cron.err crond[24255]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:45:41 2021 cron.err crond[25189]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:45:46 2021 cron.err crond[25595]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:45:46 2021 cron.err crond[25832]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:46:15 2021 cron.err crond[27169]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:46:15 2021 cron.err crond[27481]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:46:45 2021 cron.err crond[28576]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:46:45 2021 cron.err crond[28810]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:47:15 2021 cron.err crond[30051]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:47:15 2021 cron.err crond[30284]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:47:44 2021 cron.err crond[31464]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:47:45 2021 cron.err crond[31697]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:48:14 2021 cron.err crond[476]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:48:14 2021 cron.err crond[713]: crond (busybox 1.33.1) started, log level 8
Sat Sep 11 22:48:44 2021 cron.err crond[1978]: crond (busybox 1.33.1) started, log level 8
[... 647 lines cut ...]
Sun Sep 12 01:16:15 2021 cron.err crond[11093]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:16:45 2021 cron.err crond[12316]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:16:45 2021 cron.err crond[12553]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:17:15 2021 cron.err crond[13782]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:17:15 2021 cron.err crond[14017]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:17:44 2021 cron.err crond[15267]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:17:44 2021 cron.err crond[15505]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:18:14 2021 cron.err crond[16800]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:18:14 2021 cron.err crond[17033]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:18:44 2021 cron.err crond[18275]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:18:44 2021 cron.err crond[18514]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:19:13 2021 cron.err crond[19784]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:19:14 2021 cron.err crond[20019]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:19:43 2021 cron.err crond[21262]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:19:43 2021 cron.err crond[21527]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:20:13 2021 cron.err crond[22811]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:20:13 2021 cron.err crond[23044]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:20:42 2021 cron.err crond[24207]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:20:43 2021 cron.err crond[24521]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:21:12 2021 cron.err crond[27833]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:21:12 2021 cron.err crond[28072]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:21:42 2021 cron.err crond[29303]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:21:42 2021 cron.err crond[29537]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:22:12 2021 cron.err crond[30786]: crond (busybox 1.33.1) started, log level 8
Sun Sep 12 01:22:12 2021 cron.err crond[31074]: crond (busybox 1.33.1) started, log level 8

deplives, 2021-09-12:
j4125 esxi
也有类似的问题，自己写了个ddns的脚本，计划任务每五分钟执行一次，然后我就收到了大量的系统错误日志
Sat Sep 11 18:10:00 2021 cron.err crond[26623]: USER root pid 12765 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:15:00 2021 cron.err crond[26623]: USER root pid 13612 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:20:00 2021 cron.err crond[26623]: USER root pid 14355 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:25:00 2021 cron.err crond[26623]: USER root pid 15095 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:30:00 2021 cron.err crond[26623]: USER root pid 15841 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:35:00 2021 cron.err crond[26623]: USER root pid 16590 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:40:00 2021 cron.err crond[26623]: USER root pid 17336 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:45:00 2021 cron.err crond[26623]: USER root pid 18089 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:50:00 2021 cron.err crond[26623]: USER root pid 18856 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 18:55:00 2021 cron.err crond[26623]: USER root pid 19602 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:00:00 2021 cron.err crond[26623]: USER root pid 20347 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:05:00 2021 cron.err crond[26623]: USER root pid 21094 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:10:00 2021 cron.err crond[26623]: USER root pid 21847 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:15:00 2021 cron.err crond[26623]: USER root pid 22586 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:20:00 2021 cron.err crond[26623]: USER root pid 23414 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:25:00 2021 cron.err crond[26623]: USER root pid 24168 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:30:00 2021 cron.err crond[26623]: USER root pid 24920 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:35:00 2021 cron.err crond[26623]: USER root pid 25737 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:40:00 2021 cron.err crond[26623]: USER root pid 26490 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:45:00 2021 cron.err crond[26623]: USER root pid 27382 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:50:00 2021 cron.err crond[26623]: USER root pid 28167 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 19:55:00 2021 cron.err crond[26623]: USER root pid 28918 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 20:00:00 2021 cron.err crond[26623]: USER root pid 29666 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log
Sat Sep 11 20:05:00 2021 cron.err crond[26623]: USER root pid 30473 cmd /usr/bin/python3 -u /root/ddns.py >> /root/ddns.log

但是实际脚本是运行了，只是在系统日志里面显示 cron.err 
不只有这一个计划任务是这样
包括 ssrp coremark adbyby 也是一样的
Wed Sep  8 03:00:00 2021 cron.err crond[9949]: USER root pid 19898 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >/var/log/ssrplus.log
Wed Sep  8 04:00:00 2021 cron.err crond[9949]: USER root pid 28987 cmd /etc/coremark.sh
Wed Sep  8 06:00:00 2021 cron.err crond[9949]: USER root pid 14383 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1

Kisoul, 2021-09-15:
问题找到了，我发现我之前设置的NodeJS版的网易云音乐解锁插件没有正常运行，换了Golang版本后一切正常了，但是话说为什么会有这个问题？

closed 2021-09-15 by Kisoul