coolsnowwolf/lede #7850: 编译ax3600时遭遇交互，但不明白什么意思！

JohyC, 2021-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
ax3600编译时遇到选项
`Miscellaneous devices (INPUT_MISC) `
全部选N后，固件编译出来了。但是不知道会不会有啥问题！
望大佬看见，解答一下！

### (2) 路由器型号和固件版本

A：xiaomiAX3600

### (3) 详细日志

A：完整日志丢失了。。只有交互这部分保留了下来。
[Miscellaneous devices.txt](https://github.com/coolsnowwolf/lede/files/7148983/Miscellaneous.devices.txt)

Boos4721, 2021-09-12:
https://github.com/Boos4721/openwrt/commit/0445bbebc4655b5402fb57ab3fb522fe9e9a0520

JohyC, 2021-09-12:
> [Boos4721/openwrt@0445bbe](https://github.com/Boos4721/openwrt/commit/0445bbebc4655b5402fb57ab3fb522fe9e9a0520)

大佬！我就是用你的.config配置修改编译的！ 
以下我的配置！
[ax3600.txt](https://github.com/coolsnowwolf/lede/files/7149032/ax3600.txt)
🤣

Boos4721, 2021-09-12:
> > [Boos4721/openwrt@0445bbe](https://github.com/Boos4721/openwrt/commit/0445bbebc4655b5402fb57ab3fb522fe9e9a0520)
> 
> 大佬！我就是用你的.config配置修改编译的！
> 以下我的配置！
> [ax3600.txt](https://github.com/coolsnowwolf/lede/files/7149032/ax3600.txt)
> 🤣

两者不一样

JohyC, 2021-09-12:
> > > [Boos4721/openwrt@0445bbe](https://github.com/Boos4721/openwrt/commit/0445bbebc4655b5402fb57ab3fb522fe9e9a0520)
> > 
> > 
> > 大佬！我就是用你的.config配置修改编译的！
> > 以下我的配置！
> > [ax3600.txt](https://github.com/coolsnowwolf/lede/files/7149032/ax3600.txt)
> > 🤣
> 
> 两者不一样

我看到了！我正在修改我的源码库！谢谢大佬提示！

JohyC, 2021-09-12:
> [Boos4721/openwrt@0445bbe](https://github.com/Boos4721/openwrt/commit/0445bbebc4655b5402fb57ab3fb522fe9e9a0520)

大佬！我用你的源码进行编译，提示以下交互！
我想请问一下！如何才能避免交互呢？

`  INSTALL /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/user_headers/include
make[5]: Leaving directory '/home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63'
grep '=[ym]' /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/.config.set | LC_ALL=C sort | /home/johy/openwrt/staging_dir/host/bin/mkhash md5 > /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/.vermagic
touch /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/.configured
rm -f /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/vmlinux /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63/System.map
make -C /home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63 KCFLAGS="-fmacro-prefix-map=/home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl" HOSTCFLAGS="-O2 -I/home/johy/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="Boos4721" KBUILD_BUILD_HOST="Boos" KBUILD_BUILD_TIMESTAMP="Sat Sep 11 10:06:49 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/johy/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/johy/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.63 CC="ccache_cc" Image dtbs modules
make[5]: Entering directory '/home/johy/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.63'
  SYNC    include/config/auto.conf.cmd
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:480:warning: multi-line strings not supported
*
* Restart config...
*
*
* Scheduler features
*
Enable utilization clamping for RT/FAIR tasks (UCLAMP_TASK) [N/y/?] (NEW)

*
* Miscellaneous devices
*
Miscellaneous devices (INPUT_MISC) [Y/n/?] y
  Analog Devices AD714x Capacitance Touch Sensor (INPUT_AD714X) [N/m/?] n
  Atmel Capacitive Touch Button Driver (INPUT_ATMEL_CAPTOUCH) [N/m/?] n
  BMA150/SMB380 acceleration sensor support (INPUT_BMA150) [N/m/?] n
  NI Ettus Research USRP E3xx Button support. (INPUT_E3X0_BUTTON) [N/m/?] n
  Qualcomm PM8941 power key support (INPUT_PM8941_PWRKEY) [N/m/?] (NEW) n
  Qualcomm PM8XXX vibrator support (INPUT_PM8XXX_VIBRATOR) [N/m/?] (NEW) n
  MMA8450 - Freescale's 3-Axis, 8/12-bit Digital Accelerometer (INPUT_MMA8450) [N/m/?] n
  Generic GPIO Beeper support (INPUT_GPIO_BEEPER) [N/m/?] n
  Polled GPIO Decoder Input driver (INPUT_GPIO_DECODER) [N/m/?] n
  GPIO vibrator support (INPUT_GPIO_VIBRA) [N/m/?] n
  ATI / Philips USB RF remote control (INPUT_ATI_REMOTE2) [N/m/?] n
  Keyspan DMR USB remote control (INPUT_KEYSPAN_REMOTE) [N/m/?] n
  Kionix KXTJ9 tri-axis digital accelerometer (INPUT_KXTJ9) [N/m/?] n
  Griffin PowerMate and Contour Jog support (INPUT_POWERMATE) [N/m/?] n
  Yealink usb-p1k voip phone (INPUT_YEALINK) [N/m/?] n
  C-Media CM109 USB I/O Controller (INPUT_CM109) [N/m/?] n
  Regulator haptics support (INPUT_REGULATOR_HAPTIC) [N/m/?] n
  User level driver support (INPUT_UINPUT) [N/m/?] n
  PCF8574 Keypad input device (INPUT_PCF8574) [N/m/?] n
  Rotary encoders connected to GPIO pins (INPUT_GPIO_ROTARY_ENCODER) [N/m/?] n
  Analog Devices ADXL34x Three-Axis Digital Accelerometer (INPUT_ADXL34X) [N/m/?] n
  IMS Passenger Control Unit driver (INPUT_IMS_PCU) [N/m/?] n
  Azoteq IQS269A capacitive touch controller (INPUT_IQS269A) [N/m/?] n
  VTI CMA3000 Tri-axis accelerometer (INPUT_CMA3000) [N/m/?] n
  TI DRV260X haptics support (INPUT_DRV260X_HAPTICS) [N/m/?] n
  TI DRV2665 haptics support (INPUT_DRV2665_HAPTICS) [N/m/?] n
  TI DRV2667 haptics support (INPUT_DRV2667_HAPTICS) [N/m/?] n
*
* Power supply class support
*
Power supply class support (POWER_SUPPLY) [Y/?] y
  Power supply debug (POWER_SUPPLY_DEBUG) [N/y/?] n
[... 6 lines cut ...]
  DS2782/DS2786 standalone gas-gauge (BATTERY_DS2782) [N/m/y/?] n
  SBS Compliant gas gauge (BATTERY_SBS) [N/m/y/?] n
  SBS Compliant charger (CHARGER_SBS) [N/m/y/?] n
  BQ27xxx battery driver (BATTERY_BQ27XXX) [N/m/y/?] n
  Maxim MAX17040 Fuel Gauge (BATTERY_MAX17040) [N/m/y/?] n
  Maxim MAX17042/17047/17050/8997/8966 Fuel Gauge (BATTERY_MAX17042) [N/m/y/?] n
  MAX8903 Battery DC-DC Charger for USB and Adapter Power (CHARGER_MAX8903) [N/m/y/?] n
  TI/National Semiconductor LP8727 charger driver (CHARGER_LP8727) [N/m/y/?] n
  GPIO charger (CHARGER_GPIO) [N/m/y/?] n
  Battery charger manager for multiple chargers (CHARGER_MANAGER) [N/m/y/?] n
  Analog Devices LT3651 charger (CHARGER_LT3651) [N/m/y/?] n
  Maxim MAX14656 USB charger detector (CHARGER_DETECTOR_MAX14656) [N/m/y/?] n
[... cut ...]
  TI BQ25890 battery charger driver (CHARGER_BQ25890) [N/m/y/?] n
  TI BQ25980 battery charger driver (CHARGER_BQ25980) [N/m/y/?] n
  Summit Microelectronics SMB3XX Battery Charger (CHARGER_SMB347) [N/m/y/?] n
  LTC2941/LTC2943 Battery Gauge Driver (BATTERY_GAUGE_LTC2941) [N/m/y/?] n
  RT5033 fuel gauge support (BATTERY_RT5033) [N/m/y/?] (NEW) n
  Richtek RT9455 battery charger driver (CHARGER_RT9455) [N/m/y/?] n
  Microchip UCS1002 USB Port Power Controller (CHARGER_UCS1002) [N/m/y/?] n
  ROHM bd99954 charger driver (CHARGER_BD99954) [N/m/y/?] n
`

closed 2021-09-12 by JohyC