coolsnowwolf/lede #7852: 采用最新源码编译的固件，n1盒子会突然重启

baby195lxl, 2021-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：采用最新源码编译的固件，n1盒子会突然重启

### (2) 路由器型号和固件版本

A：斐讯N1盒子

### (3) 详细日志

A：OpenWrt login: root
Password: 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
\    ____                 _       __     __  /
\   / __ \____  ___  ____| |     / /____/ /_ /
\  / / / / __ \/ _ \/ __ \ | /| / / ___/ __/ /
\ / /_/ / /_/ /  __/ / / / |/ |/ / /  / /_   /
\ \____/ .___/\___/_/ /_/|__/|__/_/   \__/   /
\     /_/  W I R E L E S S   F R E E D O M   /
\                                            /
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 Base on OpenWrt R21.8.6 by lean & lienol
 Kernel 5.4.144-flippy-63+o
 Packaged by Rosaorin on 2021-09-11
 SOC: s905d     BOARD: n1

设备信息： Phicomm N1
CPU 型号:  AArch64 : Cortex-A53 x 4
系统负载:  0.36 0.19 0.07       运行时间:  1分钟 52秒
环境温度:  45.0 °C              当前频率:  667 Mhz
内存已用:  8% of 1807MB         IP  地址:  192.168.123.2
启动存储:  12% of 511.0M        系统存储:  41% of 720.0M 

root@OpenWrt:~# logread
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Sun Sep 12 23:26:28 2021 kern.notice kernel: [    0.000000] Linux version 5.4.144-flippy-63+o (root@gtking-pro) (gcc version 11.2.1 20210816 [revision 056e324ce46a7924b5cf10f61010cf9dd2ca10e9] (SUSE Linux)) #318 SMP PREEMPT Fri Sep 3 23:04:42 CST 2021
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Machine model: Phicomm N1
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] efi: Getting EFI parameters from FDT:
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] efi: UEFI not found.
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x0000000063800000, size 256 MiB
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 478464
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 7616 pages used for memmap
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000]   DMA32 zone: 478464 pages, LIFO batch:63
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] psci: PSCIv0.2 detected in firmware.
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] psci: Trusted OS migration not required
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s47576 r8192 d30248 u86016
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s47576 r8192 d30248 u86016 alloc=21*4096
Sun Sep 12 23:26:28 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 470848
Sun Sep 12 23:26:28 2021 kern.notice kernel: [    0.000000] Kernel command line: root=UUID=bbdc794a-9652-4674-9cef-23953fd823b1 rootfstype=btrfs rootflags=compress=zstd console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 mac=FC:7C:02:89:E4:7C
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Sep 12 23:26:28 2021 kern.info kernel: [    0.000000] Memory: 1582300K/1913856K available (12476K kernel code, 1446K rwdata, 7412K rodata, 896K init, 851K bss, 69412K reserved, 262144K cma-reserved)
Sun Sep 12 23:26:28 2021 kern.notice kernel: [    0.000000] random: get_random_u64 called from __kmem_cache_create+0x38/0x570 with crng_init=0
[... 476 lines cut ...]
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]:   terminal type: xterm-256color
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]: LWS: 4.1.6-20200915-807-g12d4b91c8, loglevel 7
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]: NET CLI SRV H1 H2 WS IPV6-off
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]:    /usr/lib/libwebsockets-evlib_uv.so
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]:  Using foreign event loop...
Sun Sep 12 23:26:32 2021 daemon.notice ttyd[2745]:  Listening on port: 7681
Sun Sep 12 23:26:32 2021 daemon.info procd: - init complete -
Sun Sep 12 23:26:32 2021 daemon.err dockerd[2581]: time="2021-09-12T23:26:32.983844127+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sun Sep 12 23:26:32 2021 daemon.err dockerd[2581]: time="2021-09-12T23:26:32.986437211+08:00" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sun Sep 12 23:26:33 2021 daemon.info avahi-daemon[1634]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.31.0.1.
Sun Sep 12 23:26:33 2021 daemon.info avahi-daemon[1634]: New relevant interface docker0.IPv4 for mDNS.
Sun Sep 12 23:26:33 2021 daemon.info avahi-daemon[1634]: Registering new address record for 172.31.0.1 on docker0.IPv4.
Sun Sep 12 23:28:11 2021 daemon.info pdnsd[3116]: pdnsd-1.2.9b-par starting.
Sun Sep 12 23:28:11 2021 auth.info sshd[3376]: Accepted password for root from 192.168.123.226 port 63894 ssh2
Sun Sep 12 23:28:18 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Sep 12 23:28:18 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Sep 12 23:28:19 2021 cron.err crond[4293]: crond (busybox 1.33.1) started, log level 5
Sun Sep 12 23:28:19 2021 daemon.info dnsmasq[4170]: Connected to system UBus
Sun Sep 12 23:28:34 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Sep 12 23:29:38 2021 daemon.notice ttyd[2745]: HTTP / - 192.168.123.226
Sun Sep 12 23:29:38 2021 daemon.notice ttyd[2745]: HTTP /token - 192.168.123.226
Sun Sep 12 23:29:38 2021 daemon.notice ttyd[2745]: WS   /ws - 192.168.123.226, clients: 1
Sun Sep 12 23:29:38 2021 daemon.notice ttyd[2745]: started process, pid: 4564
Sun Sep 12 23:29:43 2021 auth.info login[4565]: root login on 'pts/1'
root@OpenWrt:~#

coolsnowwolf, 2021-09-12:
SFE 的补丁，N1 这边的 Patch 需要更新

baby195lxl, 2021-09-12:
> SFE 的补丁，N1 这边的 Patch 需要更新

谢谢回复，所以是SFE的问题吗？

unifreq, 2021-09-17:
5.4.146-flippy-64+o、5.10.65-flippy-64+内核已修复此问题。

closed 2021-09-17 by baby195lxl