coolsnowwolf/lede #7856: SSR Plus +

peter123456789, 2021-09-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
无端端丢弃我的节点信息，amytelecom机场无法获取节点

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
问题现象：
2021-09-12 23:06:51 订阅更新成功
2021-09-12 23:06:51 新增节点数量: 0 删除节点数量: 0
2021-09-12 23:06:51 成功解析节点数量: 41
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 新加坡 04
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 新加坡 03
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 新加坡 02
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 新加坡 01
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 台湾 05
2021-09-12 23:06:51 丢弃无效节点: ss 节点, 台湾 04

[image]

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
esri v9 高大全最新版本

baby195lxl, 2021-09-13:
这是OP的库，你为啥要在这里提issue问别的库的东西呢

peter123456789, 2021-09-13:
> 这是OP的库，你为啥要在这里提issue问别的库的东西呢

没有的，源码在这里。没有修改，只是按照源码编译

EnnawYang, 2021-09-13:
源码不在这里

peter123456789, 2021-09-13:
> 源码不在这里

帮忙给下源码地址  谢谢了

minlang112, 2021-09-13:
> > 源码不在这里
> 
> 帮忙给下源码地址 谢谢了

出门左转→ https://github.com/fw876/helloworld

1715173329, 2021-09-16:
很简单，你的机场还在用过时的流加密套件，建议更换。

coolsnowwolf, 2026-09-06:
SSR Plus+ 订阅丢弃节点通常由订阅格式、节点参数或应用自身解析/过滤规则决定；报告没有订阅样本、解析器版本和可复现配置，无法确认现行 LEDE 包缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf